espressif/arduino-esp32 #7437: SyntaxError: invalid syntax in library

luutoan322001, 2022-11-07:
### Board

ESP32-C3-32S

### Device Description

It connect directly to a Ubuntu 16.04 system

### Hardware Configuration

Nothing

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 18.

### Operating System

Ubuntu 16.04

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

It cannot build the script, even the blank one. I try with the example code, it is also not work

### Sketch

```cpp
//Blank code
```

### Debug Message

```plain
Arduino: 1.8.19 (Linux), Board: "ESP32C3 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 160MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None, Disabled"

Traceback (most recent call last):
  File "/home/lou3hc/arduino-1.8.19/hardware/espressif/esp32/tools/esptool/esptool.py", line 31, in <module>
    import esptool
  File "/home/lou3hc/arduino-1.8.19/hardware/espressif/esp32/tools/esptool/esptool/__init__.py", line 42, in <module>
    from esptool.cmds import (
  File "/home/lou3hc/arduino-1.8.19/hardware/espressif/esp32/tools/esptool/esptool/cmds.py", line 14, in <module>
    from .bin_image import ELFFile, ImageSegment, LoadFirmwareImage
  File "/home/lou3hc/arduino-1.8.19/hardware/espressif/esp32/tools/esptool/esptool/bin_image.py", line 14, in <module>
    from .loader import ESPLoader
  File "/home/lou3hc/arduino-1.8.19/hardware/espressif/esp32/tools/esptool/esptool/loader.py", line 274
    raise FatalError(f"Could not open {port}, the port doesn't exist")
                                                                    ^
SyntaxError: invalid syntax
exit status 1
Error compiling for board ESP32C3 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-11-07:
It seems that ESP32 Boards from Arduino IDE has not been correctly installed.
Another possibility is that installed Python is way too old.

Could you please open a shell and type `python --version`?

VojtechBartoska, 2022-12-15:
Closing as expired, if needed you can reopen this.

closed 2022-12-15 by VojtechBartoska