espressif/arduino-esp32 #7440: Wifi problem in ESP32 Board

suryanshRoy, 2022-11-07:
### Board

Esp32 Dev Module 

### Device Description

Esp32 Dev module ,  chip is cp2102

### Hardware Configuration

No

### Version

v2.0.3

### IDE Name

Arduino IDE 

### Operating System

Windows 10

### Flash frequency

80mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

In the first Time my board gets connected to the WiFi and after 10 to 20 second the Board gets disconnected to the WiFi and unable to reconnect to the WiFi until I press the EN button in my board 
and it print some error in the Serial Monitor whenever it gets disconnected from the WiFi.
       Please help 

### Sketch

```cpp
case ARDUINO_EVENT_WIFI_STA_DISCONNECTED:
      digitalWrite(wifiLed, false);
      break;
      case WIFI_PROV_STA_CONNECTING:
      {
      Serial.println("\nDisconnected. Connecting to the AP again... ");
      delay(20000);
      }
      break;
    case ARDUINO_EVENT_WIFI_STA_CONNECTED:
      Serial.printf("\nConnected to Wi-Fi!\n");
      digitalWrite(wifiLed, true);
      break;
```

### Debug Message

```plain
E (621984) TRANSPORT_BASE: poll_read select error 113, errno = Software caused connection abort, fd = 48
E (621986) MQTT_CLIENT: Poll read error: 119, aborting connection
E (643762) esp-tls: couldn't get hostname for :a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0
E (643762) esp-tls: Failed to open new connection
E (643767) TRANSPORT_BASE: Failed to open a new connection
E (643772) MQTT_CLIENT: Error transport connect
E (643776) esp_mqtt_glue: MQTT_EVENT_ERROR
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-11-07:
Please update to 2.0.5, enable Core Debug to level Verbose (in the board menu) and try again. Then post the logs so we can see what is the WiFi doing. The logs you provided are from MQTT

mrengineer7777, 2022-11-07:
I recommend turning on Verbose debugging as me-no-dev suggested.

May be resolved by #7344.

suryanshRoy, 2022-11-08:
I did as me-no-dev said but I maked this code for ESP RAINMAKER and now it is unable to connect with the WiFi in ESP RAINMAKER and now in verbose it printing something like this :-

[    23][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 0 - WIFI_READY[   188][I][WiFiProv.cpp:142] beginProvision(): Starting AP using BLE. service_name : ESP, pop : ESP RAIN MAKER
[   188][V][WiFiGeneric.cpp:440] _arduino_event_cb(): Provisioning Initialized![   350][V][WiFiGeneric.cpp:283] _arduino_event_cb(): STA Started
[   336][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 33 - PROV_INIT
[   363][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 2 - STA_START
[   785][V][WiFiGeneric.cpp:446] _arduino_event_cb(): Provisioning Start!
[   785][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 35 - PROV_START
[183372][V][WiFiGeneric.cpp:317] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 1
[183372][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 1 - SCAN_DONE

So there is nothing by doing this if ESP RAINMAKER Is not Connecting to the WiFi

me-no-dev, 2022-11-08:
Log shows only that provisioning started, but not that it completed, so it could not even start connecting to any WiFi. Maybe you need to give a bit more details on what you are doing, code you are running and so on.

suryanshRoy, 2022-11-09:
/**********************************************************************************
 *  TITLE: ESP RainMaker + IR + Manual Switch control 8 Relays using ESP32 (store last GPIO state + Real time feedback + no WiFi control)
 *  Click on the following links to learn more. 
 *  YouTube Video: https://youtu.be/s_DGYsd4T6I
 *  Related Blog : https://iotcircuithub.com/esp32-project-with-alexa-google-home/
 *  by Tech StudyCell
 *  Preferences--> Aditional boards Manager URLs : 
 *  http://arduino.esp8266.com/stable/package_esp8266com_index.json,https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
 *  
 *  Download Board ESP32 (2.0.3): https://github.com/espressif/arduino-esp32
 *
 *  Download the libraries 
 *  IRremote Library (3.6.1): https://github.com/Arduino-IRremote/Arduino-IRremote
 **********************************************************************************/

#include "RMaker.h"
#include "WiFi.h"
#include "WiFiProv.h"
#include <IRremote.h>
#include <Preferences.h>
Preferences pref;

const char *service_name = "PROV_1234RB8";
const char *pop = "12345RB8";

// define the Chip Id
uint32_t espChipId = 0;

// define the Node Name
char nodeName[] = "ESP32";

// define the Device Names
char deviceName_1[] = "Light 1";
char deviceName_2[] = "Light 2";
char deviceName_3[] = "Light 3";
char deviceName_4[] = "Light 4";

//Update the HEX code of IR Remote buttons 0x<HEX CODE>
#define IR_Button_1   0x1FE50AF
#define IR_Button_2   0x1FED827
#define IR_Button_3   0x1FEF807
#define IR_Button_4   0x1FE30CF
#define IR_All_Off    0x1FEE01F
#define IR_All_On     0x1FE48B7

// define the GPIO connected with Relays and switches
static uint8_t RelayPin1 = 23;  //D23
static uint8_t RelayPin2 = 22;  //D22
static uint8_t RelayPin3 = 21;  //D21
static uint8_t RelayPin4 = 19;  //D19

static uint8_t SwitchPin1 = 13;  //D13
static uint8_t SwitchPin2 = 12;  //D12
static uint8_t SwitchPin3 = 14;  //D14
static uint8_t SwitchPin4 = 27;  //D27

static uint8_t wifiLed      = 2;  // D2
static uint8_t gpio_reset   = 0;  // Press BOOT for reset WiFi
static uint8_t IR_RECV_PIN  = 35; // D35 (IR receiver pin)

/* Variable for reading pin status*/
// Relay State
bool toggleState_1 = LOW; //Define integer to remember the toggle state for relay 1
bool toggleState_2 = LOW; //Define integer to remember the toggle state for relay 2
bool toggleState_3 = LOW; //Define integer to remember the toggle state for relay 3
bool toggleState_4 = LOW; //Define integer to remember the toggle state for relay 4

// Switch State
bool SwitchState_1 = LOW;
bool SwitchState_2 = LOW;
[... 326 lines cut ...]
          Serial.printf("Reset Wi-Fi.\n");
          // If key pressed for more than 3secs, but less than 10, reset Wi-Fi
          RMakerWiFiReset(2);
        }
    }
    delay(100);

    if (WiFi.status() != WL_CONNECTED)
    {
      //Serial.println("WiFi Not Connected");
      digitalWrite(wifiLed, false);
    }
    else
    {
      //Serial.println("WiFi Connected");
      digitalWrite(wifiLed, true);
    }

    ir_remote(); //IR remote Control
    manual_control();
}

This is the Code and there are Two problems In the whole code 
1st Is in ir_remote and that is after about pressing 10 to 20 times it receives a signal of Ir remote and It works Properly when I set Core debug level to Verbose .
2nd Problem is In Wi-Fi My board gets disconnect when I restart The Wi-Fi .

mrengineer7777, 2022-11-09:
1) IR issue should be unrelated to WiFi.  That's up to you to solve.  If debug printing causes it to work then you have a timing issue.
2) When you say "restart the Wi-Fi" do you mean restarted your router?  The upcoming 2.0.6 release includes #7344, which may resolve that issue.

If you're looking for fellow developers to help with your code, I recommend sharing it on a GitHub repository.  Obviously remove passwords, private keys, etc.  No guarantees we'll fix your project though.  We all have our own projects.

closed 2022-11-14 by suryanshRoy