espressif/arduino-esp32 #7447: HTTPS Server example

mpjmichala, 2022-11-08:
### Related area

Webserver

### Hardware specification

ESP32

### Is your feature request related to a problem?

I've found an information that a https server feature will be added in next major release:
https://github.com/espressif/arduino-esp32/issues/3902#issuecomment-781258806

I see that there are some files deep in folders, like this esp_https_server header file.

But to be honest, I'm pretty lost. Is this feature ready yet, and are there any examples how to use it? 

### Describe the solution you'd like

I would like to know if there are any examples available.
Like those pretty examples for standard http server.

Thanks a lot for any clue!

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

mpjmichala, 2022-11-14:
Thanks a lot!
So there is a bad situation now, when trying to create a project with https server.
Official arduino-esp32 is only http not https, and best known alternative from: https://github.com/fhessel/esp32_https_server
is based on opensll, which will be removed in next major esp-idf tools release and the library probably won't be rewritten (it is quiet abandoned).

VojtechBartoska, 2022-12-15:
cc @pedrominatel & @PilnyTomas

SuGlider, 2022-12-16:
Arduino may use IDF for HTTPS as described in the IDF [ESP HTTPS SERVER](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_https_server.html) documentation.

An IDF example can be found at https://github.com/espressif/esp-idf/tree/edd815af2e/examples/protocols/https_server

VojtechBartoska, 2023-02-16:
@PilnyTomas Can you please help with triage? Thanks

PilnyTomas, 2023-08-11:
I have nothing special to add. The IDF example works. It should be possible to copy them to our code base so it is more accessible to our users, but implementing a pretty Arduino-like API is in my opinion a low-priority feature request.

VojtechBartoska, 2023-10-23:
Closing this issue as wont do for now, thanks for understanding.

closed 2023-10-23 by VojtechBartoska

hemangjoshi37a, 2024-06-12:
has this been solved yet ?