espressif/arduino-esp32 #7448: EXT_RAM_NOINIT_ATTR random data on reboot

handmade0octopus, 2022-11-08:
### Board

Not board specific

### Device Description

WROOVER module, custom board

### Hardware Configuration

Not hardware specific

### Version

v2.0.5

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

quick question on WROOVER module with PSRAM

I use flags in platformio:
```build_flags = 
    -DBOARD_HAS_PSRAM=1
    -DCORE_DEBUG_LEVEL=1
    -DPCB_VER=40
#    -DCONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY=1 # tested with and without
    -DCONFIG_SPIRAM_ALLOW_NOINIT_SEG_EXTERNAL_MEMORY=1
```

And I try to use this type of data with `EXT_RAM_NOINIT_ATTR ` flag

```cpp
struct DABuffer {
    uint32_t timestamp;
    uint32_t digital;
    int16_t analog[ANALOG_SIZE];
};

uint32_t EXT_RAM_NOINIT_ATTR bufferSize = 0;
DABuffer EXT_RAM_NOINIT_ATTR buffer[BUFFER_SIZE];
```

Problem is on each reboot (either from RST pin or software reset):

This is example of data check during runtime:
```cpp
[ 10469][E][Devices.hpp:107] addToBuffer(): Fps  10.10 Buffsize:   100 Stmp: 10435 D:   00000000   A0: 25829 9.697 A1:     0 0.000 A2:     0 0.000 A3:     0 0.000
```

Log from random corrupted data after each reboot
```cpp
[   524][E][Devices.hpp:120] devicesTask(): bufferSize -1431657526  Stamp: 1073470136  Digi[0] 003FF797  Anal[0] -2153 -0.808

[   513][E][Devices.hpp:120] devicesTask(): bufferSize 0  Stamp: 1073470136  Digi[0] 003F8E3F  Anal[0] -29733 -11.163
```

Note I am using esp_spiram_writeback_cache() every so often and before issuing ESP.restart();

Many thanks for helping me how can I debug this!

### Sketch

```cpp
void restart() {
	esp_spiram_writeback_cache();
	delay(1);
	ESP.restart();
}
```

### Debug Message

```plain
In description
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-11-09:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/memory-types.html#noinit-dram
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/sdkconfig#L656
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-allow-noinit-seg-external-memory

@handmade0octopus - it seems not configured for Arduino Core.

In order to get it, you may need to move to Arduino as IDF Component project and change sdkconfig manually.

handmade0octopus, 2022-11-09:
@SuGlider thank you I will try that (although I am not a fan of how IDF recompiles take ages) and let you know.

I see noinit for PSRAM is a #ifdef inside Arduino headers so any idea it would be disabled and if its possible to make it enabled in new version? What is the reason it would be disabled otherwise?

SuGlider, 2022-11-09:
This is a Linker Section attribute. 
Try this in your stetch:

``` cpp
undef EXT_RAM_NOINIT_ATTR
#define EXT_RAM_NOINIT_ATTR _SECTION_ATTR_IMPL(".ext_ram_noinit", __COUNTER__)
```

I can see that ESP32 `sections.ld` has it.
It may work even not touching the sdkconfig file.

handmade0octopus, 2022-11-09:
@SuGlider It's the same unfortunately.

When I clicked on definition before it showed this:

```cpp
#if CONFIG_SPIRAM_ALLOW_NOINIT_SEG_EXTERNAL_MEMORY
// Forces data into external memory noinit section to avoid initialization after restart.
#define EXT_RAM_NOINIT_ATTR _SECTION_ATTR_IMPL(".ext_ram_noinit", __COUNTER__)
#else
```
Inside `platformio\packages\framework-arduinoespressif32\tools\sdk\esp32\include\esp_common\include\esp_attr.h`
It seems like it was defined already there. This is why I added `-DCONFIG_SPIRAM_ALLOW_NOINIT_SEG_EXTERNAL_MEMORY=1` to my build flags for platformio.ini

So not sure why it's not working though..

handmade0octopus, 2022-11-11:
Some more testing I done:

interesting, been playing with my PSRAM and something is not right..

Just after boot I run command `heap_caps_get_largest_free_block(MALLOC_CAP_SPIRAM)` to get my heap and everything is fine, then if I try to run the same command it crashes and my PSRAM is decimated for some uknown reason :0

```cpp
[   542][E][Config.hpp:68] configTask(): HEAP(KB)    311.39 /  343.84 Min:  306.02 Max:  115.99
[   550][E][Config.hpp:69] configTask(): PSRAM (KB) 4093.89 / 4093.89 Min: 4093.89 Max:    0.00
[  2556][E][Config.hpp:68] configTask(): HEAP(KB)    311.05 /  343.84 Min:  305.50 Max:  115.99
[  2566][E][Config.hpp:69] configTask(): PSRAM (KB)  255.99 / 4093.89 Min: 4194267.81 Max:    0.00
```

And here is crash if I call MAX after that:
```cpp
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008d7d8  PS      : 0x00060233  A0      : 0x8008feec  A1      : 0x3ffb4680
A2      : 0x00000214  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060223
A6      : 0x00060220  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0xffffffff
A10     : 0x3ffb481c  A11     : 0x3ffc16cc  A12     : 0x0000003f  A13     : 0x00000062
A14     : 0xc0800214  A15     : 0x003fffff  SAR     : 0x0000001b  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000214  LBEG    : 0x40087d9c  LEND    : 0x40087da7  LCOUNT  : 0x00000000

Backtrace:0x4008d7d5:0x3ffb46800x4008fee9:0x3ffb46c0 0x4009002e:0x3ffb46e0 0x400902a3:0x3ffb4700 0x400df746:0x3ffb4720 0x400df795:0x3ffb4760 0x400d3ee9:0x3ffb47a0 0x400d1c1e:0x3ffb47c0 0x400d1d17:0x3ffb4800

```

I was using `_SECTION_ATTR_IMPL(".ext_ram_noinit", __COUNTER__)` for one of the buffers and it caused this error

When I tried to use `_SECTION_ATTR_IMPL(".ext_ram.bss", __COUNTER__)` on the buffer something even stranger happend - I wasn't manage to force it to ext ram instead it used DRAM (as usage of RAM in sketch increased and with bigger buffer I wasn't able to use it).

So I tried `_SECTION_ATTR_IMPL(".ext_ram", __COUNTER__)` and it crashes nearly instantly (but I manage to get single heap size limit read).

So as you can see I am really confused.

usage is unrealistic `PSRAM (KB)  255.99 / 4093.89` is not how much I use in any way (even if I use only like 10 bytes it happens

So potential linker issue?

SuGlider, 2022-11-11:
When PSRAM is enabled, ESP32 Arduino will try to allocate anything over 4KB on it, even when using C `malloc()` or `new()`
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/sdkconfig#L654
Therefore, other tasks or Arduino core code can end up using it as well.

It's possible to control where a memory chuck is allocated by using the right `heap_caps_malloc(...)` prameter.

handmade0octopus, 2022-11-11:
> It's possible to control where a memory chuck is allocated by using the right heap_caps_malloc(...) prameter.
Ok well this is my solution for now but its not the best one.

Thank you for help.

handmade0octopus, 2022-11-13:
@SuGlider ok well heap allocation isn't the best either, it causes some kind of corruption on data (that isn't apparent when I use 8bit internal DRAM), I am not sure if its due to heap_malloc or due to some PSRAM checking when device boots but I cannot use this approach either.

```
   521][E][Devices.hpp:137] Buffer(): MaxBuffers:  42 PSram PSRAM (KB) 4093.85 / 4093.87 Min: 4093.85 Max: 4031.99 Diff 96588
[   531][E][Devices.hpp:112] check(): Chk: "CHECKST" noBuffers  0
[   537][E][Devices.hpp:87] check(): Force zero 0 buff          31  00000000
[   544][E][Devices.hpp:87] check(): Force zero 0 buff           0  AAAAF542
[   551][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   558][E][Devices.hpp:87] check(): Force zero 0 buff -1431588414  00000000
[   564][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   571][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   578][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   585][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   591][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   598][E][Devices.hpp:87] check(): Force zero 0 buff           0  AAA9E75A
[   605][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   612][E][Devices.hpp:87] check(): Force zero 0 buff -1431395366  00000000
[   619][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   625][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   632][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   639][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   646][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   652][E][Devices.hpp:87] check(): Force zero 0 buff           0  AAAC9152
[   659][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   666][E][Devices.hpp:87] check(): Force zero 0 buff -1431544366  00000000
[   673][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   679][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   686][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   693][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   700][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   707][E][Devices.hpp:87] check(): Force zero 0 buff           0  AAA380AA
[   713][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   720][E][Devices.hpp:87] check(): Force zero 0 buff -1432137942  00000000
[   727][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   734][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   740][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   747][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   754][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   761][E][Devices.hpp:87] check(): Force zero 0 buff           0  AAA6B2A2
[   767][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   774][E][Devices.hpp:87] check(): Force zero 0 buff -1431930590  00000000
[   781][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   788][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   795][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   801][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   808][E][Devices.hpp:87] check(): Force zero 0 buff           0  00000000
[   815][E][Devices.hpp:87] check(): Force zero 0 buff           0  AAA5ACBA
[   829][E][Config.hpp:30] configTask(): Reason: 3 Sw
[   836][E][Config.hpp:38] configTask(): HEAP(KB)    311.61 /  343.80 Min:  306.24 Max:  115.99 PSRAM (KB)  156.17 / 4093.85 Min:  156.17 Max:  155.99
```
^ I am checking current element of array (left column) and [0] index of such array (right column). What is interesting exactly the same pattern is visible even when I change size of array they still appear on same indexes which is strange even more.

When I use internal DRAM with heap malloc just first element is fully FF.

me-no-dev, 2022-11-16:
@handmade0octopus we can look into adding it to future versions.

@igrr are you aware of any negative consequences if we enable that option?

VojtechBartoska, 2024-02-20:
postponing to 3.1.0, we need to evaluate this.

closed 2025-03-13 by me-no-dev