espressif/arduino-esp32 #7449: How to view SDK version through bin file?

hzhh110, 2022-11-09:
### Related area

ESP32-WROOM-32D

### Hardware specification

ESP32-WROOM-32D

### Is your feature request related to a problem?

How to view SDK version through bin file? You need to read the SDK version to troubleshoot problems and see if there are any differences

### Describe the solution you'd like

How to view SDK version through bin file

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2022-11-09:
are you asking how to get the version from already compiled file, or how to get it programmatically?

igrr, 2022-11-09:
The question has been answered in https://github.com/espressif/esp-idf/issues/10144.
@hzhh110 please avoid creating the same issue in multiple repositories in the future.

closed 2022-11-09 by SuGlider