espressif/arduino-esp32 #7453: Issue in IRremote receiver  and Linked with Esp rainmaker

suryanshRoy, 2022-11-10:
### Board

Esp32 Dev Module 

### Device Description

Esp32 Dev module chip is Cp2102 

### Hardware Configuration

Gpio 35 is connected to Tsop 1838 Ir receiver 

### Version

v2.0.5

### IDE Name

Arduino IDE 

### Operating System

Windows 10

### Flash frequency

80mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

When I upload the code and press the Ir Remote button about 10 to 20 times My board receives One Signal only and when I enable Core Debug to level Verbose my Board starts receiving all Signal 
But a new problem starts on verbose mode and that is in Esp Rainmaker It is unable to Print a QR code. So how can I do without enabling Core debug Level. 

### Sketch

```cpp
#include <IRremote.h>
//Update the HEX code of IR Remote buttons 0x
#define IR_Button_1 0x1FE50AF
#define IR_Button_2 0x1FED827
#define IR_Button_3 0x1FEF807
#define IR_Button_4 0x1FE30CF
#define IR_All_Off 0x1FEE01F
#define IR_All_On 0x1FE48B7

// define the GPIO connected with Relays
static uint8_t RelayPin1 = 23; //D23
static uint8_t RelayPin2 = 22; //D22
static uint8_t RelayPin3 = 21; //D21
static uint8_t RelayPin4 = 19; //D19

static uint8_t IR_RECV_PIN = 35; // D35 (IR receiver pin)

IRrecv irrecv(IR_RECV_PIN);
decode_results results;

void ir_remote(){
if (irrecv.decode(&results)) {
switch(results.value){
case IR_Button_1:
digitalWrite(RelayPin1, toggleState_1);
toggleState_1 = !toggleState_1;
pref.putBool("Relay1", toggleState_1);
my_switch1.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_1);
delay(100);
break;
case IR_Button_2:
digitalWrite(RelayPin2, toggleState_2);
toggleState_2 = !toggleState_2;
pref.putBool("Relay2", toggleState_2);
my_switch2.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_2);
delay(100);
break;
case IR_Button_3:
digitalWrite(RelayPin3, toggleState_3);
toggleState_3 = !toggleState_3;
pref.putBool("Relay3", toggleState_3);
my_switch3.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_3);
delay(100);
break;
case IR_Button_4:
digitalWrite(RelayPin4, toggleState_4);
toggleState_4 = !toggleState_4;
pref.putBool("Relay4", toggleState_4);
my_switch4.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_4);
delay(100);
break;
case IR_All_Off:
all_SwitchOff();
break;
case IR_All_On:
all_SwitchOn();
break;
default : break;
}
//Serial.println(results.value, HEX);
irrecv.resume();
}
}

void all_SwitchOff(){
toggleState_1 = 0; digitalWrite(RelayPin1, LOW); pref.putBool("Relay1", toggleState_1); my_switch1.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_1); delay(100);
toggleState_2 = 0; digitalWrite(RelayPin2, LOW); pref.putBool("Relay2", toggleState_2); my_switch2.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_2); delay(100);
toggleState_3 = 0; digitalWrite(RelayPin3, LOW); pref.putBool("Relay3", toggleState_3); my_switch3.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_3); delay(100);
toggleState_4 = 0; digitalWrite(RelayPin4, LOW); pref.putBool("Relay4", toggleState_4); my_switch4.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_4); delay(100);
}

void all_SwitchOn(){
toggleState_1 = 1; digitalWrite(RelayPin1, HIGH); pref.putBool("Relay1", toggleState_1); my_switch1.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_1); delay(100);
toggleState_2 = 1; digitalWrite(RelayPin2, HIGH); pref.putBool("Relay2", toggleState_2); my_switch2.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_2); delay(100);
toggleState_3 = 1; digitalWrite(RelayPin3, HIGH); pref.putBool("Relay3", toggleState_3); my_switch3.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_3); delay(100);
toggleState_4 = 1; digitalWrite(RelayPin4, HIGH); pref.putBool("Relay4", toggleState_4); my_switch4.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_4); delay(100);
}

void setup()
{
Serial.begin(115200);

// Set the Relays GPIOs as output mode
pinMode(RelayPin1, OUTPUT);
pinMode(RelayPin2, OUTPUT);
pinMode(RelayPin3, OUTPUT);
pinMode(RelayPin4, OUTPUT);
pinMode(IR_RECV_PIN, INPUT);

// Write to the GPIOs the default state on booting
digitalWrite(RelayPin1, !toggleState_1);
digitalWrite(RelayPin2, !toggleState_2);
digitalWrite(RelayPin3, !toggleState_3);
digitalWrite(RelayPin4, !toggleState_4);
digitalWrite(wifiLed, LOW);

irrecv.enableIRIn(); // Enabling IR sensor

void loop()
{
ir_remote(); //IR remote Control
}
```

### Debug Message

```plain
Ir_receiver has been enabled please use decoded value without any parameter
```

### Other Steps to Reproduce

I also tried this code in my friend windows but It didn't work

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-11-10:
IRremote.h is not one of the libraries in this repository.   According to a web search it is an Arduino library.  Top result:
https://gist.github.com/reefwing/f87764523ace1c383ab002759f6369d4#:~:text=The%20standard%20IRRemote.h%20Arduino%20library%20only%20supports%20receiving,library%20and%20add%20it%20to%20your%20IDE%3A%20https%3A%2F%2Fgithub.com%2FSensorsIot%2FDefinitive-Guide-to-IR%2Ftree%2Fmaster%2FESP32-IRremote

I'm not sure how your library works but I'll point out a few delays:
`'delay(100)` on most (but not all!) of your switch cases.  May cause it to miss events.
`pref.putBool("Relay3", toggleState_3);` If you are saving settings to NVS, be aware this probably writes to flash immediately, causing a small delay.
`my_switch4.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_4);' Is this a URL request to cloud?  Variable delay here 50ms to 90000ms.
`irrecv.resume()` Was there a pause somewhere?

It's probably best to set flags or put events into a que and then process those events where they won't block.

mrengineer7777, 2022-11-10:
Try
```
if (irrecv.decode(&results)) {
   irrecv.resume();
   //Serial.println(results.value, HEX);
}
switch(results.value) {
   case IR_Button_1:
   ....
}
```

SuGlider, 2022-11-10:
Is this the issue? 

>But a new problem starts in verbose mode and that is in Esp Rainmaker I am linked with Esp Rainmaker and It is unable to Print a QR code.

Or is it related to the IR receiver?

suryanshRoy, 2022-11-11:
@SuGlider It is related to Ir receiver.

@mrengineer7777 when I was not linked with ESP RAINMAKER it is working but I need to press Ir Remote button about 5 to 10 times  and same thing happens when I am linked with Esp Rainmaker and also happens when I arranged them in que.   :⁠^⁠)

SuGlider, 2022-11-11:
@suryanshRoy 
I think that you may consider changing the IR Library to one that uses RMT. 
One example is https://github.com/Darryl-Scott/ESP32-RMT-Library-IR-code-RAW
Not sure how easy that would be to adapt it to your code.
But using Hardware Peripherals from ESP32 would be more efficient than using something based on Timers or delay(), such as the current library used in your code.

EPS32 Arduino has many tasks running together and it may cause delays that prevent it from reading IR correctly.

suryanshRoy, 2022-11-13:
I don't know how to use that type of library @SuGlider  so In the old library It prints something like this :-

The function decode(&results)) is deprecated and may not work as expected! Just use decoded ( ) without a parameter and IrReceiver.decodedIRData.<fieldname>  .

SuGlider, 2022-11-13:
It may be necessary to read the IR Library code in order to adjust the application correctly.
I'm not sure about what may be going wrong with your application and RainMaker... You may need to debug it further and try to figure it out.

Anyway, I don't see, so far, any Arduino Core issue here.

suryanshRoy, 2022-11-14:
I have used Tsop1838 and It is fine it is working.
When I change the library in the code IRremote.h  3.9.0 version to IRremoteESP8266 2.8.4 version by testing I find that new library is also not working So simply it means that there is something problem in my code. 
So I will try to fix it soon and I am closing the Issue. Thanks

closed 2022-11-14 by suryanshRoy