espressif/arduino-esp32 #7454: esp32s2 ble support

BennyShani, 2022-11-10:
### Board

ESP32-S2

### Device Description

Can not compile the BLE-Server example.
It works for ESP32-S3 but for the S2 the toolchain is broken for BLE

### Hardware Configuration

NO

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

Not relevent

### Description

Open Arduino IDE
Load ESP32 BLE_Server example
Select board : ESP32S2 Dev Module
Compile: Fail
Select board : ESP32S3 Dev Module
Compile: OK

### Sketch

```cpp
/*
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleServer.cpp
    Ported to Arduino ESP32 by Evandro Copercini
    updates by chegewara
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("Long name works now");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic->setValue("Hello World says Neil");
  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}
```

### Debug Message

```plain
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino: In function 'void setup()':
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:21:3: error: 'BLEDevice' has not been declared
   BLEDevice::init("Long name works now");
   ^~~~~~~~~
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:22:3: error: 'BLEServer' was not declared in this scope
   BLEServer *pServer = BLEDevice::createServer();
   ^~~~~~~~~
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:22:3: note: suggested alternative: 'Server'
   BLEServer *pServer = BLEDevice::createServer();
   ^~~~~~~~~
   Server
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:22:14: error: 'pServer' was not declared in this scope
   BLEServer *pServer = BLEDevice::createServer();
              ^~~~~~~
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:22:14: note: suggested alternative: 'Server'
   BLEServer *pServer = BLEDevice::createServer();
              ^~~~~~~
              Server
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:22:24: error: 'BLEDevice' has not been declared
   BLEServer *pServer = BLEDevice::createServer();
                        ^~~~~~~~~
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:23:3: error: 'BLEService' was not declared in this scope
   BLEService *pService = pServer->createService(SERVICE_UUID);
   ^~~~~~~~~~
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:23:15: error: 'pService' was not declared in this scope
   BLEService *pService = pServer->createService(SERVICE_UUID);
               ^~~~~~~~
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:23:15: note: suggested alternative: 'Serial'
   BLEService *pService = pServer->createService(SERVICE_UUID);
               ^~~~~~~~
               Serial
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:24:3: error: 'BLECharacteristic' was not declared in this scope
   BLECharacteristic *pCharacteristic = pService->createCharacteristic(
   ^~~~~~~~~~~~~~~~~
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:24:3: note: suggested alternative: 'Character_h'
   BLECharacteristic *pCharacteristic = pService->createCharacteristic(
   ^~~~~~~~~~~~~~~~~
   Character_h
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:24:22: error: 'pCharacteristic' was not declared in this scope
   BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                      ^~~~~~~~~~~~~~~
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:24:22: note: suggested alternative: 'Character_h'
   BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                      ^~~~~~~~~~~~~~~
                      Character_h
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:26:42: error: 'BLECharacteristic' is not a class, namespace, or enumeration
                                          BLECharacteristic::PROPERTY_READ |
                                          ^~~~~~~~~~~~~~~~~
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:27:42: error: 'BLECharacteristic' is not a class, namespace, or enumeration
                                          BLECharacteristic::PROPERTY_WRITE
                                          ^~~~~~~~~~~~~~~~~
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:33:3: error: 'BLEAdvertising' was not declared in this scope
   BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
   ^~~~~~~~~~~~~~
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:33:19: error: 'pAdvertising' was not declared in this scope
   BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
                   ^~~~~~~~~~~~
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:33:34: error: 'BLEDevice' has not been declared
   BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
                                  ^~~~~~~~~
C:\Users\benny\AppData\Local\Temp\.arduinoIDE-unsaved20221010-13068-1dx5tgm.zkjk\BLE_server\BLE_server.ino:38:3: error: 'BLEDevice' has not been declared
   BLEDevice::startAdvertising();
   ^~~~~~~~~

exit status 1

Compilation error: 'BLEDevice' has not been declared
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-11-10:
S2 has no BLE

closed 2022-11-10 by SuGlider

BennyShani, 2022-11-10:
Thank you,

I mistake esp32 for esp32s2

Sorry

Benny