espressif/arduino-esp32 #7457: USB HID header file for ESP32-S3

PeterHindes, 2022-11-10:
### Related area

USB HID header file for usb keyboard and mouse examples included with arduino ide.

### Hardware specification

ESP32-S3 (and S2)

### Is your feature request related to a problem?

[image]
[image]
This file "HID.h" should be supplied by the board library.

### Describe the solution you'd like

Please add support for this stock example using the keyboard library by adding an appropriate HID.h file to support usb hid keyboard (and other usb hid use cases).

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2022-11-10:
Our HID library is not 100% compatible with Arduino's HID. We do provide our own examples here: https://github.com/espressif/arduino-esp32/tree/master/libraries/USB/examples

closed 2022-11-10 by me-no-dev

PeterHindes, 2022-11-10:
Thank you. I seem to have missed that example in my haste.
For anyone who stumbles across this the main difference is using this instead of importing keyboard.h 
```C++
#include "USB.h"
#include "USBHIDKeyboard.h"
USBHIDKeyboard Keyboard;
```
You can open the example if you have your esp environment setup under the esp32 specific examples for usb.
[image]