espressif/arduino-esp32 #7458: Starting carrier wae and other RF tests from ESP32 core since the RF test firmware doesnt work in CDC on the C3

0x0fe, 2022-11-10:
### Board

ESP32-C3 dev module

### Device Description

ESP32C3 board

### Hardware Configuration

nothing but UART0 not aailable

### Version

v2.0.5

### IDE Name

arduino IDE

### Operating System

windows 10

### Flash frequency

80M

### PSRAM enabled

no

### Upload speed

CDC

### Description

The RF test frmware does not work with the ESP32C3 n CDC mode, we can connect and send commands howeer there is only empty reples from the ESP32C3. The problem is that in some case the UART is not accessible, only USB is accessible.
So i would ike to know wether there is a way to use the RF test function from the HAL. IF there is no way (at least setting the carrier wae) then is there a CDC ersion of the RF test firmware for C3?

### Sketch

```cpp
none
```

### Debug Message

```plain
none
```

### Other Steps to Reproduce

just use the manual RF test tool wth an ESP32C3 a the USB port (CDC) we can connect to it, flash the RF test firmware, send commands, but we only get empty replies, and the C3 does not go into test mode, for example no carrier is present when we enable it.
[image]

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

0x0fe, 2022-11-10:
so, there is no access to these function at all? how do we do with the C3 when there is no UART then?

me-no-dev, 2022-11-10:
I really can not answer this question :) RF Testing is outside of what we do and we do not have access to the code and modules required. If you need it, you probably are a customer of Espressif that needs to certify a product or similar. In that case you can contact Espressif directly and get help with the tool. It is possible that such firmware already exists

0x0fe, 2022-11-12:
Thank you for your reply, yes, we will see with the Shenzhen office.