espressif/arduino-esp32 #7459: Let the WiFiClientSecure set an hostname different from the ip address

lu-zero, 2022-11-10:
### Related area

WiFiClientSecure

### Hardware specification

Any

### Is your feature request related to a problem?

The function

```
int start_ssl_client(sslclient_context *ssl_client, const char *host, uint32_t port, int timeout, const char *rootCABuff, bool useRootCABundle, const char *cli_cert, const char *cli_key, const char *pskIdent, const char *psKey, bool insecure, const char **alpn_protos
```

uses `host` to resolve the ip and use the it as expected CN. In some cases the CN information is provided by other means 

### Describe the solution you'd like

Provide a mean to set the CN when it is different.

### Additional context

https://github.com/espressif/arduino-esp32/issues/7350 is tangentially related.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

cziter15, 2022-12-08:
There is a room for refactoring in context of SSL and WiFiClientSecure.
When you look at WiFiClientSecure, you'll see that it simply converts IPAddress to string, passing it to start_ssl_client function. 

Why? I suspect that the idea was to have a proper hostname for certificate validation (parameter of mbedtls_ssl_set_hostname). The solution is simple - introduce another optional (NULL by default) sslHostname parameter and replace current hostname parameter with IPAddress.

```
int WiFiClientSecure::connect(IPAddress ip, uint16_t port, const char *CA_cert, const char *cert, const char *private_key)
{
    return connect(ip.toString().c_str(), port, CA_cert, cert, private_key);
}

int WiFiClientSecure::connect(const char *host, uint16_t port, const char *CA_cert, const char *cert, const char *private_key)
{
    int ret = start_ssl_client(sslclient, host, port, _timeout, CA_cert, _use_ca_bundle, cert, private_key, NULL, NULL, _use_insecure, _alpn_protos);
    _lastError = ret;
    if (ret < 0) {
        log_e("start_ssl_client: %d", ret);
        stop();
        return 0;
    }
    _connected = true;
    return 1;
}
```

cziter15, 2022-12-31:
I've issued PR that aims to resolve this issue.

- Added another connect variant in WiFiClientSecure that can take both IP and hostname from the user.
- All other connect methods ('higher level') will call that underlying function. Hostname is now translated to IP in WiFiClientSecure, not in ssl_client code.

closed 2025-01-16 by Parsaabasi