espressif/arduino-esp32 #7461: typo in camera_index.h OV2460 -> OV2640

beta-tester, 2022-11-10:
### Board

n/a

### Device Description

n/a

### Hardware Configuration

n/a

### Version

v2.0.5

### IDE Name

Arduino

### Operating System

n/a

### Flash frequency

n/a

### PSRAM enabled

yes

### Upload speed

n/a

### Description

the title of the Camera web site is wrong for the OV2640 sensor
[camera_index.h](https://github.com/espressif/arduino-esp32/blob/bcc1d758fc343887de03affeedfbb33522ff2523/libraries/ESP32/examples/Camera/CameraWebServer/camera_index.h#L5)

it extracts to:
```
<!doctype html>
<html>
    <head>
        <meta charset="utf-8">
        <meta name="viewport" content="width=device-width,initial-scale=1">
        <title>ESP32 OV2460</title>
...
```

but it should be:
```
<!doctype html>
<html>
    <head>
        <meta charset="utf-8">
        <meta name="viewport" content="width=device-width,initial-scale=1">
        <title>ESP32 OV2640</title>
...
```

### Sketch

```cpp
n/a
```

### Debug Message

```plain
n/a
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-11-10:
Thanks for reporting!
@me-no-dev can fix it for sure!

me-no-dev, 2022-11-11:
yup, will do. Working on updated API of the camera driver itself and once that is done, I will update the HTML as part of the changes necessary here.

@SuGlider I would not call this a "bug" though :) it's just string typo

SuGlider, 2022-11-11:
>@SuGlider I would not call this a "bug" though :) it's just string typo

I couldn't find a softer tag in the list :-) - I agree it is just a typo.

beta-tester, 2023-05-29:
any progress here?

Parsaabasi, 2025-01-09:
Hello, 

I close this since this report contains the release we no longer support. Please try the new versions and in case the issue persists, feel free to reopen it. 

Thanks

closed 2025-01-09 by Parsaabasi

beta-tester, 2025-01-10:
but the current [master of camera_index.h](https://github.com/espressif/arduino-esp32/blob/a81e2d48f57c0a4c0b90c4a06596904d0905805a/libraries/ESP32/examples/Camera/CameraWebServer/camera_index.h#L4) still shows this wrong title. so the typo still exist.

to verify,
1. copy the content (comma separated hex-values) between the curled brackets of
    ```C
    const unsigned char index_ov2640_html_gz[] = {
      ...
    };
    ```
2. and paste it to the input field of the [CyberChef page](https://gchq.github.io/CyberChef/#recipe=From_Hex('Auto')Gunzip())

```html
<!doctype html>
<html>
    <head>
        <meta charset="utf-8">
        <meta name="viewport" content="width=device-width,initial-scale=1">
        <title>ESP32 OV2460</title>
```

beta-tester, 2025-01-12:
the issue is not resolved nor done yet.
how can i re-open this issue?