espressif/arduino-esp32 #7462: Can't dynamically allocate memory even though heap shows enough is free

av4625, 2022-11-10:
### Board

ESP32 Dev Module

### Device Description

DevKitC

### Hardware Configuration

NA

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

macOS

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

```C++
ESP.getFreeHeap();
ESP.getMaxAllocHeap();
```

Output:
```
161484
100340
```

I have a fairly large system that dynamically allocates a lot of things up front using smart pointers. 
When setting things up 2 large vectors are reserved up front. If `vec` reserves 3000 and `vec1` reserves 2500, I get the output above. If I raise `vec1` to reserve 2750 I get a constant boot loop. With `|<-CORRUPTED` beside the backtrace. From the output above it looks like I have plenty of space to be able to reserve those vectors upfront.

### Sketch

```cpp
++
std::vector<std::pair<int32_t, double> > vec;
vec.reserve(3000);
std::vector<std::pair<int32_t, double> > vec1;
vec1.reserve(2750);
```

### Debug Message

```plain
0x40083a61: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 402
0x4008daad: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c line 128
0x40093025: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c line 46
0x4016ccb7: __cxa_call_terminate(_Unwind_Exception*) at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_call.cc line 58
0x4016ccfe: std::get_terminate() at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc line 97
0x4015e69b: __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc line 90
0x4015e1ae: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/no-rtti/libstdc++-v3/include/bits/exception.h line 63
0x400d5a79: __gnu_cxx::new_allocator   >::allocate(unsigned int, void const*) at /Users/avance/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/xtensa-esp32-elf/include/c++/8.4.0/ext/new_allocator.h line 99
0x400d5c17: std::vector  , std::allocator   > >::reserve(unsigned int) at /Users/avance/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_vector.h line 1395
0x400d5db5: component::timing::context_impl::context_impl(std::shared_ptr  const&, std::shared_ptr  const&, std::shared_ptr  const&, std::shared_ptr  const&, std::shared_ptr  const&, std::shared_ptr  const&, std::shared_ptr  const&) at /Users/avance/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/xtensa-esp32-elf/include/c++/8.4.0/bits/shared_ptr_base.h line 1018
0x400d678d: component::timing::timing::create(std::shared_ptr  const&, std::shared_ptr  const&, std::shared_ptr  const&, std::shared_ptr  const&, std::shared_ptr  const&, std::shared_ptr  const&) at /Users/avance/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/xtensa-esp32-elf/include/c++/8.4.0/bits/shared_ptr_base.h line 1182
0x400d3c1a: _GLOBAL__sub_I_status_led() at /Users/avance/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/xtensa-esp32-elf/include/c++/8.4.0/bits/shared_ptr_base.h line 1167
0x400f719f: start_cpu0_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c line 422
0x40083162: call_start_cpu0 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32/include/hal/cpu_ll.h line 39
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-11-11:
Are you allocating in the global? Have you tried to do it in `setup()`?

av4625, 2022-11-11:
Yea I am, just thought this could be an issue after I made this last night. Will try moving it later today to see if it helps

mrengineer7777, 2022-11-11:
FYI getMaxAllocHeap() calls heap_caps_get_largest_free_block.
```
uint32_t EspClass::getMaxAllocHeap(void)
{
    return heap_caps_get_largest_free_block(MALLOC_CAP_INTERNAL);
}
```

av4625, 2022-11-11:
I moved all the allocation to setup and it has fixed the issue! 

I had a separate branch where I was moving all startup initialisation to a separate class and out of the global section, I just hadn't got round to the new change of reserving the vectors! Darn!

closed 2022-11-14 by VojtechBartoska