espressif/arduino-esp32 #7464: GPIO33 and GPIO34 not working as input

rubenfrolic, 2022-11-11:
### Board

ESP32-S3FN8

### Device Description

ESP32-S3FN8 custom board

### Hardware Configuration

ESP32-S3FN8 with digital inputs on GPIO33 and 34

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

macOS

### Flash frequency

40mHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I'm working on a custom designed pcb equipped with an ESP32-S3FN8 SoC. There are 2 digital inputs connected to GPIO33 and GPIO34 but they seem to be unresponsive. There are no problems with other GPIO's. 

The simple attached code doesn't work.

Is there a specific way to define these pins since they have multiple functions/configurations?

### Sketch

```cpp
int gpioPin = 33;

void setup() {

  Serial.begin(115200);
  pinMode(gpioPin, INPUT);
}

void loop() {

  int gpioState = digitalRead(gpioPin);
  Serial.println(gpioState);
  delay(1000);
}
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-11-11:
I am not sure of the reason. We do not impose any such limitation in Arduino (not sure about is ESP-IDF does so), but I notice that 33 and 34 are not exposed on any of Espressif's modules or dev-boards

SuGlider, 2022-11-11:
There is some information about it in https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/gpio.html

```
SPI0/1: GPIO26-32 are usually used for SPI flash and PSRAM and not recommended for other uses.
When using Octal Flash or Octal PSRAM or both, GPIO33~37 are connected to SPIIO4 ~ SPIIO7 and SPIDQS.
Therefore, on boards embedded with ESP32-S3R8 / ESP32-S3R8V chip, GPIO33~37 are also not recommended for other uses.
```

But for the ESP32-S3FN8 it seems that it only has QIO Flash and GPIO33/34 would be free for using.
https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf
Chapter 1,2 says:

[image]

SuGlider, 2022-11-11:
I have a ESP32-S3-DevKitM-1 with a MINI-1 (S3FN8) in my desk.
I'll test it and report here.

SuGlider, 2022-11-11:
@rubenfrolic - Itested it using a ESP32-S3-DevKitM-1 with a MINI-1 (S3FN8). It exposes both pins 33 and 34.

I can't reproduce the issue. 
It reads GPIO33 and GPIO34 correctly.

SuGlider, 2022-11-11:
You can check the eFuses to see if the SPI default pins are pre-configured.

```
80      FLASH_TYPE                      EFUSE_BLK0        137              1
...
92      SPI_PAD_CONFIG_CLK              EFUSE_BLK1         48              6
93      SPI_PAD_CONFIG_Q_D1             EFUSE_BLK1         54              6
94      SPI_PAD_CONFIG_D_D0             EFUSE_BLK1         60              6
95      SPI_PAD_CONFIG_CS               EFUSE_BLK1         66              6
96      SPI_PAD_CONFIG_HD_D3            EFUSE_BLK1         72              6
97      SPI_PAD_CONFIG_WP_D2            EFUSE_BLK1         78              6
98      SPI_PAD_CONFIG_DQS              EFUSE_BLK1         84              6
99      SPI_PAD_CONFIG_D4               EFUSE_BLK1         90              6
100     SPI_PAD_CONFIG_D5               EFUSE_BLK1         96              6
101     SPI_PAD_CONFIG_D6               EFUSE_BLK1        102              6
102     SPI_PAD_CONFIG_D7               EFUSE_BLK1        108              6
```

Using `espefuse.py --port com15 summary`, in my Win11, the S3 is connected to COM15, I get this (among other info):

```
FLASH_TYPE (BLOCK0)                                Selects SPI flash type                             = 4 data lines R/W (0b0)
...
Spi_Pad_Config fuses:
SPI_PAD_CONFIG_CLK (BLOCK1)                        SPI CLK pad                                        = 0 R/W (0b000000)
SPI_PAD_CONFIG_Q (BLOCK1)                          SPI Q (D1) pad                                     = 0 R/W (0b000000)
SPI_PAD_CONFIG_D (BLOCK1)                          SPI D (D0) pad                                     = 0 R/W (0b000000)
SPI_PAD_CONFIG_CS (BLOCK1)                         SPI CS pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_HD (BLOCK1)                         SPI HD (D3) pad                                    = 0 R/W (0b000000)
SPI_PAD_CONFIG_WP (BLOCK1)                         SPI WP (D2) pad                                    = 0 R/W (0b000000)
SPI_PAD_CONFIG_DQS (BLOCK1)                        SPI DQS pad                                        = 0 R/W (0b000000)
SPI_PAD_CONFIG_D4 (BLOCK1)                         SPI D4 pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D5 (BLOCK1)                         SPI D5 pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D6 (BLOCK1)                         SPI D6 pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D7 (BLOCK1)                         SPI D7 pad                                         = 0 R/W (0b000000)
```

If I try the same `espsfuse.py` command on a ESP32-S3-WROOM that has OCTOSPI, where GPIO33 and 34 are not available, I get this information:
```
FLASH_TYPE (BLOCK0)                                Selects SPI flash type                             = 8 data lines R/W (0b1)
...
Spi_Pad_Config fuses:
SPI_PAD_CONFIG_CLK (BLOCK1)                        SPI CLK pad                                        = 0 R/W (0b000000)
SPI_PAD_CONFIG_Q (BLOCK1)                          SPI Q (D1) pad                                     = 0 R/W (0b000000)
SPI_PAD_CONFIG_D (BLOCK1)                          SPI D (D0) pad                                     = 0 R/W (0b000000)
SPI_PAD_CONFIG_CS (BLOCK1)                         SPI CS pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_HD (BLOCK1)                         SPI HD (D3) pad                                    = 0 R/W (0b000000)
SPI_PAD_CONFIG_WP (BLOCK1)                         SPI WP (D2) pad                                    = 0 R/W (0b000000)
SPI_PAD_CONFIG_DQS (BLOCK1)                        SPI DQS pad                                        = 0 R/W (0b000000)
SPI_PAD_CONFIG_D4 (BLOCK1)                         SPI D4 pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D5 (BLOCK1)                         SPI D5 pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D6 (BLOCK1)                         SPI D6 pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D7 (BLOCK1)                         SPI D7 pad                                         = 0 R/W (0b000000)

````

SuGlider, 2022-11-11:
Another question is if the IDE is correctly configured using one of those 3 options: DIO 80MHz / QIO 80MHz / QIO 120MHz

rubenfrolic, 2022-11-14:
I've been trying quite some thing, but still didn't get it to work. using espefuse.py gives me the following. But I can't find any cues in there either (also fairly new to me): 

```=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
TEMP_SENSOR_CAL (BLOCK2)                           ??? Temperature calibration                        = -3.5 R/W (0b100100011)
ADC1_MODE0_D2 (BLOCK2)                             ??? ADC1 calibration 1                             = 40 R/W (0x0a)
ADC1_MODE1_D2 (BLOCK2)                             ??? ADC1 calibration 2                             = 72 R/W (0x12)
ADC1_MODE2_D2 (BLOCK2)                             ??? ADC1 calibration 3                             = -272 R/W (0xc4)
ADC1_MODE3_D2 (BLOCK2)                             ??? ADC1 calibration 4                             = 184 R/W (0x2e)
ADC2_MODE0_D2 (BLOCK2)                             ??? ADC2 calibration 5                             = 360 R/W (0x5a)
ADC2_MODE1_D2 (BLOCK2)                             ??? ADC2 calibration 6                             = -452 R/W (0xf1)
ADC2_MODE2_D2 (BLOCK2)                             ??? ADC2 calibration 7                             = -292 R/W (0xc9)
ADC2_MODE3_D2 (BLOCK2)                             ??? ADC2 calibration 8                             = -448 R/W (0xf0)
ADC1_MODE0_D1 (BLOCK2)                             ??? ADC1 calibration 9                             = 16 R/W (0b000100)
ADC1_MODE1_D1 (BLOCK2)                             ??? ADC1 calibration 10                            = 80 R/W (0b010100)
ADC1_MODE2_D1 (BLOCK2)                             ??? ADC1 calibration 11                            = -52 R/W (0b101101)
ADC1_MODE3_D1 (BLOCK2)                             ??? ADC1 calibration 12                            = 12 R/W (0b000011)
ADC2_MODE0_D1 (BLOCK2)                             ??? ADC2 calibration 13                            = -60 R/W (0b101111)
ADC2_MODE1_D1 (BLOCK2)                             ??? ADC2 calibration 14                            = 76 R/W (0b010011)
ADC2_MODE2_D1 (BLOCK2)                             ??? ADC2 calibration 15                            = 32 R/W (0b001000)
ADC2_MODE3_D1 (BLOCK2)                             ??? ADC2 calibration 16                            = 64 R/W (0b010000)

Config fuses:
DIS_ICACHE (BLOCK0)                                Disables ICache                                    = False R/W (0b0)
DIS_DCACHE (BLOCK0)                                Disables DCache                                    = False R/W (0b0)
DIS_DOWNLOAD_ICACHE (BLOCK0)                       Disables Icache when SoC is in Download mode       = False R/W (0b0)
DIS_DOWNLOAD_DCACHE (BLOCK0)                       Disables Dcache when SoC is in Download mode       = False R/W (0b0)
DIS_FORCE_DOWNLOAD (BLOCK0)                        Disables forcing chip into Download mode           = False R/W (0b0)
DIS_CAN (BLOCK0)                                   Disables the TWAI Controller hardware              = False R/W (0b0)
DIS_APP_CPU (BLOCK0)                               Disables APP CPU                                   = False R/W (0b0)
FLASH_TPUW (BLOCK0)                                Configures flash startup delay after SoC power-up, = 0 R/W (0x0)
                                                    unit is (ms/2). When the value is 15, delay is 7.
                                                   5 ms                                              
DIS_DIRECT_BOOT (BLOCK0)                           Disables direct boot mode                          = False R/W (0b0)
DIS_USB_SERIAL_JTAG_ROM_PRINT (BLOCK0)             Disables USB-Serial-JTAG ROM printing              = False R/W (0b0)
FLASH_ECC_MODE (BLOCK0)                            Configures the ECC mode for SPI flash             
   = 16-byte to 18-byte mode R/W (0b0)
DIS_USB_SERIAL_JTAG_DOWNLOAD_MODE (BLOCK0)         Disables USB-Serial-JTAG download feature in UART  = False R/W (0b0)
                                                   download boot mode                                
UART_PRINT_CONTROL (BLOCK0)                        Sets the default UART boot message output mode     = Enabled R/W (0b00)
FLASH_TYPE (BLOCK0)                                Selects SPI flash type                             = 4 data lines R/W (0b0)
FLASH_PAGE_SIZE (BLOCK0)                           Sets the size of flash page                        = 0 R/W (0b00)
FLASH_ECC_EN (BLOCK0)                              Enables ECC in Flash boot mode                     = False R/W (0b0)
FORCE_SEND_RESUME (BLOCK0)                         Forces ROM code to send an SPI flash resume comman = False R/W (0b0)
                                                   d during SPI boot                                 
DIS_USB_OTG_DOWNLOAD_MODE (BLOCK0)                 Disables USB-OTG download feature in UART download = False R/W (0b0)
                                                    boot mode                                        
DISABLE_WAFER_VERSION_MAJOR (BLOCK0)               Disables check of wafer version major              = False R/W (0b0)
DISABLE_BLK_VERSION_MAJOR (BLOCK0)                 Disables check of blk version major                = False R/W (0b0)
[... cut ...]
VDD_SPI_FORCE (BLOCK0)                             Force using VDD_SPI_XPD and VDD_SPI_TIEH to config = True R/W (0b1)
                                                   ure VDD_SPI LDO                                   
PIN_POWER_SELECTION (BLOCK0)                       Sets default power supply for GPIO33..37           = VDD3P3_CPU R/W (0b0)

Wdt Config fuses:
WDT_DELAY_SEL (BLOCK0)                             Selects RTC WDT timeout threshold at startup       = 0 R/W (0b00)

Flash voltage (VDD_SPI) set to 3.3V by efuse.
```

SuGlider, 2022-11-14:
The eFuses are OK. The same I have here for my ESP32-S3-Mini. 
Your eFuses reports that it uses 4 wires for Flash, therefore, IO33 and IO34 shall be free to be used.

A possible test is to verify if the board is fine.
For that, I'd suggest to set those pins as OUTPUT and blink them in a Sketch. This way you will verify if the pads are connected to the S3 module and the PCB/Soldering has no failure.

rubenfrolic, 2022-11-15:
I think we found the problem and indeed it seemed the problem was elsewhere in the hardware. The IC that was producing the signals turned out not to be strong enough be able to fully pull up the signal to 3.3v. The signal kept floating around 1.2v and thus unable to reach the HIGH-LOW threshold. It took a bit of time to verify this since we couldn't to an analog read on GPIO33/34 and it was physically very challenging to actually reach those signal lines for measuring. Having figured out this made the fix very easy by using input_pullup instead of a regular input. 

Thanks for the help though!

closed 2022-11-16 by VojtechBartoska

ProfFan, 2023-10-24:
I think this issue has not been resolved. When choosing QSPI GPIO34 access immediately triggers a WDG failure in ESP32-S3-MINI-1

ProfFan, 2023-10-24:
GPIO34 is SPIIO4 which should be free when not using Octal SPI

me-no-dev, 2023-10-25:
Actually GPIOs 35 to 37 are free when Octal is not used. 34 is used.

0x0fe, 2024-06-08:
> Actually GPIOs 35 to 37 are free when Octal is not used. 34 is used.

```
GPIO26-32 are usually used for SPI flash and PSRAM and not recommended for other uses.
When using Octal Flash or Octal PSRAM or both, GPIO33~37 are connected... 
```

Can you clarify that? the S8FN8 is quad, how comes GPIO34 is used?

0x0fe, 2024-06-08:
[image]