espressif/arduino-esp32 #7465: Wrover-kit SPI not working SD0-3 CLK

Pilowsite, 2022-11-11:
### Board

ESP-WROVER-KIT V4.1

### Device Description

wrover-kit connected to LoRa Module SX1262

### Hardware Configuration

SD0 connected to MISO of sx1262
SD1 connected to MOSI of sx1262
GPIO5 connected to NSS of SX1262
CLK connected to CLK of sx1262
GPIO27 connected to RST of sx1262
GPIO36 connected to BUSY of sx1262
GPIO26 connected to DIO1 of sx1262

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Ubuntu 22.04

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I'm currently trying to use ESP-WROVER-KIT and interface it with a LoRa SX1262 by using the RadioLib library.

In the description of the board, they say I can use CLK,SD0,SD1,SD2,SD3 for SPI and choose any free GPIO for chip select.

I'm not sure who is MISO, MOSI out of all these SD0-3 pins. I also tried to run SX126x_Transmit example. I get the error that no device was found (error -2)

Also, I tried using my Logic Analyzer Logic 8 Saleae and even the SCK pin is flat..

### Sketch

```cpp
/*
   RadioLib SX126x Transmit with Interrupts Example

   This example transmits LoRa packets with one second delays
   between them. Each packet contains up to 256 bytes
   of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)

   Other modules from SX126x family can also be used.

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx126x---lora-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <SPI.h>
#include <RadioLib.h>

// SX1262 has the following connections:
// NSS pin:   19
// DIO1 pin:  26
// NRST pin:  27
// BUSY pin:  36
SX1262 radio = new Module(5, 26, 27, 36);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1262 radio = RadioShield.ModuleA;

// or using CubeCell
//SX1262 radio = new Module(RADIOLIB_BUILTIN_MODULE);

// save transmission state between loops
int transmissionState = RADIOLIB_ERR_NONE;

void setup() {
  Serial.begin(9600);

  // initialize SX1262 with default settings
  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio.begin();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // set the function that will be called
  // when packet transmission is finished
  radio.setDio1Action(setFlag);

  // start transmitting the first packet
  Serial.print(F("[SX1262] Sending first packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  transmissionState = radio.startTransmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67,
                      0x89, 0xAB, 0xCD, 0xEF};
    state = radio.startTransmit(byteArr, 8);
  */
}

// flag to indicate that a packet was sent
volatile bool transmittedFlag = false;

// disable interrupt when it's not needed
volatile bool enableInterrupt = true;

// this function is called when a complete packet
// is transmitted by the module
// IMPORTANT: this function MUST be 'void' type
//            and MUST NOT have any arguments!
#if defined(ESP8266) || defined(ESP32)
  ICACHE_RAM_ATTR
#endif
void setFlag(void) {
  // check if the interrupt is enabled
  if(!enableInterrupt) {
    return;
  }

  // we sent a packet, set the flag
  transmittedFlag = true;
}

void loop() {
  // check if the previous transmission finished
  if(transmittedFlag) {
    // disable the interrupt service routine while
    // processing the data
    enableInterrupt = false;

    // reset flag
    transmittedFlag = false;

    if (transmissionState == RADIOLIB_ERR_NONE) {
      // packet was successfully sent
      Serial.println(F("transmission finished!"));

      // NOTE: when using interrupt-driven transmit method,
      //       it is not possible to automatically measure
      //       transmission data rate using getDataRate()

    } else {
      Serial.print(F("failed, code "));
      Serial.println(transmissionState);

    }

    // clean up after transmission is finished
    // this will ensure transmitter is disabled,
    // RF switch is powered down etc.
    radio.finishTransmit();

    // wait a second before transmitting again
    delay(1000);

    // send another one
    Serial.print(F("[SX1262] Sending another packet ... "));

    // you can transmit C-string or Arduino string up to
    // 256 characters long
    transmissionState = radio.startTransmit("Hello World!");

    // you can also transmit byte array up to 256 bytes long
    /*
      byte byteArr[] = {0x01, 0x23, 0x45, 0x67,
                        0x89, 0xAB, 0xCD, 0xEF};
      int state = radio.startTransmit(byteArr, 8);
    */

    // we're ready to send more packets,
    // enable interrupt service routine
    enableInterrupt = true;
  }
}
```

### Debug Message

```plain
[SX1262] Initializing ... failed, code -2
```

### Other Steps to Reproduce

n/a

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-11-11:
> In the description of the board, they say I can use CLK,SD0,SD1,SD2,SD3 for SPI and choose any free GPIO for chip select.

Can you link us to that description?

You can not use those pins for SPI. You need to use any of the other pins. Default GPIOs for SPI are found here: 
[image]

Pilowsite, 2022-11-11:
Oh I did not see this diagram interesting! 

Here's the link: 
[image]

And then bellow is mentioned that: 

[image]

Here's what I got when using the mentioned pins:
[image]

It seems better but I see that the NSS pin is not working the right way for example.

I was wondering also, could I use any other free GPIO other than GPIO5 right? Because GPIO5 is used to interface with the LCD (backlight). Or should I use a dedicated alternate CS?

SuGlider, 2022-11-11:
You can use whatever SPI pins you may need. 
For that just add this `  SPI.begin(int8_t sck, int8_t miso, int8_t mosi, int8_t ss);` before `radio.begin();`

SS is the CS pin. Yes, you can change it to any other PIN.
There is a method for using CS from the SPI Peripheral (automaticaly when there is a transmission) or controlling it manually (which I think is the case for this Library). By default it is manually (false).
`void SPIClass::setHwCs(bool use)`

Pilowsite, 2022-11-11:
Update: I realized that I flipped the logic analyzer connector. I still have issue but now I know that SPI is working. I'm using the following pins as me-no-dev said: 
MISO: GPIO19
MOSI: GPIO23
SCK: GPIO18
NSS: GPIO5

SuGlider, 2022-11-11:
What is the issue? I understood that SPI is working fine.

Pilowsite, 2022-11-16:
Yes it is working fine but the documentation is inaccurate I think. They mention pins SD0 SD1 SD2 SD3 CLK being SPI peripherals that we can use when in fact they are not. It should be mentioned in the ESP-WROVER-KIT DOC that it's the other Pins. 
For VSPI: 
MISO: GPIO19
MOSI: GPIO23
SCK: GPIO18
NSS: GPIO5

and maybe mention where are the HSPI pins exactly.
That would make it easier to find.

I'm referring to this page: 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-wrover-kit.html

and this picture
[image]

SuGlider, 2022-11-17:
Are you talking about this link?
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-wrover-kit.html#spi-flash-jp2

It is just saying that the **SPI FLASH** JP2 exposes these pins.

If your application is running code from IRAM, it could, in theory, use those pins for something else... but this is really more advanced usage.

By other hand, if the ESP32 is set to use DIO FLASH mode, the pins 9 and 10 (SD2 and SD3) will be always free for any other purpose.

NezhaZahri, 2022-12-12:
It could be interesting to maybe mention that it is more advanced usage? And if possible mention the readily usable SPI pins in the doc which are : 
MOSI : GPIO23
MISO : GPIO19

SuGlider, 2022-12-12:
A PR https://github.com/espressif/esp-idf/pull/10356/files is pushed in order to make the document clearer regarding not using JP2 Flash SPI pins.

SuGlider, 2022-12-12:
https://github.com/espressif/esp-idf/pull/10356

SuGlider, 2022-12-14:
Merged changes to the documentation in 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-wrover-kit.html#spi-flash-jp2

[image]

closed 2022-12-14 by SuGlider