espressif/arduino-esp32 #7467: The 16MB partition tables fail to upload

CoryCharlton, 2022-11-12:
### Board

Unexpected Maker FeatherS3 (but likely any board with a 16MB flash)

### Device Description

Unexpected Maker FeatherS3

### Hardware Configuration

Nothing connected

### Version

v2.0.5

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

460800

### Description

If you create a filesystem image using `default_16MB.csv` or `large_spiffs_16MB.csv` it fails to upload (see: https://github.com/espressif/esptool/issues/795).

A custom partition table such as the following works fine:

```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     36K,     20K,
factory,  app,  factory, 64K,     4M,
spiffs,   data, spiffs,  ,        10M,
```

### Sketch

```cpp
Not related to a specific sketch.
```

### Debug Message

```plain
esptool.py v4.3
Serial port COM10
Connecting....
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:42:03:2c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00c90000 to 0x00ffffff...
Compressed 3604480 bytes to 6617...
Writing at 0x00c90000... (100 %)

A fatal error occurred: Invalid head of packet (0x6F): Possible serial noise or corruption.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

pauldupuyjr, 2022-11-12:
Confirm same "Invalid head of packet (0x6F)" error with Unexpected Maker ProS3 16M board [ESP32-S3 (revision v0.1)] and that the custom partition table here fixes the issue in PlatformIO 2.5.5.

SuGlider, 2022-11-12:
@CoryCharlton - Could you please test the UM FeatherS3 with the Arduino IDE?
I tested Arduino IDE and a S3 Board with 32MB Flash and it worked fine.

Configuration for the UM board in the IDE would be:

[image]

SuGlider, 2022-11-12:
The Arduino Core 2.0.5 uses esptool.py v4.2.1

CoryCharlton, 2022-11-13:
@SuGlider I updated the title to be more clear that the issue appears to be related specifically to [`default_16MB.csv`](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default_16MB.csv) or [`large_spiffs_16MB.csv`](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/large_spiffs_16MB.csv) partition tables.

I browse the repo but I'm not clear which partition table csv at https://github.com/espressif/arduino-esp32/tree/master/tools/partitions corresponds to the "16M Flash (2MB APP/12.5MB FATFS)" you show in the UI. If you can point me to the correct file I can compare the differences to see if anything jumps out.

That being said I'm not in a position right now to test the Arduino IDE but I did test earlier using default and large spiffs and encountered the same error I did when using PlatformIO or the esptool directly.

CoryCharlton, 2022-11-13:
Then again the data in the `large_spiffs_16mb.csv` looks accurate: https://docs.google.com/spreadsheets/d/1TukMJ4aA4ZDjClvgUbY9OfiUPdwDmbYCW7Wj_OyY7aU

I modified `large_spiffs_16mb.csv` to make the spiffs partition 6.5MB instead of the default 7MB which leaves 0.5MB free space and it uploaded fine. 

```
# Name,   Type, SubType,  Offset,   Size,  Flags
nvs, data, nvs, 0x9000, 0x5000,
otadata, data, ota, 0xE000, 0x2000,
app0, app, ota_0, 0x10000, 0x480000,
app1, app, ota_1, 0x490000, 0x480000,
spiffs, data, spiffs, 0x910000, 0x680000,
```

So this may not be an issue with the partition tables or Arduino Core and the [esptool issue](https://github.com/espressif/esptool/issues/795) might still be the cause (I'll let you two teams fight it out :P)

SuGlider, 2022-11-13:
Have you consider that it could be a Flash chip failure?
Can you try it in some other equivalient board?

SuGlider, 2022-11-13:
Another possible test, would be to lower the flashing baud rate to, maybe, 115200... 
I had a chip that I only could completely flash it when in 115200 baud.

SuGlider, 2022-11-13:
`A fatal error occurred: Invalid head of packet (0x6F): Possible serial noise or corruption.`
 It sounds like a communication issue, maybe not related to the partition definition...

CoryCharlton, 2022-11-13:
> Have you consider that it could be a Flash chip failure? Or even a bad cable issue... Can you try it in some other equivalient board/cable?

Here's what I have tried, all of which resulted in the same error. All of these combinations work fine if the filesystem image is slightly smaller than the entire 16MB.

- Three different UM FeatherS3 boards
- Multiple USB cables
- Flashing via the native ESP32-S3 USB port (no USB-to-UART bridge)
- Flashing via multiple external USB-to-UART bridge
- Flashing at different baud rates

I do not have any ESP32-S3 dev boards with 16MB flash that are not UM FeatherS3 but @pauldupuyjr is using an UM ProS3.

As far as flash chip failure goes it seems highly unlikely that multiple chips would all fail within the last 1KB of address space but work fine otherwise. Flashing application code or smaller filesystems works fine.

> `A fatal error occurred: Invalid head of packet (0x6F): Possible serial noise or corruption.` It sounds like a communication issue, maybe not related to the partition definition...

I agree which is why I started on the esptool side. 

The fact that I could modify the partition and get it to work then made me assume that it was related to the partition definition and open this issue. Admittedly I never actually validated the definitions but now that I have they appear correct; which is pointing me back at the esptool side (or even the ESP32-S3 🤷).

@SuGlider two and a half questions:

1. Do you have an ESP32-S3 dev board with 16MB flash that you can try with the `large_spiffs_16MB.csv` definition? If not can you try it on the 32MB board you referenced earlier?

2. Can you tell me which partition table csv at https://github.com/espressif/arduino-esp32/tree/master/tools/partitions corresponds to the "16M Flash (2MB APP/12.5MB FATFS)" you show in your screenshot?

lbernstone, 2022-11-13:
Does it work if you make your spiffs bin 1K smaller?  I don't know exactly how spiffs handles it, but ffat has 1K overhead that it is using for wear levelling, so the fs image must be 1k smaller than the partition size.  Perhaps the platform.io tool is not calculating correctly?

SuGlider, 2022-11-13:
> @SuGlider two and a half questions:
> 
> 1. Do you have an ESP32-S3 dev board with 16MB flash that you can try with the `large_spiffs_16MB.csv` definition? If not can you try it on the 32MB board you referenced earlier?

Yes, I just tried the ESP32-S3-Box board. https://www.espressif.com/en/news/ESP32-S3-BOX_video
This board (box) has 16MB QIO Flash and 8MB QIO PSRAM. 
I thinK it matches the UM S3 board.

I used Arduino IDE 1.8.15 and Arduino Core 2.0.5 to flash a simple Blink.ino, using as selected board the `UM FeatherS3`.

It worked fine with https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default_16MB.csv
This is `Default (6.25MB APP/3.43MB SPIFFS)` UM FeatherS3 partition in the Arduino IDE.

It also worked fine with https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/large_spiffs_16MB.csv
This is the `Large SPIFFS (4.5MB APP/6.93MB SPIFFS)` partition option the the Arduino IDE Menu.

The 32MB board I referenced earlier is an ESP32-S3-DevKitC-1 v1.1 board with an ESP32-S3-WROOM-2-N32R8V module with Octal SPI.

> 2. Can you tell me which partition table csv at https://github.com/espressif/arduino-esp32/tree/master/tools/partitions corresponds to the "16M Flash (2MB APP/12.5MB FATFS)" you show in your screenshot?

This is the file:
https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/ffat.csv

=========
Therefore, I can't reproduce this issue.

SuGlider, 2022-11-13:
In case it can help, the complete successful building/flashing log, using Win11, is:

```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\espUser\AppData\Local\Arduino15\packages -hardware C:\Users\espUser\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\espUser\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\espUser\Documents\Arduino\libraries -fqbn=esp32:esp32:feathers3:LoopCore=1,EventsCore=1,USBMode=default,CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,UploadMode=cdc,PSRAM=enabled,PartitionScheme=large_spiffs,CPUFreq=240,FlashMode=qio,UploadSpeed=921600,DebugLevel=none,EraseFlash=none -vid-pid=303A_1001 -ide-version=10815 -build-path C:\Users\espUser\AppData\Local\Temp\arduino_build_334590 -warnings=default -build-cache C:\Users\espUser\AppData\Local\Temp\arduino_cache_200775 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.mkspiffs.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.esptool_py.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.2.1 -prefs=runtime.tools.esptool_py-4.2.1.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.2.1 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.mklittlefs.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -verbose C:\Users\espUser\AppData\Local\Temp\arduino_modified_sketch_188141\Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\espUser\AppData\Local\Arduino15\packages -hardware C:\Users\espUser\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\espUser\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\espUser\Documents\Arduino\libraries -fqbn=esp32:esp32:feathers3:LoopCore=1,EventsCore=1,USBMode=default,CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,UploadMode=cdc,PSRAM=enabled,PartitionScheme=large_spiffs,CPUFreq=240,FlashMode=qio,UploadSpeed=921600,DebugLevel=none,EraseFlash=none -vid-pid=303A_1001 -ide-version=10815 -build-path C:\Users\espUser\AppData\Local\Temp\arduino_build_334590 -warnings=default -build-cache C:\Users\espUser\AppData\Local\Temp\arduino_cache_200775 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.mkspiffs.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.esptool_py.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.2.1 -prefs=runtime.tools.esptool_py-4.2.1.path=C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.2.1 -prefs=runtime.tools.riscv32-es [... cut ...]
Hash of data verified.
Compressed 252976 bytes to 144152...
Writing at 0x00010000... (11 %)
Writing at 0x0001cfda... (22 %)
Writing at 0x00022fd6... (33 %)
Writing at 0x0002869a... (44 %)
Writing at 0x0002da15... (55 %)
Writing at 0x00032f3a... (66 %)
Writing at 0x0003b66a... (77 %)
Writing at 0x000437fe... (88 %)
Writing at 0x00049038... (100 %)
Wrote 252976 bytes (144152 compressed) at 0x00010000 in 3.0 seconds (effective 682.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

SuGlider, 2022-11-13:
I noticed a few difference with your upload log....

`Chip is ESP32-S3 (revision v0.1)`  ==> `Chip is ESP32-S3`
`Changing baud rate to 460800`  ==> `Changing baud rate to 921600`
`Flash will be erased from 0x00c90000 to 0x00ffffff...`  ==> 
```
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x0004dfff...
```
`Writing at 0x00c90000... (100 %)`  ==>
```
Writing at 0x00010000... (11 %)
Writing at 0x0001cfda... (22 %)
Writing at 0x00022fd6... (33 %)
Writing at 0x0002869a... (44 %)
Writing at 0x0002da15... (55 %)
Writing at 0x00032f3a... (66 %)
Writing at 0x0003b66a... (77 %)
Writing at 0x000437fe... (88 %)
Writing at 0x00049038... (100 %)
```

SuGlider, 2022-11-13:
`Compressed 3604480 bytes to 6617...` ==> this is odd... how can it compress 3.5MB to 6K?
 
Arduino  IDE --> `Compressed 252976 bytes to 144152...` 250K to 144K seems correct, for instance.

 When I use Arduino IDE I get:
 
 ```
 Compressed 15040 bytes to 10363...
Writing at 0x00000000... (100 %)
Wrote 15040 bytes (10363 compressed) at 0x00000000 in 0.3 seconds (effective 399.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 481.3 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 644.5 kbit/s)...
Hash of data verified.
Compressed 252976 bytes to 144152...
```

SuGlider, 2022-11-13:
I think that there is something wrong with PIO building/flashing process.
@Jason2866 - Have you seen anything like that before?

lbernstone, 2022-11-13:
@SuGlider They are uploading filesystem images, not firmware.  The equivalent in Arduino IDE would be using https://github.com/me-no-dev/arduino-esp32fs-plugin.

Jason2866, 2022-11-13:
Mhh, it does work in general since i have a S3 16MB flash /8MB PSRAM (qio/opi) which does work when flashed from PIO.
Using upload speed above `460800` never worked reliable for me. So i would try that first. You are using `921600`

SuGlider, 2022-11-13:
> @SuGlider They are uploading filesystem images, not firmware.  The equivalent in Arduino IDE would be using https://github.com/me-no-dev/arduino-esp32fs-plugin.

Ok. I see now.
In that case, I think that the compression is wrong?!

Anyway, this issue isn't related to Arduino.

SuGlider, 2022-11-13:
I still think this is a PIO issue instead...
Or maybe a issue for the Esptool.py GH.

CoryCharlton, 2022-11-14:
You guys are missing the issue. Application code uploads fine. 

This is an issue related to uploading a filesystem that contains files (a simple TXT for is enough to reproduce). 

Uploading a filesystem is not directly possible in Arduino IDE and you would need a plugin: https://randomnerdtutorials.com/install-esp32-filesystem-uploader-arduino-ide/#installing-windows (this may have changed in Arduino IDE 2+).

All PIO or the Arduino IDE plug-in does is call out to the esptools.

So long story short this is likely an issue with the esptool at this point but I'm still waiting on them to confirm/reproduce.

SuGlider, 2022-11-14:
OK, thanks. Indeed I didn't got that at first glance.
Just as a report/information about the plugin from https://github.com/me-no-dev/arduino-esp32fs-plugin that allows Arduino IDE to upload fiiles to the SPIFFS partition:

It doesn't work for the S3... it only works for the ESP32. I got this error mesage:
```
A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?
SPIFFS Upload failed!
```

@me-no-dev FYI

SuGlider, 2022-11-14:
@CoryCharlton Regarding ESPTOOL cmd line, try adding `--no-compress` to it.

C:\Users\ccharlton\Downloads\esptool-v4.3-win64\esptool-v4.3-win64\esptool.exe --chip esp32s3 --port "COM10" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 16MB `--no-compress` 13172736 .pio\build\debug\spiffs.bin

SuGlider, 2022-11-14:
I just tried: `esptool.py -p com18 -b 115200 write_flash --no-compress 9502720 README.md` to my ESP32-S3-Box

I used this partition for a simple `blink.ino` : 
https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/large_spiffs_16MB.csv
 
### Successful Output:
```
esptool.py v3.3.2-dev
Serial port com18
Connecting...
Failed to get PID of a device on com18, using standard reset sequence.

Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e1:8b:78
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00910000 to 0x00911fff...
Wrote 16384 bytes at 0x00910000 in 0.2 seconds (660.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```

SuGlider, 2022-11-14:
I also just tried it with a large file (4MB bin file compressed to 1.8MB):

```
esptool.py -p com18 -b 115200 write_flash 9502720 esp32_s3_box_original_firmware.bin
esptool.py v3.3.2-dev
Serial port com18
Connecting...
Failed to get PID of a device on com18, using standard reset sequence.
.
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e1:8b:78
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00910000 to 0x00d0ffff...
Compressed 4194304 bytes to 1866032...
Wrote 4194304 bytes (1866032 compressed) at 0x00910000 in 38.0 seconds (effective 883.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

SuGlider, 2022-11-14:
It has used esptool.py `v3.3.2-dev` ... different from the 4.3 used within this issue.

SuGlider, 2022-11-14:
Tested with esptool.py `v4.2.1` - working fine:

```
C:\Users\espUser\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.2.1>
.\esptool.exe -p com18 -b 115200 write_flash 9502720 esptool.exe
esptool.py v4.2.1
Serial port com18
Connecting...
Failed to get PID of a device on com18, using standard reset sequence.

Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e1:8b:78
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00910000 to 0x00f5efff...
Compressed 6614264 bytes to 6367162...
Wrote 6614264 bytes (6367162 compressed) at 0x00910000 in 82.3 seconds (effective 643.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

CoryCharlton, 2022-11-14:
@SuGlider thanks for the suggestion. I did try the `--no-compress` option previously and it made no difference.

I think the `Compressed 3604480 bytes to 6617...` is a valid compression ratio since the partition in this example was 3.5MB but only contained a single test.txt file. So I would expect 3.5MB of zeroes to compress extremely well.

When I fill up the filesystem, especially with compressed data [ie: MP3 files], it compresses less:

`Compressed 12516352 bytes to 3527687...`

~12MB partition with ~3.5MB of MP3 files.

SuGlider, 2022-11-14:
@CoryCharlton - I think it narrows down to the esptool.py version. I see that your testing used version `4.3`
I tried versions `3.3` and `4.2` - it seems that both work fine. At least I could upload a file to flash at a specific address.

CoryCharlton, 2022-11-14:
@SuGlider I think that might be the different as well. You're writing a single file while I'm flashing the entire partition. Can you try uploading this [spiffs.zip](https://github.com/espressif/arduino-esp32/files/9999454/spiffs.zip) [had to zip the .bin so it would upload].

PIO used the following command to build the image from the `large_spiffs_16MB.csv` table: `"mkspiffs_espressif32_arduino" -c data -s 7274496 -p 256 -b 4096 .pio\build\debug\spiffs.bin` where `data` is the directory that contains the files I want inside the spiffs partition.

I suppose `mkspiffs_espressif32_arduino` could also be where the issue/bug is 🤷

lbernstone, 2022-11-14:
@CoryCharlton That command shows the data size as 0x6F0000 (7274496), whereas your partition is 0x68000.  Can you try manually running mkspiffs with the correct size, and see if that uploads properly?  This certainly looks like a problem with the pio script that is building the image file.

CoryCharlton, 2022-11-14:
@lbernstone that is using the `large_spiffs_16MB.csv` table from this repo that has the spiffs partition set to 0x6F0000.

My custom partition shown above works fine and decreased the spiffs partition by 1KB to 0x680000

Jason2866, 2022-11-14:
Just to be sure... Verified that the created spiffs partition with files is correctly generated?

CoryCharlton, 2022-11-14:
@Jason2866 I'm not sure how I would verify the bin file with the partition but I attached it in this [comment](https://github.com/espressif/arduino-esp32/issues/7467#issuecomment-1313085792). That being said the fact I can generate and upload a working partition bin by only decreasing the size slightly I'm going to assume it is a valid file (assumptions always bite me but I still like making them :P)

This [comment ](https://github.com/espressif/esptool/issues/795#issuecomment-1313240654) from @dobairoland also points out that the error is being triggered by the ESP32-S3 throwing an exception. Unfortunately I do not have a non-S3 board with a 16MB flash to test with but I'm willing to make the following assumptions (see, I love them) at this point:

1. The partition table is valid (this is less of an assumption because the table looks correct to me)
2. The filesystem generated is valid (see reasoning above)
3. This is a bug / issue specific to the S3 

Based on those assumptions I'm going to submit a PR shortly that will decrease the 16MB spiffs partitions by 1KB. To me this seems like an acceptable work around based on the fact that it's only 1KB being wasted.

The other alternative would be to create S3 specific partition tables but this seems like more work (ie: the boards list would need to be changed for them all). That also assumes the error is specific to the S3 which is 100% assumption on my part at this time.

SuGlider, 2022-11-14:
Another detail... I think that esptool.py doesn't care about the partition scheme. 
Therefore, sending a PR to change it wouldn't really fix anything.

It seems a issue with PlatformIO instead. 
PIO may calculate size, positions etc based on the partition table information.

CoryCharlton, 2022-11-14:
@SuGlider correct, `esptool` and `mkspiffs_espressif32_arduino` doesn't know about the partition scheme and just uses the parameters passed in to perform their work.

However the IDE (in this case PIO) is aware of the partition scheme and uses those values to determine what parameters to pass. PIO is just passing the values as they are specified in the partition CSV file and everything indicates it's doing the correct thing at this point (based on the data in the CSV).

Here's an example:

[default_16MB.csv](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default_16MB.csv) - From this repo

```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x640000,
app1,     app,  ota_1,   0x650000,0x640000,
spiffs,   data, spiffs,  0xc90000,0x370000,
```

SPIFFS offset: 0xc90000 - 13172736
SPIFFS size: 0x370000 - 3604480

Make filesystem:

```
>  C:\Users\ccharlton\.platformio\penv\Scripts\platformio.exe run --verbose --target buildfs --environment debug
Processing debug (build_type: debug; build_flags: -DARDUINO_USB_MODE=1 -DCORE_DEBUG_LEVEL=5; platform: espressif32; board: um_feathers3; board_build.partitions: default_16MB.csv; framework: arduino; monitor_port: COM14; upload_port: COM14; lib_deps: earlephilhower/ESP8266Audio@^1.9.7)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/um_feathers3.html
PLATFORM: Espressif 32 (5.2.0) > Unexpected Maker FeatherS3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5)
 - tool-esptoolpy @ 1.40201.0 (4.2.1)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP8266Audio @ 1.9.7 (License: Unknown, Path: C:\src\temp\i2s-audio-test\.pio\libdeps\debug\ESP8266Audio)
|   |-- SPIFFS @ 2.0.0 (License: Unknown, Path: C:\Users\ccharlton\.platformio\packages\framework-arduinoespressif32\libraries\SPIFFS)
|   |   |-- FS @ 2.0.0 (License: Unknown, Path: C:\Users\ccharlton\.platformio\packages\framework-arduinoespressif32\libraries\FS)
|   |-- I2S @ 1.0 (License: Unknown, Path: C:\Users\ccharlton\.platformio\packages\framework-arduinoespressif32\libraries\I2S)
|   |-- FS @ 2.0.0 (License: Unknown, Path: C:\Users\ccharlton\.platformio\packages\framework-arduinoespressif32\libraries\FS)
|   |-- HTTPClient @ 2.0.0 (License: Unknown, Path: C:\Users\ccharlton\.platformio\packages\framework-arduinoespressif32\libraries\HTTPClient)
|   |   |-- WiFi @ 2.0.0 (License: Unknown, Path: C:\Users\ccharlton\.platformio\packages\framework-arduinoespressif32\libraries\WiFi)
|   |   |-- WiFiClientSecure @ 2.0.0 (License: Unknown, Path: C:\Users\ccharlton\.platformio\packages\framework-arduinoespressif32\libraries\WiFiClientSecure)
|   |   |   |-- WiFi @ 2.0.0 (License: Unknown, Path: C:\Users\ccharlton\.platformio\packages\framework-arduinoespressif32\libraries\WiFi)
|   |-- SD @ 2.0.0 (License: Unknown, Path: C:\Users\ccharlton\.platformio\packages\framework-arduinoespressif32\libraries\SD)
|   |   |-- FS @ 2.0.0 (License: Unknown, Path: C:\Users\ccharlton\.platformio\packages\framework-arduinoespressif32\libraries\FS)
|   |   |-- SPI @ 2.0.0 (License: Unknown, Path: C:\Users\ccharlton\.platformio\packages\framework-arduinoespressif32\libraries\SPI)
|   |-- SPI @ 2.0.0 (License: Unknown, Path: C:\Users\ccharlton\.platformio\packages\framework-arduinoespressif32\libraries\SPI)
|-- I2S @ 1.0 (License: Unknown, Path: C:\Users\ccharlton\.platformio\packages\framework-arduinoespressif32\libraries\I2S)
|-- SPIFFS @ 2.0.0 (License: Unknown, Path: C:\Users\ccharlton\.platformio\packages\framework-arduinoespressif32\libraries\SPIFFS)
|   |-- FS @ 2.0.0 (License: Unknown, Path: C:\Users\ccharlton\.platformio\packages\framework-arduinoespressif32\libraries\FS)
Building in debug mode
**"mkspiffs_espressif32_arduino" -c data -s 3604480 -p 256 -b 4096 .pio\build\debug\spiffs.bin**
/test.txt
**============================================================================================== [SUCCESS] Took 5.17 seconds ==============================================================================================**
```

Upload filesystem:

```
>  C:\Users\ccharlton\.platformio\penv\Scripts\platformio.exe run --verbose --target uploadfs --environment debug
Processing debug (build_type: debug; build_flags: -DARDUINO_USB_MODE=1 -DCORE_DEBUG_LEVEL=5; platform: espressif32; board: um_feathers3; board_build.partitions: default_16MB.csv; framework: arduino; monitor_port: COM14; upload_port: COM14; lib_deps: earlephilhower/ESP8266Audio@^1.9.7)
[... cut ...]
**============================================================================================= [SUCCESS] Took 50.67 seconds =============================================================================================**
```

So you can see PIO is just passing the values supplied from the partition CSV but without the 1KB buffer the partition fails to upload (due to a bug in the ESP32-S3?).

Long story short my PR will workaround this issue and allow users to create and upload a SPIFFS filesystem to an ESP32-S3 with 16MB flash. 

As it stands now users are unable to do this due to a mysterious/vague error message that may not be able to get any clearer.

Jason2866, 2022-11-14:
@CoryCharlton  The PR is not needed. This [partitions.csv](https://github.com/arendst/Tasmota/blob/development/partitions/esp32_partition_app2944k_fs10M.csv) works for me. Platformio uploads to a S3 without any issue. No need to reduce size.
Still thinking you have a hardware problem `A fatal error occurred: Invalid head of packet (0x6F): Possible serial noise or corruption`

SuGlider, 2022-11-14:
Files Used in this testing (remove, if necessary, the .txt extension of the file):
[spiffs_3.604.480_bytes.bin.txt](https://github.com/espressif/arduino-esp32/files/10007506/spiffs_3.604.480_bytes.bin.txt)
[spiffs_3.604.476_bytes.bin.txt](https://github.com/espressif/arduino-esp32/files/10007508/spiffs_3.604.476_bytes.bin.txt)

lbernstone, 2022-11-14:
@SuGlider Are you generating the files with mkspiffs? Are they created the correct size?  It ought to be aligned to the page/block dimensions. I'm sure when the tool was built it was never tested against an image larger than 1MB...

SuGlider, 2022-11-14:
No, I just created it manually... but it allowed me to verify the issue anyway,
The sizes are correct. The files can be downloaded from this comment:
https://github.com/espressif/arduino-esp32/issues/7467#issuecomment-1314484711

SuGlider, 2022-11-14:
@CoryCharlton - Please test it with these suggested spiffs partition sizes:

tools/partitions/default_16MB.csv
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x640000,
app1,     app,  ota_1,   0x650000,0x640000,
spiffs,   data, spiffs,  0xc90000,36FFFC,
```

tools/partitions/large_spiffs_16MB.csv
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x480000,
app1,     app,  ota_1,   0x490000,0x480000,
spiffs,   data, spiffs,  0x910000,0x6EFFFC,
```

CoryCharlton, 2022-11-14:
@SuGlider confirmed that decreasing by 4 bytes resolves the issue. PR updated.

Jason2866, 2022-11-15:
The PR is against the docs https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/api-guides/partition-tables.html

dobairoland, 2022-11-15:
If you make a file of 16 kB of all zeros and try to flash it to the end of the flash region (`0xffbfff` - `0xffffff`) then are you getting the same error? Could you please try with `--compress` and `--no-compress` as well and provide the traces (`--trace`)?

CoryCharlton, 2022-11-15:
@dobairoland that test failed with the same error. 

Here's the data I used:

[empty_16KB.bin.txt](https://github.com/espressif/arduino-esp32/files/10015631/empty_16KB.bin.txt)

And the output:

[trace_empty_16kb_compress.txt](https://github.com/espressif/arduino-esp32/files/10015629/trace_empty_16kb_compress.txt)
[trace_empty_16kb_no_compress.txt](https://github.com/espressif/arduino-esp32/files/10015630/trace_empty_16kb_no_compress.txt)

dobairoland, 2022-11-16:
@CoryCharlton Thank you for testing and providing reports.

@SuGlider I see you've got an ESP32-S3-Box. Could you please try https://github.com/espressif/arduino-esp32/issues/7467#issuecomment-1315171483. I was not able to reproduce with 8MB flash (https://github.com/espressif/esptool/issues/795#issuecomment-1316763104) and don't have access to a board with 16MB flash.

beckmx, 2022-12-02:
@dobairoland I tried that one, but was not able to compile my esptool locally

06GitHub, 2022-12-08:
For sharing, I got the same error "**Invalid head of packet (0x6F): Possible serial noise or corruption**" with **Arduino IDE** when uploading data files on **16Mb** Flash (using arduino-esp32fs-plugin)
- Arduino IDE 1.8.18
- ESP32-S3-WROOM-1(N16R8)
- USBOTG (No FTDI)

As highlighted by @SuGlider, **decreasing flash size by 4 bytes corrects the error**, see KO partition table and OK partition table below.

Thanks to @SuGlider for this workaround 😄 👍 

This error does not occur when flash size is set to 8Mb or less.

**KO**
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x140000,
app1,     app,  ota_1,   0x150000,0x140000,
spiffs,   data, spiffs,  0x290000,0xD70000,
[...]
->Writing at 0x00ff9cc2... (99 %)
->Writing at 0x00ffe568... (100 %)
->
->A fatal error occurred: Invalid head of packet (0x6F): Possible serial noise or corruption.
LittleFS Upload failed!
```

**OK**
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x140000,
app1,     app,  ota_1,   0x150000,0x140000,
spiffs,   data, spiffs,  0x290000,0xD6FFFC,
[...]
->Writing at 0x00a37149... (99 %)
->Writing at 0x00ab1afb... (100 %)
->Wrote 14090236 bytes (4642084 compressed) at 0x00290000 in 62.7 seconds (effective 1798.5 kbit/s)...
->Hash of data verified.
->
->Leaving...
->Hard resetting via RTS pin.
```

Jason2866, 2022-12-08:
Forked esptool.py and changed the stub loader code as suggested (not from me!) to fix this issue. You can try it is [here](https://github.com/Jason2866/esptool/releases)
[Source](https://github.com/Jason2866/esptool)

CoryCharlton, 2023-01-06:
This has been resolved in the [esptool](https://github.com/espressif/esptool/commit/21e59148e73e384947327a0883a753685e03d143).

closed 2023-01-06 by CoryCharlton