espressif/arduino-esp32 #7468: ESP32 not connected to WIFI after update from Arduino ESP32 board 1.06 to 2.05

Nulista, 2022-11-13:
### Board

Lolin32

### Device Description

Only Wemos Lolin32 V1.0.0

### Hardware Configuration

notting connected to the board

### Version

v2.0.5

### IDE Name

Arduino 1.8.19

### Operating System

windows 11

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

This simple code works perfectly on arduinosp32 V1.06, after update to v2.05 I can't connect to WiFi.
I'm sure ssid and password are correct.
A scan function works good(see debug)

### Sketch

```cpp
#include "WiFi.h"

char* ssid = "--------";
char* password =  "------------";

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);
  Serial.println("scan start");

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks();
  Serial.println("scan done");
  if (n == 0) {
    Serial.println("no networks found");
  } else {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i) {
      // Print SSID and RSSI for each network found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN) ? " " : "*");
      delay(10);
    }
  }
  Serial.println("");
  delay(500);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.println("Connecting to WiFi..");
  }
  Serial.print("Connected to the WiFi network, IP = ");
  Serial.println(WiFi.localIP());
}
void loop() {}
```

### Debug Message

```plain
[     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    57][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   145][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
scan start
[  6254][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE
scan done
4 networks found
1: Tenda_2208E0 (-57)*
2: SMA1992011276 (-76)*
3: Proximus Public Wi-Fi (-90)*
4: Proximus-Home-0418 (-90)*

Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
[ 12811][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 12812][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 12815][D][WiFiGeneric.cpp:966] _eventCallback(): WiFi Reconnect Running
Connecting to WiFi..
Connecting to WiFi..
```

### Other Steps to Reproduce

I tried another board, I tried another computer, I even tried via platformio.
I have tried another router but all without success.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

felmue, 2022-11-13:
Hello @Nulista 

hmm, your code works for me with either arduinoesp32 framework version v1.0.6 or v2.0.5. If anything at all it takes a tick longer to connect with v2.0.5.

Have you tried without scanning first?

Thanks
Felix

Nulista, 2022-11-13:
> Van "felmue" ***@***.***>
> Aan "espressif/arduino-esp32" ***@***.***>
> CC "ludo nulens" ***@***.***>; "Mention"
> Datum 13/11/2022 15:15:27
> Onderwerp Re: [espressif/arduino-esp32] ESP32 not connected to WIFI
> after update from Arduino ESP32 board 1.06 to 2.05 (Issue #7468)

Hoi Felix,

Yes, i tried without scanning first.
i tried with multiWiFi also, but no result.
With arduinoesp32 framework version v2.0.0 works it good, but no version 
higher.

Yes it's a mystery to me what the error could be.
My router is ± 8 years old, a Tenda N630 V2 Wireless WIFI Router 300Mbps 
802.11 a/b/g/n/3/3 
<https://www.aliexpress.com/item/32245075403.html?spm=a2g0o.order_list.0.0.21ef79d2yTRNj4>

What couldt i trie?

Nulista, 2022-11-13:
Hello,

I tried to change the seruritysettings in the router from Wpa-PSK  to  
Wpa/Wpa2 - PSK.
And now it works good with version 2.05.

Thanks a lot Felix for trying my code en the reply!

problem solved

Van "Ludo Nulens" ***@***.***>

Datum 13/11/2022 16:00:30
Onderwerp Re[2]: [espressif/arduino-esp32] ESP32 not connected to WIFI

P-R-O-C-H-Y, 2022-11-14:
@Nulista Closing this as solved :)

closed 2022-11-14 by P-R-O-C-H-Y