espressif/arduino-esp32 #7476: ESP32 don't run void setup() if it is being powered by arduino uno

avryux, 2022-11-15:
### Board

NODEMCU ESP32-S

### Device Description

Arduino Uno  and NODEMCU ESP32-S

### Hardware Configuration

RX of Arduino UNO connected to TX of ESP32

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I was testing the code of the ESP32 connected to USB and watching the COM and it works properly. But when i connect the Arduino UNO with a battery 9V and i connect the ESP32 to the Arduino (3.3V) the ESP doesn't turn on the BL (this is on the void setup(), so looks like the ESP it's not turning on correctly )

### Sketch

```cpp
ESP32 code:
...
void setup() {
  Serial.begin(115200);
  SerialBT.begin("VLOCK"); //Bluetooth device name
}

void loop() {
  if (SerialBT.available()) {
      dato = SerialBT.read();
      if (dato == '/') {
        if (contador == 0) {
          action = MensajeTemp; //action tendra el valor de donde proviene la info, 9 = VLockAPP
        }
        else if (contador == 1 && action == "9") {
          action_door = MensajeTemp;
        }
        else if (contador == 2 && action == "9") {
          Input_password = MensajeTemp;  
        } 
        contador += 1;
        MensajeTemp = "";
        if (contador == 3) {
          contador = 0;
          action = "";
          Serial.print("p/"+ action_door + "/" + Input_password + "/");
          //Check privileges in DB
          if (Firebase.ready() && signupOK && (millis() - sendDataPrevMillis > 15000 || sendDataPrevMillis == 0)) { //owners
            sendDataPrevMillis = millis();
            rutanodo = "/owners/";
            rutanodo += Input_password;
            if (Firebase.RTDB.getBool(&fbdo, rutanodo)) {
              if (fbdo.dataType() == "boolean") {
                Serial.print("f/true/");
              }
            } else { // admins
                rutanodo = "/admins/";
                rutanodo += Input_password;
                 if (Firebase.RTDB.getBool(&fbdo, rutanodo)) {
                  if (fbdo.dataType() == "boolean") {
                    Serial.print("f/true/");
                  } 
                 } else { //users
                    rutanodo = "/users/";
                    rutanodo += Input_password;
                    if (Firebase.RTDB.getArray(&fbdo, rutanodo)) {
                      if (fbdo.dataType() == "array") {
                        FirebaseJsonArray &arr = fbdo.jsonArray();
                        FirebaseJsonData result;
                        struct tm timeinfo;
                        if(!getLocalTime(&timeinfo)){
                          Serial.println("Failed to obtain time");
                        }
                        char timeWeekDay[10];
                        strftime(timeWeekDay,10, "%a", &timeinfo);
                        if (strcmp(timeWeekDay, "Mon") == 0) {
                          arr.get(result, 0);
                        } else if (strcmp(timeWeekDay, "Tue") == 0) {
                          arr.get(result, 1);
                        } else if (strcmp(timeWeekDay, "Wed") == 0) {
                          arr.get(result, 2);
                        } else if (strcmp(timeWeekDay, "Thu") == 0) {
                          arr.get(result, 3);
                        } else if (strcmp(timeWeekDay, "Fri") == 0) {
                          arr.get(result, 4);
                        } else if (strcmp(timeWeekDay, "Sat") == 0) {
                          arr.get(result, 5);
                        } else if (strcmp(timeWeekDay, "Sun") == 0) {
                          arr.get(result, 6);
                        }
                        if(result.to<String>() == "true") {
                          Serial.print("f/true/");
                        }
                      }
                    }
                  }
              }
          }
        }
      } else {
        MensajeTemp += dato;
        }
  
  }
}

Arduino code:
// variable para entender de donde proviene la información
String action;
char dato;
//variables para dividir la información por "/"
int contador = 0;
int contadorf = 0;
String MensajeTemp;
// variables de VLockapp
String action_door;
String Input_password;

int SensorPin = 3;
int state;
int RelePin = 4;

String PassOff1 = "Off1";
String PassOff2 = "SJ2043";
String PassOff3 = "UsuarioOff";

void setup() {
  Serial.begin(115200);
  pinMode(SensorPin, INPUT_PULLUP);
  pinMode(RelePin, OUTPUT);
  digitalWrite(RelePin, HIGH);
}

void loop() {
  if (Serial.available()) {
    dato = Serial.read();
    if (dato == '/') {
      if (contador == 0) {
        action = MensajeTemp; //action tendra el valor de donde proviene la info, p = ESP32 manda p/action/password/
        }
      if (contador == 1 && action == "p") {
        action_door = MensajeTemp;
        }
      if (contador == 1 && action == "f") { //Abrir / cerrar la puerta
          contadorf = 1;
      }
      if (contador == 2 && action == "p") {
        Input_password = MensajeTemp;  
      }
      contador += 1;
      MensajeTemp = "";
      if (contador == 3) {
        contador = 0;
        action = "";
        Serial.println("El privilegio es: " + Input_password + " y la acción es: " + action_door);
        Serial.println("Action: " + action);
        Serial.println("Action_door: " + action_door);
        Serial.println("Password: " + Input_password);
        if (Input_password == PassOff1 || Input_password == PassOff2 || Input_password == PassOff3) { //Abrir / cerrar puerta usuarios offline
          Serial.print("Abriendo la puerta");
          digitalWrite(RelePin, LOW);
          state = 1;
          delay(4000);
          digitalWrite(RelePin, HIGH);
          state = 0;
        }
      } else if (contadorf == 1) {
          Serial.print("Abriendo la puerta");
          digitalWrite(RelePin, LOW);
          state = 1;
          delay(4000);
          digitalWrite(RelePin, HIGH);
          state = 0;
          MensajeTemp = "";
          contador = 0;
          action = "";
          contadorf = 0;
        }
    } else {
      MensajeTemp += dato;
      }
  }
}
```

### Debug Message

```plain
No debug because the error happen without USB connected
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-11-15:
The power regulator on the uno probably does not supply enough current to run the esp32, let alone both devices.
You can run 9V to the input pins (5V) on both devices, but I have no guarantees that the regulator & capacitors on the esp32 are rated for 9V.

avryux, 2022-11-15:
The ESP doesn't support 9V :S, also, do you know if i can use the default Serial for communicate the ESP with the Arduino if i don't use USB?

lbernstone, 2022-11-15:
This is a forum for issues with the code repository here.  For general questions about not causing fires, please use http://esp32.com

SuGlider, 2022-11-16:
Just to close this issue:

1- When ESP32 Bluetooth is turned on, it consumes a peak of current. The 3.3V of the UNO may not supply such current, therefore, it may reset on brownout. As suggested, you may try to use an other appropriate LDO regulator to convert from 9V to 5V and supply it to the ESP32 board.

2- Yes, you can connect UNO UART to ESP32 UART directly. I have done it before and it works fine. Just don't forget to cross connect RX and TX, plus use a common GND.

closed 2022-11-16 by SuGlider