espressif/arduino-esp32 #7479: src/parsebytes.h

sbx59, 2022-11-16:
### Board

upload

### Device Description

esp32 cam

### Hardware Configuration

esp + esp cam

### Version

latest master (checkout manually)

### IDE Name

Arduino

### Operating System

Win10

### Flash frequency

40mhz

### PSRAM enabled

no

### Upload speed

11520

### Description

When uploading I got this error message

esp32-cam-webserver:8:10: fatal error:  src/parsebytes.h : No such file or directory
 #include < src/parsebytes.h >
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1
src/parsebytes.h : No such file or directory

[image]

### Sketch

```cpp
#include <esp_camera.h>
#include <esp_int_wdt.h>
#include <esp_task_wdt.h>
#include <WiFi.h>
#include <DNSServer.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <src/parsebytes.h>
#include "time.h"
#include <ESPmDNS.h>
```

### Debug Message

```plain
esp32-cam-webserver:8:10: fatal error:  src/parsebytes.h : No such file or directory
 #include < src/parsebytes.h >
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
src/parsebytes.h : No such file or directory
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-11-16:
you probably copied the code from somewhere else. We do have a camera example in this repo that you can try instead.

sbx59, 2022-11-16:
@me-no-dev the file is at the root of the folder. he is present.

me-no-dev, 2022-11-16:
then maybe you need just `#include "parsebytes.h"`

playerfreak, 2023-05-03:
i get the same 

#include "parsebytes.h"
          ^~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: parsebytes.h: No such file or directory