espressif/arduino-esp32 #7481: Can't access or find 'gen_esp32part.exe' file compilation errors

nikstp, 2022-11-16:
### Board

ESP32 Adafruit Huzzah 

### Device Description

[Adafruit Huzzah ESP32 Feather](https://learn.adafruit.com/adafruit-huzzah32-esp32-feather)

### Hardware Configuration

None

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

921000

### Description

Can't seem to compile any of the sketches because I'm missing the "gen_esp32part.exe" file. I've reinstalled the boards and the library multiple times through the Arduino IDE, but I'm not sure if the file is made from the py file or is needed separately. Right now I'm considering doing a manual install, but it'd be nice if it also just worked "out of the box".

I've tried Adafruit ESP32 Feather and Adafruit ESP32-S2 Feather boards for compiling and I get the same result. 

### Sketch

```cpp
Any sketch.
```

### Debug Message

```plain
fork/exec C:\Users\nikstp\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5/tools/gen_esp32part.exe: Access is denied.

Compilation error: fork/exec C:\Users\nikstp\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5/tools/gen_esp32part.exe: Access is denied.
```

### Other Steps to Reproduce

N/A

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-11-16:
Perhaps your antivirus is eating it?  It is certainly a part of both the manual and boards manager installs.

nikstp, 2022-11-16:
@lbernstone yeah, I've gotten both "not found" and "access denied". I've tried starting the Arduino IDE with admin rights but I still get the same issue which points to the fact that it's just missing. Not sure why that is so; I'll try to manually replace all of those files with the files that are in this core repo (with executables present) and see what happens.

nikstp, 2022-11-16:
Very interesting, I'd assume it's something to do with my system protection software. Even when I managed to drag and drop the file into the correct folder from the zip of this repo, it was hidden and by default owned by the system. Even changing administration privileges to full control for myself as a user and an admin did not work right. When I just extract the repo into, say, a downloads folder, the file is missing. It's the only one, 'get.exe' and 'espota.exe' are still there. I might ask IT about this.

SuGlider, 2022-11-21:
It seems not related to ESP32 Arduino github... please advise.

nikstp, 2022-11-21:
@SuGlider we can close this for now; seems it's a local issue.

closed 2022-11-21 by VojtechBartoska