espressif/arduino-esp32 #7483: [2.0.3+] LWIP PPP disabled in sdkconfig/menuconfig

chegewara, 2022-11-17:
### Board

esp32

### Device Description

not important

### Hardware Configuration

not important

### Version

v2.0.3

### IDE Name

platformio

### Operating System

linux

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

Maybe i am just using old and outdated esp_modem library, but here is the problem.
- `#include "esp_netif_ppp.h"`
- this header exist
- declarations of missing functions also exist in this header
- looks like definitions are not compiled

With v2.0.0 everything works fine, but i wanted to migrate to v2.0.3 and onward. 
After comparing sdkconfig files i am guessing its because PPP is disabled:
```
< 2.0.0 
> 2.0.3
< #define CONFIG_LWIP_MAX_SOCKETS 10
> #define CONFIG_LWIP_MAX_SOCKETS 16
> #define CONFIG_LWIP_DHCP_DISABLE_VENDOR_CLASS_ID 1
> #define CONFIG_LWIP_DHCP_OPTIONS_LEN 128
< #define CONFIG_LWIP_IPV6_RDNSS_MAX_DNS_SERVERS 0
< #define CONFIG_LWIP_PPP_SUPPORT 1
< #define CONFIG_LWIP_PPP_ENABLE_IPV6 1
< #define CONFIG_LWIP_PPP_PAP_SUPPORT 1
< #define CONFIG_LWIP_PPP_CHAP_SUPPORT 1
< #define CONFIG_LWIP_PPP_MSCHAP_SUPPORT 1
< #define CONFIG_LWIP_PPP_MPPE_SUPPORT 1
> #define CONFIG_LWIP_SNTP_MAX_SERVERS 3
> #define CONFIG_LWIP_DHCP_GET_NTP_SRV 1
< #define CONFIG_PPP_CHAP_SUPPORT CONFIG_LWIP_PPP_CHAP_SUPPORT
< #define CONFIG_PPP_MPPE_SUPPORT CONFIG_LWIP_PPP_MPPE_SUPPORT
< #define CONFIG_PPP_MSCHAP_SUPPORT CONFIG_LWIP_PPP_MSCHAP_SUPPORT
< #define CONFIG_PPP_PAP_SUPPORT CONFIG_LWIP_PPP_PAP_SUPPORT
< #define CONFIG_PPP_SUPPORT CONFIG_LWIP_PPP_SUPPORT
> #define CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP
```

Now the question is, is it considered as a bug or is it on purpose, so i knew what to do with it next.

Thanks

[sdkconfig-2.0.0.txt](https://github.com/espressif/arduino-esp32/files/10028159/sdkconfig-2.0.0.txt)
[sdkconfig-2.0.3.txt](https://github.com/espressif/arduino-esp32/files/10028160/sdkconfig-2.0.3.txt)

### Sketch

```cpp
too big project, cant deliver small sketch
```

### Debug Message

```plain
/home/chegewara/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/roobuck/src/networking/lte_pppos.cpp.o:(.literal._Z13CreateLteInitv+0x20): undefined reference to `NETIF_PPP_STATUS'
/home/chegewara/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/chegewara/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/lib/libesp_netif.a(esp_netif_lwip_defaults.c.obj):/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_netif/lwip/esp_netif_lwip_defaults.c:52: undefined reference to `esp_netif_lwip_ppp_input'
/home/chegewara/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/roobuck/lib015/libmodem.a(esp_modem_netif.c.o):(.literal.esp_modem_post_attach_start+0x1c): undefined reference to `esp_netif_ppp_set_params'
/home/chegewara/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/roobuck/lib015/libmodem.a(esp_modem_netif.c.o): in function `esp_modem_post_attach_start':
../lib/modem/src/esp_modem_netif.c:82: undefined reference to `esp_netif_ppp_set_params'
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-11-17:
neither bug or on purpose. consequence of changing the build system in the lib-builder or change of defaults in ESP-IDF. Feel free to submit a PR in the lib-builder to enable that :)

SuGlider, 2022-11-17:
It was done on purpose because the final WiFi binary grows about 60KB with PPP enabled.
@me-no-dev, we removed it from sdkconfig file when users were complaining about the binary size in version 2. 0.0+.

SuGlider, 2022-11-17:
If the application needs LWIP PPP, I'd suggest using Arduino as Component + enabling it in the sdkconfig.
I think that the users that need PPP may be in the more expert side of the spectrum, thus, it won't be a problem going with Arduino as IDF Component project.

chegewara, 2022-11-17:
Dont take me wrong, i am not complaining, just i was surprised by the fact its working in 2.00 and not working in 2.0.3+ (maybe earlier).
Like you said, "more advanced" users can enable it, and voila, its enabled already:
https://github.com/chegewara/arduino-esp32-custom/tree/lwip_ppp_enabled

I can accept any option, suggested by @me-no-dev or by @SuGlider , so feel free to close topic.

Thanks

chegewara, 2022-11-17:
I also did quick comparison. v2.0.0 vs my custom build with PP enabled, and its not that bad like you said, its not 60kB:
```
esp-idf v4.4.3 with most recent arduino-builder and PPP enabled
RAM:   [==        ]  17.9% (used 58504 bytes from 327680 bytes)
Flash: [========  ]  77.3% (used 1570701 bytes from 2031616 bytes)

arduino-esp32 v2.0.0
RAM:   [==        ]  16.5% (used 53980 bytes from 327680 bytes)
Flash: [========  ]  76.8% (used 1560077 bytes from 2031616 bytes)
```
Its about 10kB and it may be anything, not just a PPP, because between v2.0.0 and 2.0.5 there is a lot changes, which could cause this. Also there may be a lot changes, which could cause shrinking code size and enabling PPP actually caused 60kB grow in size.

BTW this is huge project, with wifi, mqtt, lte, ble, dozen tasks and more, and with all enabled still leaves about 50kB free heap on esp32 without psram

SuGlider, 2022-11-18:
Ok, let's consider your analysis for the next release.

garudaonekh, 2023-01-05:
> 

Hello, how to install your board in replace of original ESP32 2.0.6?

Elekronixx, 2023-03-22:
(maybe earlier). Like you said, "more advanced" users can enable it, and voila, its enabled already: https://github.com/chegewara/arduino-esp32-custom/tree/lwip_ppp_enabled

@chegewara How can I use it in the IDE ? I tied : Update the 2.0.0 to 2.0.5 by boardmanager (working) and then copy your github files into the /arduino_ESP32_IDF2.0.5/portable/packages/esp32/hardware/esp32/2.0.5 directory. But I get 

```
Board esp32 (Plattform esp32, Paket esp32) is unknown

Error compiling for Board ESP32 Dev Module.
```
BR 'less advanced user' :-)

chegewara, 2023-03-22:
platformio maybe?

there is few more options, but this is one i am using

Elekronixx, 2023-03-22:
Thank you very much. I tired to install it - but it look that I first have to update things, install an new OS, buy a new computer ... Not easy at all.

chegewara, 2023-03-22:
Sorry, i cant help you with that.
Like i said, you have more options. Try to install 2 instances of arduino-esp32, the original one and the one from my repo.
This is where is my second copy of arduino-esp32 on linux
[image]

I cant help with windows thou.

Elekronixx, 2023-03-22:
Yes,clear. I will test some alternative ways. Thank you for your help and for the test to enable PPP again.

Thermelgy-Repo, 2023-06-06:
> Dont take me wrong, i am not complaining, just i was surprised by the fact its working in 2.00 and not working in 2.0.3+ (maybe earlier). Like you said, "more advanced" users can enable it, and [...]

Dear @chegewara or @SuGlider or @me-no-dev,

I am facing the same issue, But not able to solve it. I am getting some Linking issue like this :

```
Linking .pio\build\esp32dev\firmware.elf
c:/users/asish/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/lib5ce/libPPPCom_Lib.a(PPPCom_Lib.cpp.o):(.literal._ZN5PPPoS17pppos_client_taskEPv+0x3c): undefined reference to `pppapi_pppos_create'
c:/users/asish/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/lib5ce/libPPPCom_Lib.a(PPPCom_Lib.cpp.o):(.literal._ZN5PPPoS17pppos_client_taskEPv+0x40): undefined reference to `ppp_free'
c:/users/asish/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/lib5ce/libPPPCom_Lib.a(PPPCom_Lib.cpp.o):(.literal._ZN5PPPoS17pppos_client_taskEPv+0x48): undefined reference to `pppapi_set_notify_phase_callback'
c:/users/asish/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/lib5ce/libPPPCom_Lib.a(PPPCom_Lib.cpp.o):(.literal._ZN5PPPoS17pppos_client_taskEPv+0x4c): undefined reference to `pppapi_set_default'
c:/users/asish/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/lib5ce/libPPPCom_Lib.a(PPPCom_Lib.cpp.o):(.literal._ZN5PPPoS17pppos_client_taskEPv+0x50): undefined reference to `pppapi_set_auth'
c:/users/asish/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/lib5ce/libPPPCom_Lib.a(PPPCom_Lib.cpp.o):(.literal._ZN5PPPoS17pppos_client_taskEPv+0x54): undefined reference to `pppapi_connect'
c:/users/asish/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/lib5ce/libPPPCom_Lib.a(PPPCom_Lib.cpp.o):(.literal._ZN5PPPoS17pppos_client_taskEPv+0x5c): undefined reference to `pppapi_close'
c:/users/asish/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/lib5ce/libPPPCom_Lib.a(PPPCom_Lib.cpp.o):(.literal._ZN5PPPoS17pppos_client_taskEPv+0x60): undefined reference to `pppos_input_tcpip'
c:/users/asish/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/lib5ce/libPPPCom_Lib.a(PPPCom_Lib.cpp.o): in function `PPPoS::pppos_client_task(void*)':
F:\Aspiration Energy\Heat Pump Monitoring\Hardware Development\Thermelgy tMY Future\Firmware\Development\Thermelgy Gen2 Gateway\Thermelgy_Gen2_Gateway/lib/PPPCom_Lib/PPPCom_Lib.cpp:397: undefined reference to `pppapi_pppos_create'
c:/users/asish/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: F:\Aspiration Energy\Heat Pump Monitoring\Hardware Development\Thermelgy tMY Future\Firmware\Development\Thermelgy Gen2 Gateway\Thermelgy_Gen2_Gateway/lib/PPPCom_Lib/PPPCom_Lib.cpp:408: undefined reference to `ppp_free'
c:/users/asish/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: F:\Aspiration Energy\Heat Pump Monitoring\Hardware Development\Thermelgy tMY Future\Firmware\Development\Thermelgy Gen2 Gateway\Thermelgy_Gen2_Gateway/lib/PPPCom_Lib/PPPCom_Lib.cpp:590: undefined reference to `pppapi_set_notify_phase_callback'
c:/users/asish/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: F:\Aspiration Energy\Heat Pump Monitoring\Hardware Development\Thermelgy tMY Future\Firmware\Development\Thermelgy Gen2 Gateway\Thermelgy_Gen2_Gateway/lib/PPPCom_Lib/PPPCom_Lib.cpp:590: undefined reference to `pppapi_set_default'
c:/users/asish/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: F:\Aspiration Energy\Heat Pump Monitoring\Hardware Development\Thermelgy tMY Future\Firmware\Development\Thermelgy Gen2 Gateway\Thermelgy_Gen2_Gateway/lib/PPPCom_Lib/PPPCom_Lib.cpp:592: undefined reference to `pppapi_set_auth'
c:/users/asish/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: F:\Aspiration Energy\Heat Pump Monitoring\Hardware Development\Thermelgy tMY Future\Firmware\Development\Thermelgy Gen2 Gateway\Thermelgy_Gen2_Gateway/lib/PPPCom_Lib/PPPCom_Lib.cpp:428: undefined reference to `pppapi_connect'
[... cut ...]
This is my .ini file :

```
[platformio]
default_envs = 

[env:esp32dev]
platform = espressif32@6.3.1
board = esp32dev
framework = arduino
board_build.partitions = F:\Aspiration Energy\Heat Pump Monitoring\Hardware Development\Thermelgy tMY Future\Firmware\Development\Thermelgy Gen2 Gateway\Partition\Thermelgy_Custom_16MB.csv 
;lib_extra_dirs = C:\Users\asish\OneDrive\Documents\Arduino\libraries
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
build_flags = 

```

Any help is appreciated.

Thermelgy-Repo, 2023-06-08:
I have solve the issue, Plz check :

#https://github.com/levkovigor/ppposclient/issues/8#issuecomment-1581764405

I hope it will be helpful to someone.

kshitijgarg2609, 2023-07-17:
Hello, I actually want to enable PPPOS without changing in any sdk file if not then how can we include esp protocols in arduino library?

VojtechBartoska, 2023-10-04:
@chegewara Is this issue still valid?

chegewara, 2023-10-06:
No idea, sorry.
I am not using arduino lately and i have my own arduino build version with PPP enabled.

VojtechBartoska, 2023-12-11:
Closing this as expired, if needed you can reopen.

closed 2023-12-11 by VojtechBartoska