espressif/arduino-esp32 #7484: NOT getting rfid tag using MFRC522

beginner140BB, 2022-11-17:
### Board

ESP32

### Device Description

HARDWARE USED MFRC522

### Hardware Configuration

miso IO19
mosi IO23
sck IO18
cs IO4

### Version

other

### IDE Name

ARDUINO IDE

### Operating System

WINDOWS 10

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

I have done a customised project by making RST pin of MFRC522 to 3.3v and CS pin to IO4 of esp32 wroom.
When I checked the firmware version exampe code,it's tested okay
But the card is not reading while using the dumbinfo example code(only firmware version is printing)

### Sketch

```cpp
sample code dumb to serial is used.
```

### Debug Message

```plain
only firmware version is printing
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-11-17:
Please explain the issue regarding its relationship with ESP32 Arduino Core. 
From the short issue explanation, it seems to be an application failure, not  directly related to Arduino.

beginner140BB, 2022-11-22:
> Please explain the issue regarding its relationship with ESP32 Arduino Core. 
> From the short issue explanation, it seems to be an application failure, not  directly related to Arduino.

Its been observed that it is something related to the customized board of rfid that is created.
Not an issue of the code

beginner140BB, 2022-11-24:
@SuGlider 

we are able to read the different RFID tags multiple times using the example dumpinfo. But its not possible to read the tag after the first read while using tasks.

`
void Task1(void *pvParameters)
{
  char copy_buff[18];
  byte veh_read_block[18];
  int Vehicle_block = 4;
  char send_data[128];
  struct tm RTC_Time;
  char send_waybill[256];
 char trip_data_buff1[256];
  char fway_bill=0;
  char ftrip=0;
  char flag=0;

  for (;;)
  {
    Serial.println("Inside Task 1");
    memset(copy_buff, 0, sizeof(copy_buff));
    Serial.println("before first while");
     while ( !  mfrc522.PICC_IsNewCardPresent()) // Select one of the cards
    {
      vTaskDelay(1);
    }
    while ( ! mfrc522.PICC_ReadCardSerial()) // Select one of the cards
    {
      vTaskDelay(1);
    }

    digitalWrite(2, HIGH);
    Serial.println("Present");
    Serial.print(F("Card UID:"));
    dump_byte_UID(mfrc522.uid.uidByte, mfrc522.uid.size);
    Serial.println(c_uid);

    Serial.println();
    Serial.print(F("PICC type: "));
    MFRC522::PICC_Type piccType = mfrc522.PICC_GetType(mfrc522.uid.sak);
    Serial.println(mfrc522.PICC_GetTypeName(piccType));

    // Check for compatibility
    if (    piccType != MFRC522::PICC_TYPE_MIFARE_MINI
            &&  piccType != MFRC522::PICC_TYPE_MIFARE_1K
            &&  piccType != MFRC522::PICC_TYPE_MIFARE_4K) {
    }
    memset(&DRV_data,0,sizeof(DRV_data));
    memset(&STD_DAT,0,sizeof(STD_DAT));
    readBlock(Vehicle_block, veh_read_block);
    strcpy(DRV_data.VEH_dat,c_blkdata_VEFID);
    Serial.println(DRV_data.VEH_dat);
    strcpy(DRV_data.Route_ID,c_blkdata_RTID);
    Serial.println(DRV_data.Route_ID);                 
    strcpy(DRV_data.Trip_Type,c_blkdata_TP_UD);
    Serial.println(DRV_data.Trip_Type);
    strcpy(STD_DAT.STDID,c_blkdata_STD_ID);
    Serial.println(STD_DAT.STDID);
    mfrc522.PICC_HaltA();
    vTaskDelay(50);
  }
}

`

beginner140BB, 2022-12-09:
> 

got the solution
after reading one card to establish new card read functionality
we should call

PICC_HaltA(spi);
PCD_StopCrypto1(spi);

these two functions should called thereby oyu can establish new connection.

If there is any alternate solution.
kindly post it.

closed 2022-12-09 by beginner140BB