espressif/arduino-esp32 #7485: WiFiClientSecure: send_ssl_data() (-80) UNKNOWN ERROR CODE

Alisofin, 2022-11-17:
### Board

ESP32

### Device Description

ESP32-CAM

### Hardware Configuration

No specific configuration, no use of pins, use of OV2640 camera, no use of SD card

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

MacOs Ventura 13.0.1

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hello,
I use WiFiClientSecure library to upload an image file to a web service. The writing of HTTP request always ends up with an error (see below).

Please help !

### Sketch

```cpp
#include <Arduino.h>
#include <WiFi.h>
#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"
#include "esp_camera.h"
#include "base64.h"
#include <WiFiClientSecure.h>

// CAMERA_MODEL_AI_THINKER
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
[... 237 lines cut ...]
18:33:18.914 -> [  4701][V][ssl_client.cpp:296] start_ssl_client(): Certificate verified.
18:33:18.951 -> [  4704][V][ssl_client.cpp:311] start_ssl_client(): Free internal heap after TLS 159412
18:33:18.951 -> Connection successful!
18:33:18.951 -> [  4722][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 14 bytes...
18:33:18.951 -> [  4723][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 2 bytes...
18:33:18.951 -> POST  HTTP/1.1
18:33:18.951 -> [  4740][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 37 bytes...
18:33:18.985 -> [  4741][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 2 bytes...
18:33:18.985 -> Host: xxxxxxxxxxx
18:33:18.985 -> [  4756][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 53 bytes...
18:33:18.985 -> [  4757][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 2 bytes...
18:33:18.985 -> Authorization: Basic xxxxxxxxx
18:33:18.985 -> [  4775][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 21 bytes...
18:33:19.023 -> [  4778][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 2 bytes...
18:33:19.023 -> Content-Length: 25345
18:33:19.023 -> [  4795][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 64 bytes...
18:33:19.023 -> [  4796][V][ssl_client.cpp:364] send_ssl_data(): Writing HTTP request with 2 bytes...
18:33:19.023 -> [  4801][V][ssl_client.cpp:369] send_ssl_data(): Handling error -80
18:33:19.061 -> [  4807][E][ssl_client.cpp:37] _handle_error(): [send_ssl_data():370]: (-80) UNKNOWN ERROR CODE (0050)
18:33:19.061 -> [  4816][V][ssl_client.cpp:319] stop_ssl_socket(): Cleaning SSL connection.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mobizt, 2022-11-21:
I'm not sure is it related to your server or not? 

I test your code with my server and ESP32 core v2.0.5, and it gives the error response of invalid request instead of close connection.

To make the valid request, the line break before boundary string is required.

```cpp
 client.println("Content-Type: multipart/form-data; boundary=PapyEtLesResistances"); <-- 64 bytes of header + 2 bytes of line break.
  
 client.println(); <--- this empty line (line break) is required before sending the boundary string.

 client.print(head);  <--- I think your server may close the connection here because of missing line break before boundary string line.
```

You can test this with Postman too.

VojtechBartoska, 2023-10-23:
Hello, unfortunately I am closing this as expired due to no answers, If needed, you can reopen the issue.

closed 2023-10-23 by VojtechBartoska