espressif/arduino-esp32 #7489: Panic when tried to get bluetooth data from BLE server

robertgregor, 2022-11-18:
### Board

ESP32 Dev Module

### Device Description

Normal ESP 32 chip

### Hardware Configuration

No any connection to my TTGO board

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Win 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

I am trying to read data from BLE EQ3 radiator header. I am using the piece of code from https://github.com/pieter/ESP-EQ3.git and it's example.

### Sketch

```cpp
#include <Arduino.h>
#include <list>

#include "BLEDevice.h"
#include "eq3thermostat.h"
#include <WiFi.h>

static std::list<EQ3Thermostat> thermostats;
static EQ3Thermostat thermo = EQ3Thermostat(BLEAddress("00:1a:22:10:3c:11")); 
uint scanState = 0;

// Use this if you already know the addresses of your devices
void start() {
  thermostats.push_back(BLEAddress("00:1a:22:10:3c:11"));
  scanState = 2;
}

void startWithScan() {
  // Retrieve a Scanner and set the callback we want to use to be informed when we
  // have detected a new device.  Specify that we want active scanning and start the
  // scan to run for 5 seconds.

  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setInterval(1349);
  pBLEScan->setWindow(449);
  pBLEScan->setActiveScan(true);
  pBLEScan->start(5, [](BLEScanResults results) {
[... 185 lines cut ...]
Backtrace:0x40083675:0x3ffbeacc |<-CORRUPTED

ELF file SHA256: 0000000000000000

Seems, that the watch dog was not recet or there is endless loop????

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
```

### Other Steps to Reproduce

Based on the exception decoder, it crashed here:

Decoding stack results
0x40083675: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c line 402

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-11-21:
This is very hard to relate this issue to this ESP32 Arduino github.
The sketch uses an external BLE library,

Please narrow it down to a minimum sketch that demonstrates the issue.

VojtechBartoska, 2022-12-15:
Closing as expired, if needed you can reopen this.

closed 2022-12-15 by VojtechBartoska