espressif/arduino-esp32 #7490: Compilation error in BLE example (redefinition of 'class BLEDescriptor')

DonatelloX, 2022-11-18:
### Board

ESP32 Dev board

### Device Description

ESP32 Dev board

### Hardware Configuration

ESP32 Dev board

### Version

v2.0.5

### IDE Name

Arduino IDE 2.0.2

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

If I try to compile the BLE_server example (Example -> ESP32 BLE Arduino -> BLE_server) several compilation errors are generated.
In particular:
`[...]\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\BLE\src/BLEDescriptor.h:25:7: error: redefinition of 'class BLEDescriptor'`

### Sketch

```cpp
// Example -> ESP32 BLE Arduino -> BLE_server
/*
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleServer.cpp
    Ported to Arduino ESP32 by Evandro Copercini
    updates by chegewara
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("Long name works now");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic->setValue("Hello World says Neil");
  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}
```

### Debug Message

```plain
`[...]\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\BLE\src/BLEDescriptor.h:25:7: error: redefinition of 'class BLEDescriptor'`
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

rjrajbir, 2022-11-29:
Same issue is happening with me, in other PC it is compiling but not in my PC

DonatelloX, 2022-12-27:
Same problem also with 2.0.6:
`[...]\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\BLE\src/BLEDescriptor.h:25:7: error: redefinition of 'class BLEDescriptor'`

pschatzmann, 2023-01-11:
BLE is part of the ESP32 installation. I would expect that you get this error when you installed the ArduinoBLE library which provides the same classes

DonatelloX, 2023-02-03:
Thank you @pschatzmann.
The ArduinoBLE library in Arduino folder conflicts with ESP32 Core.

raddevus, 2023-03-20:
I just experienced the same problem. Built sketch on one computer (win10) and it worked fine using latest (2.0.4 of ArduinoIDE)
Now, just tried to build same sketch on Ubuntu 22.04 and I got the error about the redefinition of the class.
I uninstalled ArduinoBLE v1.3.3. and now the sketch compiled on Ubuntu 22.04 also.
Thanks for posting the answer. Helped greatly.

mrengineer7777, 2023-04-13:
Namespace collision on `class BLEDescriptor` between our BLE and ArduinoBLE.

closed 2025-01-16 by Parsaabasi