espressif/arduino-esp32 #7491: Custom device support with ESP32 Rainmaker and Google home

atanudasgupta, 2022-11-19:
### Board

esp232

### Device Description

esp32 devKit

### Hardware Configuration

testing with custom rainmaker device. It works well. I am able to connect with Alexa , but google Home is unable to find the rainmaker custom device
static Device my_device("TEST Geyser", "custom.device.geyser", &gpio_relay); 

### Version

v2.0.5

### IDE Name

arduino ide

### Operating System

macos

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

I want to know why google home is unable to find rainmaker custom device by alexa is able to find it\

### Sketch

```cpp
static Device my_device("TEST Geyser", "custom.device.geyser", &gpio_relay);
```

### Debug Message

```plain
google home during linking with rainmaker finds no rainmaker device
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-11-19:
https://github.com/espressif/esp-rainmaker

SuGlider, 2022-11-21:
@sanketwadekar - Any guess about it? Thanks!

sanketwadekar, 2022-11-22:
Hi @atanudasgupta
Google Home does not support custom devices but Alexa does. You can check the [list of devices](https://rainmaker.espressif.com/docs/standard-types.html#devices) supported by Alexa and GVA to find more information.

mrengineer7777, 2022-11-23:
GVA doesn't support "Other". Consider using "Outlet" instead.
[image]

VojtechBartoska, 2022-12-15:
Can I consider this as answered @atanudasgupta?

atanudasgupta, 2022-12-15:
Yes i would liked to see custom device support in rainmaker

sanketwadekar, 2022-12-22:
> @atanudasgupta - I think that @sanketwadekar  mentioned that GVA, or even GHA, doesn't support Custom Device.
> It seems that RainMaker does, as well as Alexa.
> Therefore, the issue would be with Google, not Rainmaker.
> 
> So it would be possible to have RainMaker Custom Devices with Alexa but not with Google, due to Google's limitation.
> Is that it, @sanketwadekar?

Yes, GVA doesn't support Custom devices as of now but Alexa and Rainmaker do.

VojtechBartoska, 2023-01-18:
Closing as it was already answered.

closed 2023-01-18 by VojtechBartoska