espressif/arduino-esp32 #7495: IR receiver stops receiving signal if Wi-Fi gets connected

suryanshRoy, 2022-11-20:
### Board

Esp32 Dev Module 

### Device Description

Esp32 dev module chip is cp2102

### Hardware Configuration

Gpio 35 is connected to Tsop1838 IR receiver 

### Version

v2.0.5

### IDE Name

Arduino IDE 

### Operating System

Windows 10

### Flash frequency

80mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

When I am not connected with Wi-Fi then my ESP32 Board receives signal through Tsop1838 IR receiver but when I am connected to the Wi-Fi my board stops receiving signals properly and I need to press IR Remote button around 5 to 10 times then my board receives one signal.

### Sketch

```cpp
#include "RMaker.h"
#include "WiFi.h"
#include "WiFiProv.h"
#include <IRremote.h>
#include <Preferences.h>

Preferences pref;

const char *service_name = "PROV_123";
const char *pop = "PROV";

static uint8_t wifiLed     = 2 ;   //D2

void sysProvEvent(arduino_event_t *sys_event)
{
    switch (sys_event->event_id) {      
        case ARDUINO_EVENT_PROV_START:
    #if CONFIG_IDF_TARGET_ESP32
        Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on BLE\n", service_name, pop);
        printQR(service_name, pop, "ble");
    #else
        Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on SoftAP\n", service_name, pop);
        printQR(service_name, pop, "softap");
    #endif        
      break;
    case ARDUINO_EVENT_WIFI_STA_DISCONNECTED:
        digitalWrite(wifiLed, false);
        Serial.printf("\nDisconnected from Wi-Fi\n Connecting again");
      break;
    case ARDUINO_EVENT_WIFI_STA_CONNECTED:
        Serial.printf("\nConnected to Wi-Fi!\n");
        digitalWrite(wifiLed, true);
      break;
  }
}

static uint8_t IR_RECV_PIN = 35;   //D35 to esp32
IRrecv irrecv(IR_RECV_PIN);
decode_results results;

void ir_remote()
{ 
   if (irrecv.decode(&results)) {
     irrecv.resume(); //Serial.println(results.value, HEX);
      switch(results.value){
            case 0x1FE50AF:  
            digitalWrite(RelayPin1, toggleState_1);
            toggleState_1 = !toggleState_1;
            pref.putBool("Relay1", toggleState_1);
            Serial.println("Switch 1 on - off");
            my_switch1.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_1);
            break;
            case 0x1FED827:  
            digitalWrite(RelayPin2, toggleState_2);
            toggleState_2 = !toggleState_2;
            pref.putBool("Relay2", toggleState_2);
            Serial.println("Switch 2 on - off");
            my_switch2.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_2);
            break;
            case 0x1FEF807:  
            digitalWrite(RelayPin3, toggleState_3);
          toggleState_3 = !toggleState_3;
            pref.putBool("Relay3", toggleState_3);
            Serial.println("Switch 3 on - off");
            my_switch3.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_3);
            break;
            case 0xE0E0629D:  
            digitalWrite(RelayPin4, toggleState_4);
            toggleState_4 = !toggleState_4;
            pref.putBool("Relay4", toggleState_4);
            Serial.println("Switch 4 on - off");
            my_switch4.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_4);
            break;
            case 0x1FEB04F:   
            if(toggleState_5 == 0){
              fanSpeedControl(currSpeed); //Turn ON Fan
            }
            else {
              fanSpeedControl(0); //Turn OFF Fan
            }
            toggleState_5 = !toggleState_5;
            pref.putBool("Fan_Power", toggleState_5);
            Serial.println("Switch 5 on - off");
            my_fan.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_5);
            break;
            case 0x1FE609F: 
            if(currSpeed < 4 && toggleState_5 == 1){
              currSpeed = currSpeed + 1;
              fanSpeedControl(currSpeed);
              pref.putInt("My_Speed", currSpeed);
              Serial.println("Speed up");
              my_fan.updateAndReportParam("My_Speed", currSpeed);
            }           
            break;
            case 0x1FEA05F: 
            if(currSpeed > 0 && toggleState_5 == 1){
              currSpeed = currSpeed - 1;
              fanSpeedControl(currSpeed);
              pref.putInt("My_Speed", currSpeed);
              Serial.println("Speed down");
              my_fan.updateAndReportParam("My_Speed", currSpeed);
            }       
            break;
            case 0x1FEE01F:
            all_SwitchOff();  
            break;
            case 0x1FE48B7:
            all_SwitchOn();  
            break;
            default : break;         
     }   
   } 
 }
void all_SwitchOff(){
  toggleState_1 = 0; digitalWrite(RelayPin1, LOW); pref.putBool("Relay1", toggleState_1); my_switch1.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_1);
  toggleState_2 = 0; digitalWrite(RelayPin2, LOW); pref.putBool("Relay2", toggleState_2); my_switch2.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_2); 
  toggleState_3 = 0; digitalWrite(RelayPin3, LOW); pref.putBool("Relay3", toggleState_3); my_switch3.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_3); 
  toggleState_4 = 0; digitalWrite(RelayPin4, LOW); pref.putBool("Relay4", toggleState_4); my_switch4.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_4); 
  toggleState_5 = 0; fanSpeedControl(0); pref.putBool("Fan_Power", toggleState_5); my_fan.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_5); 
}

void all_SwitchOn(){
  toggleState_1 = 1; digitalWrite(RelayPin1, HIGH); pref.putBool("Relay1", toggleState_1); my_switch1.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_1);
  toggleState_2 = 1; digitalWrite(RelayPin2, HIGH); pref.putBool("Relay2", toggleState_2); my_switch2.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_2); 
  toggleState_3 = 1; digitalWrite(RelayPin3, HIGH); pref.putBool("Relay3", toggleState_3); my_switch3.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_3);
  toggleState_4 = 1; digitalWrite(RelayPin4, HIGH); pref.putBool("Relay4", toggleState_4); my_switch4.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_4); 
  toggleState_5 = 1; fanSpeedControl(currSpeed); pref.putBool("Fan_Power", toggleState_5); my_fan.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, toggleState_5); 
}

void setup()
{
  IrReceiver.begin(IR_RECV_PIN, DISABLE_LED_FEEDBACK); // Enabling IR sensor

  pinMode(IR_RECV_PIN, INPUT);

    WiFi.onEvent(sysProvEvent);
#if CONFIG_IDF_TARGET_ESP32
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_BLE, WIFI_PROV_SCHEME_HANDLER_FREE_BTDM, WIFI_PROV_SECURITY_1, pop, service_name);
#else
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_SOFTAP, WIFI_PROV_SCHEME_HANDLER_NONE, WIFI_PROV_SECURITY_1, pop, service_name);
#endif
}

void loop()
{  
    unsigned long currentMillis = millis();
  // if WiFi is down, try reconnecting every CHECK_WIFI_TIME seconds
  if ((WiFi.status() != WL_CONNECTED) && (currentMillis - previousMillis >=interval)) {
    Serial.print(millis());
    Serial.println("Reconnecting to WiFi...");
    esp_restart();
    previousMillis = currentMillis;
  }

  // Read GPIO0 (external button to reset device
  if(digitalRead(gpio_reset) == LOW) { //Push button pressed
    Serial.printf("Reset Button Pressed!\n");
    // Key debounce handling
    delay(100);
    int startTime = millis();
    while(digitalRead(gpio_reset) == LOW) delay(50);
    int endTime = millis();

    if ((endTime - startTime) > 10000) {
      // If key pressed for more than 10secs, reset all
      Serial.printf("Reset to factory.\n");
      RMakerFactoryReset(2);
    } else if ((endTime - startTime) > 3000) {
      Serial.printf("Reset Wi-Fi.\n");
      // If key pressed for more than 3secs, but less than 10, reset Wi-Fi
      RMakerWiFiReset(2);
    } 
  }
  delay(10);

  manual_control(); //Control Switches Manualy
  fanRegularor(); //Control Fan Speed Manualy
  ir_remote(); //IR remote Control

}
```

### Debug Message

```plain
The function decode(&results)) is deprecated and may not work as expected! Just use decoded ( ) without a parameter and IrReceiver.decodedIRData.
```

### Other Steps to Reproduce

I tried it on my friend windows but I didn't work.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-11-21:
It will help a lot if you can determine how this issue is related to the ESP32 Arduino Core.
It is very hard to find an Arduino issue based on an application.

Could you narrow down to a minimum sketch that demonstrates the issue?

suryanshRoy, 2022-12-04:
In the whole code due to this line :-

IRrecv irrecv(IR_RECV_PIN); decode_results results;

it prints 

The function decode(&results)) is deprecated and may not work as expected! Just use decoded ( ) without a parameter and IrReceiver.decodedIRData.

And I also changed my Arduino core and events runs on core, 1 to 0 and 0 to 1 many times but nothing happened.

suryanshRoy, 2022-12-04:
> It will help a lot if you can determine how this issue is related to the ESP32 Arduino Core.
> It is very hard to find an Arduino issue based on an application.
> 
> Could you narrow down to a minimum sketch that demonstrates the issue?

In the whole code due to this line :-

'IRrecv irrecv(IR_RECV_PIN); decode_results results;'

it prints 

'The function decode(&results)) is deprecated and may not work as expected! Just use decoded ( ) without a parameter and IrReceiver.decodedIRData.'

And I also changed my Arduino core and events runs on core, 1 to 0 and 0 to 1 many times but nothing happened.

suryanshRoy, 2022-12-19:
@SuGlider 

I am closing this issue because today when I was working with blynk app and uploaded this IR remote code. It was not working there. But when I changed the library IRremote.h to IRremoteESP8266.h and by filling needs of IRremoteESP8266.h library it starts receiving all signals properly and when I connected my board to the wifi It was also working there and also working in ESP RAINMAKER app.
Thanks for the help.

closed 2022-12-19 by suryanshRoy