espressif/arduino-esp32 #7499: Invalid macro name if Arduino is used as a component in esp-idf and the board name contains a dash

supcik, 2022-11-20:
### Board

ttgo-lora32-v1 (and on all boards with a dash in the name)

### Device Description

Plain module. This issue is not related to a specific device.

### Hardware Configuration

This issue is not related to a specific configuration

### Version

latest master (checkout manually)

### IDE Name

VS Code

### Operating System

macOS 13.0.1

### Flash frequency

40 Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

This problem occurs when arduino-esp32 is used as a component in esp-idf (as described [here](https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html)) and when the name of the board (the "variant") contains a dash (for example  ttgo-lora32-v1).

In this case, the [CMakeList.txt](https://github.com/espressif/arduino-esp32/blob/9762b2392ad3928d8643911294e505e78fcc5b2f/CMakeLists.txt#L222) of arduino-esp32 defines the symbol `ARDUINO_${idf_target_caps}_DEV` where `idf_target_caps` is the capitalized board name (or the variant).

When the board name contains a dash (-), the symbol [is not valid](https://gcc.gnu.org/onlinedocs/gcc-12.2.0/cpp/Tokenization.html#Tokenization) and the compiler issues plenty of warnings :

``` text
<command-line>: warning: ISO C++11 requires whitespace after the macro name
```

If, inside the C++ code, you want to use conditional compilation, such as :

``` cpp
#ifdef ARDUINO_TTGO-LORA32-V1_DEV
...
#endif
```

the code still compiles, but it seems to ignore everything starting from the first dash. So it is not possible to distinguish `ARDUINO_TTGO-LORA32-V1_DEV` from `ARDUINO_TTGO-LORA32-V2_DEV`. Note that the compiler also issues a warning:

``` text
main/main.cpp:3:20: warning: extra tokens at end of #ifdef directive
 #ifdef ARDUINO_TTGO-LORA32-V1_DEV
                    ^
```

and the clang-format adds spaces around the dashes.

So to my opinion, the dashes should be replaced by underscores for the macro defined by the CMakeList.txt. Something like this:

```cmake
string(REPLACE "-" "_" idf_target_for_macro "${idf_target_caps}")
target_compile_options(${COMPONENT_TARGET} PUBLIC
    ...
    -DARDUINO_${idf_target_for_macro}_DEV
    ...
```

... but this is already part of the solution and I will provide a Pull Request for this.

### Sketch

```cpp
#include <Arduino.h>

#ifdef ARDUINO_TTGO-LORA32-V1_DEV
#define DELAY 1000
#endif

void setup() { Serial.begin(115200); }

void loop() {
    Serial.println("Hello World!");
    delay(DELAY);
}
```

### Debug Message

```plain
The error messages (warnings) are given in the description.

This warning comes from the command line because of the `-DARDUINO_TTGO-LORA32-V1_DEV` flag :

    <command-line>: warning: ISO C++11 requires whitespace after the macro name

This warning comes when you try to test for the symbol:

    main/main.cpp:3:20: warning: extra tokens at end of #ifdef directive
     #ifdef ARDUINO_TTGO-LORA32-V1_DEV
                        ^
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-11-21:
that seems interesting.... 
Let me look at that.

supcik, 2022-11-21:
I made the Pull Request #7500 with the changes described in this issue.

VojtechBartoska, 2022-11-21:
Hello @supcik, thank you for the issue and PR as well. We'll review it as soon as possible.