espressif/arduino-esp32 #7502: SD Slowing down the reading and writing process

mehrdad987, 2022-11-22:
### Board

esp32-wroom-32

### Device Description

ic esp32-wroom-32

### Hardware Configuration

static const uint8_t SS    = 5;
static const uint8_t MOSI  = 23;
static const uint8_t MISO  = 19;
static const uint8_t SCK   = 18;

### Version

other

### IDE Name

Arduino IDE

### Operating System

Windows10

### Flash frequency

80mhz

### PSRAM enabled

no

### Upload speed

512000

### Description

When the number of files in the SD card increases(800-1100), the card R/W slows down
Is there a specific reason?

### Sketch

```cpp
using 1.0.4version

void appendFile(fs::FS &fs, const char * path, const char * message) {
  File file = fs.open(path, FILE_APPEND);
    return;
  }
----
  today_filestr = ("/" + (String(datestringx)) + ".txt");//making apppend and write path
  today_file = today_filestr.c_str();
  forsd = forsdstr.c_str();
  appendFile(SD, today_file, forsd);
```

### Debug Message

```plain
It only seems that the process is sluggish after making close to 800 files 
there is no debuging messages
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-11-22:
@mehrdad987 what version of Arduino core are you using?

@P-R-O-C-H-Y Can you please help with triage? Thanks

mehrdad987, 2022-11-23:
> @mehrdad987 What version of Arduino core are you using?

esp32-wroom-32 chip/Arduino core for the ESP32
1.0.4 lib
Arduino 1.8.13 IDE

mehrdad987, 2022-11-30:
hi
It took me a week to upgrade my program and IDE :()

esp32-wroom-32 chip/Arduino core for the ESP32
2.0.5 lib
Arduino IDE 2.0.1 

and it still slowing down the process(Would it be possible for someone to check whether the writing and reading process on a SD card is slowing down?with 3000 files on SD card(the size is not important)
I will ask you to check the library if I cannot find the problem in my program
Just wanted to let you know I'm still working on it

P-R-O-C-H-Y, 2022-11-30:
Hi @mehrdad987, nice to hear you updated successfully.
Do you have any data of how much is it slowing down? Thx

mehrdad987, 2022-12-04:
how much is it slowing down was a keyword to me to find out my programs bug
There was a extra function with (while )appending to SD in my program and that was causing the SD to slow down. 
Sorry to bother you

closed 2022-12-13 by mehrdad987