espressif/arduino-esp32 #7504: If clauses dont work

borbuh, 2022-11-23:
### Board

ESP32-C3-DevKitM-1 v1.0

### Device Description

Devkit  plain module

### Hardware Configuration

Devkit connected to PC via usb and uart connected to CP2102 on pins 5 and 6

### Version

v2.0.4

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

40 MHz

### PSRAM enabled

yes

### Upload speed

460800

### Description

If the last return 0; is not commented out, if clauses start passing without condition being met. (maybe compiler is searching for next return?)
I do get a warning:
`src/cvs_ble_device_lib.cpp: In function 'uint8_t cvs_ble_check_device_flags(cvs_ble_scan_device_t*, uint8_t*)':
src/cvs_ble_device_lib.cpp:70:1: warning: control reaches end of non-void function [-Wreturn-type]
 }`

Example: 
device->pair_mac[0] is 0x62
myMac[0] is 0x10
device->flags is 0x00

If i create local variables in function eg.
`uint8_t temp1 =1;
uint8_t temp2 = 2;`
and compare them in if clause, it doestn pass as it should.

### Sketch

```cpp
uint8_t cvs_ble_check_device_flags(cvs_ble_scan_device_t *device, uint8_t *myMac)
{
    if (strcmp(device->name, BT42_NAME) == 0)
    {
        if (memcmp(device->pair_mac, myMac, 6) == 0)
        {
            if ((device->flags & 0x03) != 0)
            {
                Serial.println("Mine bt42 found");
                return 1; // valid device to manage
            }
        }
    }
    else if (strcmp(device->name, BT43_NAME) == 0)
    {
        Serial.printf("BT43 device %02X myMac %02X, flags: %02X\r\n", device->pair_mac[0], myMac[0], device->flags);

        if (myMac[0] == device->pair_mac[0])
        {
            Serial.println("Its my BT43");
            if ((device->flags & 0x03) != 0)
            {
                Serial.printf("Flags for download!\r\n");
                return 1; // valid device to manage
            }
        }
    }
    else
    {
        if ((memcmp(dfu_dev->dfu_mac, myMac, 6) == 0) || (memcmp(blec_send_target_instance.addr, myMac, 6) == 0)) // preveri že je trenutni mac shranjen za DFU
        {
            // we have NUS data for device or device needs DFU
            Serial.println("DFu target of nus target found");
            return 1;
        }

        return 0;
    }
    // return 0;
}
```

### Debug Message

```plain
[16:08:48.428] BT43 device 62 myMac 10, flags: 00

[16:08:48.430] Its my BT43

[16:08:48.433] Flags for download!
```

### Other Steps to Reproduce

Can try on other ESP32 tomorrow

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

kapyaar, 2022-11-24:
This is so weird. I spent 3 days figuring out why my if statement is working, and did not want to post  an issue saying my if statement is working. In my case, I had to put a return at the very end, in a bool function, or else the if will simply go through without regard for the condition check. No compile errors.

JiriBilek, 2022-11-24:
If I understand it correctly, it looks like a compiler bug.
Then I'd try to:
1. **Change compiler optimization level and see what happens.**
    Try to temporarily edit and change -Os to -O0 compiler flag in platform.txt file. For esp32c3 there are 3 occurrences. You can also try -O1.

or

2. **Get the assembler output for the function in question and check it for correctness.**
    The best choice would be to extract the function to a separate file (the output assembler file would be smaller) and then compile it with -S option. For the compilation I'd recommend to find the line where is the compiler invoked for this particular file in the compiler verbose output (in Arduino IDE there is an option in Preferences), then edit the line and run it in the command prompt separately. You don't need to link or compile the whole app.

mrengineer7777, 2023-05-21:
Any updates on this?

VojtechBartoska, 2023-08-21:
Closing as expired, if needed you can reopen.