espressif/arduino-esp32 #7505: Possible regression of serial communication performance since version 2.x

awawa-dev, 2022-11-23:
### Board

ESP32 dev module

### Device Description

Board capable of 2Mb serial speed (CH9102x)

### Hardware Configuration

nothing  is connected

### Version

v2.0.5

### IDE Name

Arduino IDE, PlatformIO

### Operating System

Windows, Linux

### Flash frequency

default

### PSRAM enabled

no

### Upload speed

default

### Description

As of version 2.x I observe possible performance regression compared to version 1.x when high-speed communication is used (2 000 000 baud). We verify the integrity of the frame data to make sure the frame is complete and undamaged.

Simplified sketch and python test client provided as an attachment.  'Esp32 dev module' with CH9102x 

With version 1.0.6 was working fine.
[image]

Sending 900 bytes frames without any problem to ESP32 board. The reception is confirmed.

[image]

But from version 2.x the same sketch does not work, which was first reported by our users and then confirmed by us after testing. We are unable to maintain such communication, all 900 byte frames are damaged or incomplete:

[image]

None of 900 byte frames is receive undamaged and completed.

[image]

It starts to working around 600 bytes but with high error rate:

[image]
[client_and_sketch.zip](https://github.com/espressif/arduino-esp32/files/10077331/client_and_sketch.zip)

### Sketch

```cpp
Test client and sketch provided as an attachment in the description section.
```

### Debug Message

```plain
not applicable since it's not a bug
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-11-25:
Hello @awawa-dev, thanks for the issue.

I'm tagging @SuGlider who did bunch of improvements to Serial recently.

SuGlider, 2022-11-26:
@awawa-dev - Please turn on Log messages (Core Debug Level: "Verbose") in the Arduino IDE.
You shall see this message:
`RX Buffer can't be resized when Serial is already running.`

Change your `setup()` to:
``` cpp
void setup()
{
	Serial.setRxBufferSize(MAX_BUFFER);  // before begin()
	Serial.begin(COM_SPEED);
	
	Serial.setTimeout(50);
        Serial.write("Hello tester!");
}

```

awawa-dev, 2022-11-26:
So 2.x caused that breaking change? Still it doesn't resolve the issue.
It's better for 900 byte frame but still experience degraded performance and very high error rate:
[image]

For comparison 1.0.6 could handle even 1800 byte frames:

[image]

SuGlider, 2022-11-26:
For 900, the solution is to replace `Stream::readBytes()` with `HardwareSerial::read()`

``` cpp
  uint16_t internalIndex = min(Serial.available(), MAX_BUFFER);
  if (internalIndex > 0)
  {
    lastDataTime = millis();
//    internalIndex = Serial.readBytes(buffer, internalIndex);
    internalIndex = Serial.read(buffer, internalIndex);
  }
  else if (lastDataTime + 1000 < millis())
  {
    Serial.println(framesReceived);
    lastDataTime = millis();
    framesReceived = 0;
    state = TestProtocol::HEADER_A;
  }
```

SuGlider, 2022-11-26:
Anyway, it doesn't work well with 1800. Let me analyze it further.

awawa-dev, 2022-11-26:
Thank you for your time and effort. I can confirm that it works for 1200 bytes frames now with usually no errors. Then it gets worse quickly if the frame is only slightly bigger.

SuGlider, 2022-11-27:
@awawa-dev 
PR #7525 allows the sketch presented in this issue to run perfectly fine with no errors.
I tested it with 5000 as packet size.

awawa-dev, 2022-11-27:
Great! I will test it right away.

awawa-dev, 2022-11-27:
@SuGlider Great job 😄  I've tested your PR not only using test sketch but also my main project: I confirm everything works now perfectly. Hope this PR will be merged soon. And that along with the changes you recommended here will finally allow me to upgrade my ESP32 LED driver to Arduino ESP32 ver. 2. Because of that performance issue, it was on hold for almost a year (I'm aware that it could be difficult to detect due specific serial chip capability and large frame/speed transmission configuration)

SuGlider, 2022-11-27:
Excelent!

SuGlider, 2022-12-04:
Added new change to override `Stream:readBytes()`. Now the original code using this Stream function runs correctly as well.

awawa-dev, 2022-12-04:
Thank you for the update. Just for your information, as I just received the ESP2-S2 mini lolin: it seems that the problem described here does not apply to this board, possibly due to the integrated onboard UART instead of the external one.

SuGlider, 2022-12-04:
> Thank you for the update. Just for your information, as I just received the ESP2-S2 mini lolin: it seems that the problem described here does not apply to this board, possibly due to the integrated onboard USB-UART instead of the external one.

Yes, but in that case the ESP32-S2 mini lolin uses USB CDC instead of UART. Therefore, it doesn't use `HardwareSerial` class, but `HWCDC`  instead. Those two have completely different implementations.

awawa-dev, 2022-12-04:
> Yes, but in that case the ESP32-S2 mini lolin uses USB CDC instead of UART. 

Is there any preprocessor `defined` based on the board model (platformio) that could tell us, if we compile a firmware for CDC or for UART based Serial? Seems there are more differences like the buffer size (CDC could return much larger chunk of data than UART).

SuGlider, 2022-12-05:
>Is there any preprocessor defined based on the board model (platformio) that could tell us, if we compile a firmware for CDC or for UART based Serial? Seems there are more differences like the buffer size (CDC could return much larger chunk of data than UART).

Yes, it is in the source files:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.h#L192-L207
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp#L99-L103
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HWCDC.h#L101-L107
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HWCDC.cpp#L384-L392

`ARDUINO_USB_CDC_ON_BOOT` is set in the IDE Board menu and this is set in the building process.

SuGlider, 2022-12-05:
For the ESp32-S2 mini lolin, it is set by default in the Board menu and setup in the `boards.txt` file:

https://github.com/espressif/arduino-esp32/blob/master/boards.txt#L5583

awawa-dev, 2022-12-05:
Thank you very much. Now I can handle all the cases for the different board variants I have.

closed 2022-12-13 by me-no-dev