espressif/arduino-esp32 #7506: Lilon32 (ESP-wroom-32) Rmakerswitch example cannot able to complied

saadnasim83, 2022-11-24:
### Board

Lilon32 

### Device Description

This LOLIN32 ESP32 Board comes with an on board TP4054 based lithium battery charger circuit. This Board therefore is great board for battery powered applications.

ESP32 has integrated antenna and RF balun, power amplifier, low-noise amplifiers, filters, and power management module. The entire solution takes up the least amount of printed circuit board area.This board is used with 2.4 GHz dual-mode Wi-Fi and Bluetooth chips by TSMC 40nm low power technology, best power and RF properties, safe, reliable, and scalable to a variety of applications.

The ESP32 doesn’t replace the ESP8266, but it does improve on it in every aspect. Not only does it have WiFi support, but it also features a Bluetooth 4.2 radio, making it even more versatile. The CPU is similar to the ESP8266 – it’s a 32-bit Xtensa® LX6, but the ESP32 has two cores! There’s also 128KB of ROM and 416KB SRAM, but Flash memory (for program and data storage) is still left up to an external chip (up to 64MB).

### Hardware Configuration

Lithium battery interface, 500mA Max charging current
WIFI
bluetooth
4MB Flash
High performance-price ratio
Small volume,easily embeded to other products
Strong function with support LWIP protocol,Freertos
Suppporting three modes: AP, STA, and AP+STA
Supporting Lua program, easily to develop

### Version

other

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows10

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

921600

### Description

I get following error
**Using library SimpleBLE at version 2.0.0 in folder: C:\Users\win10\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.5\libraries\SimpleBLE 
"C:\\Users\\win10\\Documents\\ArduinoData\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\gcc8_4_0-esp-2021r2-patch3/bin/xtensa-esp32-elf-size" -A "C:\\Users\\win10\\AppData\\Local\\Temp\\arduino_build_996032/RMakerSwitch.ino.elf"
Sketch uses 1786697 bytes (136%) of program storage space. Maximum is 1310720 bytes.text section exceeds available space in board

Global variables use 52988 bytes (16%) of dynamic memory, leaving 274692 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
Error compiling for board SparkFun ESP32 Thing.**

### Sketch

```cpp
//This example demonstrates the ESP RainMaker with a standard Switch device.
#include "RMaker.h"
#include "WiFi.h"
#include "WiFiProv.h"

#define bbsize 109100
byte BigBuffer[bbsize] = {0};

#define DEFAULT_POWER_MODE true
const char *service_name = "PROV_1234";
const char *pop = "abcd1234";

//GPIO for push button
#if CONFIG_IDF_TARGET_ESP32C3
static int gpio_0 = 9;
static int gpio_switch = 7;
#else
//GPIO for virtual device
static int gpio_0 = 0;
static int gpio_switch = 16;
#endif

/* Variable for reading pin status*/
bool switch_state = true;

//The framework provides some standard device types like switch, lightbulb, fan, temperaturesensor.
static Switch my_switch;

void sysProvEvent(arduino_event_t *sys_event)
{
    switch (sys_event->event_id) {
        case ARDUINO_EVENT_PROV_START:
#if CONFIG_IDF_TARGET_ESP32S2
        Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on SoftAP\n", service_name, pop);
        printQR(service_name, pop, "softap");
#else
        Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on BLE\n", service_name, pop);
        printQR(service_name, pop, "ble");
#endif
        break;
        default:;
    }
}

void write_callback(Device *device, Param *param, const param_val_t val, void *priv_data, write_ctx_t *ctx)
{
    const char *device_name = device->getDeviceName();
    const char *param_name = param->getParamName();

    if(strcmp(param_name, "Power") == 0) {
        Serial.printf("Received value = %s for %s - %s\n", val.val.b? "true" : "false", device_name, param_name);
        switch_state = val.val.b;
        (switch_state == false) ? digitalWrite(gpio_switch, LOW) : digitalWrite(gpio_switch, HIGH);
        param->updateAndReport(val);
    }
}

void setup()
{
    Serial.begin(115200);
    pinMode(gpio_0, INPUT);
    pinMode(gpio_switch, OUTPUT);
    digitalWrite(gpio_switch, DEFAULT_POWER_MODE);

    Node my_node;
    my_node = RMaker.initNode("ESP RainMaker Node");

    //Initialize switch device
    my_switch = Switch("Switch", &gpio_switch);

    //Standard switch device
    my_switch.addCb(write_callback);

    //Add switch device to the node
    my_node.addDevice(my_switch);

    //This is optional
    RMaker.enableOTA(OTA_USING_PARAMS);
    //If you want to enable scheduling, set time zone for your region using setTimeZone().
    //The list of available values are provided here https://rainmaker.espressif.com/docs/time-service.html
    // RMaker.setTimeZone("Asia/Shanghai");
    // Alternatively, enable the Timezone service and let the phone apps set the appropriate timezone
    RMaker.enableTZService();

    RMaker.enableSchedule();

    RMaker.start();

    WiFi.onEvent(sysProvEvent);
#if CONFIG_IDF_TARGET_ESP32S2
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_SOFTAP, WIFI_PROV_SCHEME_HANDLER_NONE, WIFI_PROV_SECURITY_1, pop, service_name);
#else
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_BLE, WIFI_PROV_SCHEME_HANDLER_FREE_BTDM, WIFI_PROV_SECURITY_1, pop, service_name);
#endif
}

void loop()
{
    if(digitalRead(gpio_0) == LOW) { //Push button pressed

        // Key debounce handling
        delay(100);
        int startTime = millis();
        while(digitalRead(gpio_0) == LOW) delay(50);
        int endTime = millis();

        if ((endTime - startTime) > 10000) {
          // If key pressed for more than 10secs, reset all
          Serial.printf("Reset to factory.\n");
          RMakerFactoryReset(2);
        } else if ((endTime - startTime) > 3000) {
          Serial.printf("Reset Wi-Fi.\n");
          // If key pressed for more than 3secs, but less than 10, reset Wi-Fi
          RMakerWiFiReset(2);
        } else {
          // Toggle device state
          switch_state = !switch_state;
          Serial.printf("Toggle State to %s.\n", switch_state ? "true" : "false");
          my_switch.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, switch_state);
          (switch_state == false) ? digitalWrite(gpio_switch, LOW) : digitalWrite(gpio_switch, HIGH);
      }
    }
    delay(100);
}
```

### Debug Message

```plain
sing library SimpleBLE at version 2.0.0 in folder: C:\Users\win10\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.5\libraries\SimpleBLE 
"C:\\Users\\win10\\Documents\\ArduinoData\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\gcc8_4_0-esp-2021r2-patch3/bin/xtensa-esp32-elf-size" -A "C:\\Users\\win10\\AppData\\Local\\Temp\\arduino_build_996032/RMakerSwitch.ino.elf"
Sketch uses 1786697 bytes (136%) of program storage space. Maximum is 1310720 bytes.text section exceeds available space in board

Global variables use 52988 bytes (16%) of dynamic memory, leaving 274692 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
Error compiling for board SparkFun ESP32 Thing.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-11-28:
I guess that you shall select the correct Partition Scheme in the Arduino IDE Board Menu: RainMaker.

[image]

closed 2022-12-08 by SuGlider

coding-ram, 2023-03-09:
Hello , I am facing the error issue after uploading. Please check the screenshot below.  The Serial Monitor also gives a big error. Please help me out.......
[image]
[image]
As soon as I provide the Wifi credentials , the app shows the following error.
[image]