espressif/arduino-esp32 #7507: esp32 boards(esp32s2, esp32c3, esp32s3) doesn't reset after upload

zeynepdicle, 2022-11-24:
### Board

ESP32 Boards

### Device Description

ESP32-C3-DevKitC, ESP32-S3-DevKitC, Deneyap Mini and Deneyap Kart G

### Hardware Configuration

Native USB port plugged into computer, serial convert USB port not plugged in.

### Version

v2.0.5

### IDE Name

Arduino IDE 2.0.2, Arduino IDE 2.0.1, Arduino IDE 2.0.0, Arduino IDE 1.8.19, 

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I reset my ESP boards(without usb to uart), then I can't upload. I boot manually but then reset it again so I always boot manually.

### Sketch

```cpp
any sketch
```

### Debug Message

```plain
1. 
Traceback (most recent call last):
  File "esptool.py", line 34, in <module>
  File "esptool\__init__.py", line 1004, in _main
  File "esptool\__init__.py", line 646, in main
  File "esptool\__init__.py", line 895, in get_default_connected_device
  File "esptool\loader.py", line 631, in connect
  File "esptool\loader.py", line 572, in _connect_attempt
  File "esptool\loader.py", line 435, in sync
  File "esptool\loader.py", line 365, in command
  File "esptool\loader.py", line 317, in write
  File "serial\serialwin32.py", line 325, in write
serial.serialutil.SerialTimeoutException: Write timeout
[7332] Failed to execute script 'esptool'
 due to unhandled exception!
Failed uploading: uploading error: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-11-24:
When you manually boot into download mode to upload over the native USB, the chip would not be reset into the firmware after flashing. When letting the USB reboot into Download, it behaves the same as when over UART. This limitation is outside of Arduino and ESP-IDF.

Also make sure that your USB cable is good quality.

closed 2022-11-24 by me-no-dev

contractorwolf, 2024-06-10:
@me-no-dev I know this is 2 years old, but I just hit this same issue on my Lilygo T-Deck and can figure out how to break out of it.  I am not 100% on what your response is suggesting might fix this issue.  I have got it into boot mode by clicking the boot button and the reset button (screen is off).  It compiles and starts to upload and then bails with this error each time.  I have tried the straight deploy and it appears to drop the com port as it uploads.  On know what else to try.

me-no-dev, 2024-06-11:
@contractorwolf you could try to use the serial port that is towards the top of the device to upload that way. In any case, this is not expected behavior. Could it be that cable is not great and device is losing power or something? Once in download mode, the chip runs from ROM, so there is no user software doing the work at all.