espressif/arduino-esp32 #7510: ESP32-D0WD-V3 does not eneter deep sleep with 2.x.x library

BogdanDIA, 2022-11-24:
### Board

OLIMEX-ESP32-DevKit-LiPo

### Device Description

The board is OLIMEX-ESP32-DevKit-LiPo and has ESP32-WROOM-32E module which esptool reports as ESP32-D0WD-V3.

### Hardware Configuration

No additional HW attached to the pins

### Version

v2.0.5

### IDE Name

Arduino 1.8.19

### Operating System

Ubuntu 20.04

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

If Touch wake is used, the processor will not enter deep sleep and will reboot continuously.

The problem is happening with version all versions of Arduino library from 2.0.0 to 2.0.5.

The problem does not occur with version 1.0.6 of the Arduino library.

### Sketch

```cpp
// The default TouchWakeUp example for ESP32 can be used

#define Threshold 40 /* Greater the value, more the sensitivity */

RTC_DATA_ATTR int bootCount = 0;
touch_pad_t touchPin;
/*
Method to print the reason by which ESP32
has been awaken from sleep
*/

void print_wakeup_reason(){
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case ESP_SLEEP_WAKEUP_EXT0 : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1 : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER : Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.printf("Wakeup was not caused by deep sleep: %d\n",wakeup_reason); break;
  }
}

/*
Method to print the touchpad by which ESP32
has been awaken from sleep
*/
void print_wakeup_touchpad(){
  touchPin = esp_sleep_get_touchpad_wakeup_status();

  switch(touchPin)
  {
    case 0  : Serial.println("Touch detected on GPIO 4"); break;
    case 1  : Serial.println("Touch detected on GPIO 0"); break;
    case 2  : Serial.println("Touch detected on GPIO 2"); break;
    case 3  : Serial.println("Touch detected on GPIO 15"); break;
    case 4  : Serial.println("Touch detected on GPIO 13"); break;
    case 5  : Serial.println("Touch detected on GPIO 12"); break;
    case 6  : Serial.println("Touch detected on GPIO 14"); break;
    case 7  : Serial.println("Touch detected on GPIO 27"); break;
    case 8  : Serial.println("Touch detected on GPIO 33"); break;
    case 9  : Serial.println("Touch detected on GPIO 32"); break;
    default : Serial.println("Wakeup not by touchpad"); break;
  }
}

void callback(){
  //placeholder callback function
  Serial.println("In callback");
}

void setup(){
  Serial.begin(9600);
  delay(1000); //Take some time to open up the Serial Monitor

  //Increment boot number and print it every reboot
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));

  //Print the wakeup reason for ESP32 and touchpad too
  print_wakeup_reason();
  print_wakeup_touchpad();

  Serial.print("wakeup: ");
  Serial.println(esp_sleep_is_valid_wakeup_gpio((gpio_num_t)T3));

  //Setup interrupt on Touch Pad 3 (GPIO15)
  touchAttachInterrupt(T3, callback, Threshold);

  //Configure Touchpad as wakeup source
  esp_sleep_enable_touchpad_wakeup();
  //esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);

  //Go to sleep now
  Serial.println("Going to sleep now");
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop(){
  //This will never be reached
}
```

### Debug Message

```plain
23:54:40.050 -> ⸮}⸮u 1⸮1⸮⸮Boot number: 1
23:54:40.050 -> Wakeup was not c⸮⸮⸮eL0NJ!!!⸮⸮)⸮1⸮!⸮⸮Boot number: 1
23:54:41.624 -> Wakeup was not caused by deep sleep: 0
23:54:41.657 -> Wakeup not by touchpad
23:54:41.690 -> wakeup: 1
23:54:41.690 -> Going to sleep now
23:54:42.288 -> ⸮⸮⸮
```

### Other Steps to Reproduce

I can reproduce the same issue with ESP32-PICO-V3 and library 2.0.5

Interesting is that with a board having module ESP32-WROOM-32 (reported by esptool as ESP32-D0WDQ6 Rev1) the issue does not reproduce.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-11-25:
@P-R-O-C-H-Y Can you please help with triage?

BogdanDIA, 2022-12-01:
Hi guys,

any fix for this issue?

Thanks

closed 2025-01-16 by Parsaabasi