espressif/arduino-esp32 #7515: Fatal Error with http_parser.h

CSIAUD, 2022-11-25:
### Board

WEMOS D1 mini

### Device Description

Just the arduino and a mosfet.

### Hardware Configuration

i don't know

### Version

other

### IDE Name

Arduino IDE

### Operating System

windows 11

### Flash frequency

i don't know

### PSRAM enabled

no

### Upload speed

921600

### Description

I couldn't find the WebServer library in the management, so I downloaded it and want to compile/upload, I got the 'No such file or directory' error.

### Sketch

```cpp
#include <WebServer.h>

WebServer monWebServeur(80);

void handleRoot() {
  MYDEBUG_PRINTLN("-WEBSERVER : requete root");

  String out = "";
  out += "<html><head>";
  out += "<title>TEST - PC Power</title>";
  out += "<style>body { background-color: #cccccc; font-family: Arial, Helvetica, Sans-Serif; Color: #000088; }</style>";
  out += "</head><body>";
  out += "<h1>Hello World !!</h1><br>";
  out += "</body></html>";
  monWebServeur.send(200, "text/html", out);
}
```

### Debug Message

```plain
In file included from c:\Users\csiau\Documents\Arduino\libraries\WebServer\src/WebServer.h:30,
                 from C:\Users\csiau\Documents\arduino\powerCase\main\MyWebServer.h:32,
                 from C:\Users\csiau\Documents\arduino\powerCase\main\main.ino:3:
c:\Users\csiau\Documents\Arduino\libraries\WebServer\src/HTTP_Method.h:4:10: fatal error: http_parser.h: No such file or directory
    4 | #include "http_parser.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-11-25:
WebServer is included in the board files.  The headers and examples are at https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/ for correct usage.

CSIAUD, 2022-11-25:
I know but even with the example files, the problem remains, the error comes from one of the library files : 
[image]

I have also tested with an example file directly and cannot compile.

lbernstone, 2022-11-25:
Please fill out the information in the template. Flash frequency won't be important, but the rest is needed.

CSIAUD, 2022-11-25:
> Please fill out the information in the template. Flash frequency won't be important, but the rest is needed.

Like that ?

> ### Board
> 
> WEMOS D1 mini
> ### Device Description
> 
> Just the arduino and a mosfet.
> ### Hardware Configuration
> 
> Microcontroller: ESP8266
> 
> Frequency: 80MHz
> 
> Flash: 4MB
> 
> RAM: 80KB
> ### Version
> 
> other
> ### IDE Name
> 
> Arduino IDE
> ### Operating System
> 
> windows 11
> ### Flash frequency
> 
> i don't know
> ### PSRAM enabled
> 
> no
> ### Upload speed
> 
> 921600
> ### Description
> 
> I couldn't find the WebServer library in the management, so I downloaded it and want to compile/upload, I got the 'No such file or directory' error.
> ### Sketch
> 
> ```c++
> #include <WebServer.h>
> 
> WebServer monWebServeur(80);
> 
> void handleRoot() {
>   MYDEBUG_PRINTLN("-WEBSERVER : requete root");
> 
>   String out = "";
>   out += "<html><head>";
>   out += "<title>TEST - PC Power</title>";
>   out += "<style>body { background-color: #cccccc; font-family: Arial, Helvetica, Sans-Serif; Color: #000088; }</style>";
>   out += "</head><body>";
>   out += "<h1>Hello World !!</h1><br>";
>   out += "</body></html>";
>   monWebServeur.send(200, "text/html", out);
> }
> ```
> 
> ### Debug Message
> 
> ```
> In file included from c:\Users\csiau\Documents\Arduino\libraries\WebServer\src/WebServer.h:30,
>                  from C:\Users\csiau\Documents\arduino\powerCase\main\MyWebServer.h:32,
>                  from C:\Users\csiau\Documents\arduino\powerCase\main\main.ino:3:
> c:\Users\csiau\Documents\Arduino\libraries\WebServer\src/HTTP_Method.h:4:10: fatal error: http_parser.h: No such file or directory
>     4 | #include "http_parser.h"
>       |          ^~~~~~~~~~~~~~~
> compilation terminated.
> 
> exit status 1
> 
> Compilation error: exit status 1
> ```
> 
> ### Other Steps to Reproduce
> 
> _No response_
> ### I have checked existing issues, online documentation and the Troubleshooting Guide
> 
>     * [x]  I confirm I have checked existing issues, online documentation and Troubleshooting guide.

This is the version of what ?

lbernstone, 2022-11-26:
This repo is for esp32.  https://github.com/esp8266/Arduino

VojtechBartoska, 2022-11-28:
@CSIAUD Your board is based on ESP8266. Can you please open your issue in mentioned repository above by @lbernstone?

If yes, let me know and we can close this issue. Thanks.

closed 2022-11-28 by CSIAUD