espressif/arduino-esp32 #7516: 'esp32_adc2gpio' was not declared while compiling for um_tinys3

JPZV, 2022-11-26:
### Board

Unexpected Maker TinyS3 (AKA: TinyS3)

### Device Description

N/A

### Hardware Configuration

N/A

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Everytime I try to compile for the TinyS3 using PlatformIO, it fails saying that `error: 'esp32_adc2gpio' was not declared in this scope`.

The only workaround for that is to edit manually [variants/um_tinys3/pins_arduino.h](https://github.com/espressif/arduino-esp32/blob/master/variants/um_tinys3/pins_arduino.h#L16) commenting out the line 16 (`#define analogInputToDigitalPin(p)  (((p)<20)?(esp32_adc2gpio[(p)]):-1)`) and add `#define analogInputToDigitalPin(p) (((p)<20)?(analogChannelToDigitalPin(p)):-1)` just below of it

Please refer to https://github.com/Aircoookie/WLED/pull/2905#issuecomment-1328049860 for more info

### Sketch

```cpp
//sketch.ino:
#include <Arduino.h>

void setup() { }
void loop() { }

//platformio.ini:
[env:esp32s3dev_8MB_PSRAM]
;; ESP32-TinyS3 development board, with 8MB FLASH and 8MB PSRAM (memory_type = qspi_opi)
board = um_tinys3
platform = espressif32@5.2.0
platform_packages =
upload_speed = 921600
build_unflags = ${common.build_unflags}
build_flags =  ${common.build_flags} ${esp32s3.build_flags} 
  -D CONFIG_LITTLEFS_FOR_IDF_3_2 -D WLED_WATCHDOG_TIMEOUT=0 -D ARDUINO_USB_MODE=1 -D ARDUINO_USB_CDC_ON_BOOT=0 -D ARDUINO_USB_MSC_ON_BOOT=0
  -D WLED_USE_PSRAM
lib_deps = ${esp32s3.lib_deps}
board_build.partitions = tools/WLED_ESP32_8MB.csv
board_build.f_flash = 80000000L
board_build.flash_mode = qio
monitor_filters = esp32_exception_decoder
```

### Debug Message

```plain
C:/Users/juanp/.platformio/packages/framework-arduinoespressif32/variants/um_tinys3/pins_arduino.h:16:48: error: 'esp32_adc2gpio' was not declared in this scope
 #define analogInputToDigitalPin(p)  (((p)<20)?(esp32_adc2gpio[(p)]):-1)
wled00/src/dependencies/blynk/Blynk/BlynkApiArduino.h:70:62: note: in expansion of macro 'analogInputToDigitalPin'
     #define BLYNK_DECODE_PIN(it) (((it).asStr()[0] == 'A') ? analogInputToDigitalPin(atoi((it).asStr()+1)) : (it).asInt())
                                                              ^~~~~~~~~~~~~~~~~~~~~~~
wled00/src/dependencies/blynk/Blynk/BlynkApiArduino.h:93:19: note: in expansion of macro 'BLYNK_DECODE_PIN'
     uint8_t pin = BLYNK_DECODE_PIN(it);
```

### Other Steps to Reproduce

You can fork https://github.com/Aircoookie/WLED and replace the [env:esp32s3dev_8MB_PSRAM] from that project with the above one

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

softhack007, 2022-11-26:
It seems that several -S3 variants are affected by the same problem, like `esp32s3box`, `esp32s3camlcd`, `Bee_S3`, `um_feathers3`, `um_pros3`.

SuGlider, 2022-11-28:
PR #7526 shall fix this issue.

PR #7526 (2022-12-13): fixes non existent esp32_adc2gpio
## Description of Change
Replaces non existant `esp32_adc2gpio()` with `analogChannelToDigitalPin()` in all the necessary "pins_arduino.h" files.

## Tests scenarios
Just a compilation test with the ESP32/ESP32S2/ESP32S3/ESP32C3 variants.

## Related links
Fixes #7516
closed 2022-12-13 by me-no-dev