espressif/arduino-esp32 #7518: SD_MMC

amirjak0, 2022-11-26:
### Related area

SD_MMC

### Hardware specification

Esp32

### Is your feature request related to a problem?

If possible in SD_MMC examples
add test for find file sort by name Alphabet.

### Describe the solution you'd like

_

### Describe alternatives you've considered

If possible in SD_MMC examples
add test for find file sort by name Alphabet.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2022-11-26:
Nobody is going to do your homework for you.  If you would like to add this feature, then submit a pull request.

amirjak0, 2022-11-27:
Thanks for your response

closed 2022-11-27 by amirjak0

VojtechBartoska, 2022-11-28:
Hello @amirjak0. for SD_MMC examples, please check available examples here: https://github.com/espressif/arduino-esp32/tree/master/libraries/SD_MMC/examples

Feel free to contribute to this repository, we really appreciate it.

---

Hello @lbernstone, we really appreciate your knowledge and help with issues in this repository. Please, just keep in mind that a lot of contributors here are newcomers and they are opening their very first issues or Pull Requests, so basically patience and some overview is needed and very appreciated. We would like to have this repository in polite and friendly atmosphere, so please consider this as well. Thank you.