espressif/arduino-esp32 #7521: Provide getSdkVersion()-like function for arduino-esp32 version

drws, 2022-11-27:
### Related area

General (misc.) functions

### Hardware specification

/

### Is your feature request related to a problem?

[`Esp.h`](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Esp.h) provides a runtime function `getSdkVersion()`, a wrapper for `esp-idf`'s `esp_get_idf_version()`, which reads version data from [`esp_idf_version.h`](https://github.com/espressif/esp-idf/blob/master/components/esp_common/include/esp_idf_version.h) containing `ESP_IDF_VERSION_*` macros.

A similar story is `arduino-esp32` version where [`esp_arduino_version.h`](https://github.com/espressif/arduino-esp32/commits/master/cores/esp32/esp_arduino_version.h) provides `ESP_ARDUINO_VERSION_*` macros. However, the `arduino-esp32` does not contain `getSdkVersion()`-like function to obtain `arduino-esp32` version string.

### Describe the solution you'd like

A `getSdk2Version()` function that (similarly to `getSdkVersion()`) returns a string containing `ESP_ARDUINO_VERSION_*` data.

This would also make `arduino-esp32` version as accessible as `esp-idf` one is, without needing to include `esp_arduino_version.h`. Such basic debugging information should probably be most easily accessible.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

mrengineer7777, 2022-11-28:
We found the IDF getSdkVersion() didn't work for us as it can include text for beta versions.  This is what we are currently using for one of our projects:

```
String ProdVer;

//Returns comprehensive product version info.  String is created once in life as we call this function a lot.
String MF_ProdVersion(void) {
    if(ProdVer.length()>0) return ProdVer;

    char buf[128];
    char *p;

    p = buf;
    p += sprintf(p, "%sr%s, ", PROD_MODEL, PROD_VERSION);
    p += sprintf(p, "ARD %u.%u.%u, ", ESP_ARDUINO_VERSION_MAJOR, ESP_ARDUINO_VERSION_MINOR, ESP_ARDUINO_VERSION_PATCH);
    p += sprintf(p, "IDF %u.%u.%u, ", ESP_IDF_VERSION_MAJOR, ESP_IDF_VERSION_MINOR, ESP_IDF_VERSION_PATCH);
    p += sprintf(p, "%sr%u", ESP.getChipModel(), ESP.getChipRevision() );
    ProdVer = buf;
    
    return ProdVer;
}
```

drws, 2022-12-02:
If it's not a release build of either `esp-idf` or `arduino-esp32` it is desired for `getSdkVersion()` (or `getSdk2Version()`) to include the relevant build tag.

Parsaabasi, 2025-01-16:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

closed 2025-01-16 by Parsaabasi

drws, 2025-01-26:
@Parsaabasi This is still relevant, but it seems I can't reopen it, the button is greyed out.

Somebody with access please reopen the issue.