espressif/arduino-esp32 #7522: Feasibility of implementing WiFiServerSecure in v2.* using eps-tls ESP-IDF component?

michkot, 2022-11-27:
### Related area

WiFi, TCP server, TLS, mbedlts

### Hardware specification

-

### Is your feature request related to a problem?

(Hi folks, )

I am missing WiFiSecureServer which I'd like to make secure-websokcets work (https://github.com/Links2004/arduinoWebSockets can be modified to work as WSS server, simply swapping WiFiServer with WifiServerSecure on ESP8266)

I read through this issue https://github.com/espressif/arduino-esp32/issues/3902 about WebServerSecure & its related issues

### Describe the solution you'd like

I think about writing the missing "core library class" WifFiServerSecure myself and if at least partially successful, making a pull request (I already have some practice *hacking* libraries for Arduino ESP8266 & STM32s)
Specifically it seems that I could use `eps-tls` component do to this without writing too much new logic - https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_tls.html. It implements TLS client & server related logic. However the fact that the current implementation of WiFiClientSecure does not use esp-tls and instead uses mbedlts directly alarms me that it might not be that straightforward:
https://github.com/espressif/arduino-esp32/blob/2.0.5/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp
https://github.com/espressif/arduino-esp32/blob/2.0.5/libraries/WiFiClientSecure/src/ssl_client.cpp (claims to be adapted ssl_client1 from mbedlts examples).

Given that https://github.com/espressif/arduino-esp32/issues/7447#event-7822089251 is "sort of promising" HTTPS server example, I assume that WebServerSecure itself is scheduled for 3.* and there might be some relevant work already laid and/oror there are people who might know about the reason why esp-tls **is not used** for WiFiClientSecure (did not exists at that time?) and if there would be anything bad with implementing WifFiServerSecure on top of it?

@VojtechBartoska seems to be a good first contact (🙏) 
I can see from blame of the client-related files that @ me-no-dev , @ vlastahajek , @ dirkx, @ chemicstry would be relevant.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2022-11-28:
Hello @michkot, thanks for opening new feature request and your suggestion.

Your are right on our plans, we want to investigate this for next major release 3.0.0.

Due to the fact, you are willing to help with this directly, we will take a look on your proposal and share are opinion if we will be able to decide it now.

@me-no-dev & @SuGlider, feel free to share your comments to this if you have any. Thank you.

SuGlider, 2022-11-28:
Yes, sure. 
I just think that implementing it would require some time and also it would be good to to study possible solutions. 
We can consider it for 3.0.0.

michkot, 2022-11-30:
I've created a very, very raw version that compiles, but have not tried it using it yet. I also pointed out some weird logic/semantics of connect(). 
https://github.com/espressif/arduino-esp32/compare/master...michkot:arduino-esp32:wifi-server-secure
Linking in esp-tls (well whatever get's linked when I add dependency on `esp_tls_server_session_create` and the like) adds ~7kB of used flash.

michkot, 2022-12-02:
I have been able to bring the code to a working condition! (but it's not in a state where I would attempt a PRequest against this repo yet I think)
https://github.com/michkot/arduino-esp32/pull/1

closed 2025-01-16 by Parsaabasi