espressif/arduino-esp32 #7524: InstrFetchProhibited crash on WiFi functions.

zekageri, 2022-11-27:
### Board

ESP32-Wrover-E

### Device Description

Nothing much

### Hardware Configuration

Nothing much

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Board crashes on wifi instruction calls.

I have tried with `WiFi.softAPConfig(apIP, apIP, netMsk);`
And `WiFi.mode(WIFI_STA);`

Both causes the same crash.

### Sketch

```cpp
void setup(){
    Serial.begin(115200);
    WiFi.softAPConfig(apIP, apIP, netMsk); // <-- Crash on here.
    WiFi.softAP(firstBootAP_NAME);
    dnsServer.start(53, "*", WiFi.softAPIP());
    IPAddress myIP = WiFi.softAPIP();
}
```

### Debug Message

```plain
[  2246][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x0c004136  PS      : 0x00060930  A0      : 0x800e1d78  A1      : 0x3ffb2360  
A2      : 0x00000000  A3      : 0x401a41e8  A4      : 0x3ffb2470  A5      : 0x00000000  
A6      : 0x3ffb8a28  A7      : 0x00000000  A8      : 0x800e1d3e  A9      : 0x3ffb2300  
A10     : 0x3ffb2470  A11     : 0x401a41e8  A12     : 0x00000002  A13     : 0x00000001  
A14     : 0x00060b20  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x00000014
EXCVADDR: 0x0c004134  LBEG    : 0x4008bd2d  LEND    : 0x4008bd3d  LCOUNT  : 0xfffffffd

Backtrace:0x0c004133:0x3ffb23600x400e1d75:0x3ffb2380 0x400e1e2d:0x3ffb23a0 0x400e232c:0x3ffb2530 

  #0  0x0c004133:0x3ffb2360 in ?? ??:0
  #1  0x400e1d75:0x3ffb2380 in WiFiEventCbList::WiFiEventCbList(WiFiEventCbList const&) at C:/Users/zekag/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:740
  #2  0x400e1e2d:0x3ffb23a0 in WiFiGenericClass::_eventCallback(arduino_event_t*) at C:/Users/zekag/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:1050
  #3  0x400e232c:0x3ffb2530 in _arduino_event_task(void*) at C:/Users/zekag/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:303

ELF file SHA256: 0000000000000000

Rebooting...
```

### Other Steps to Reproduce

Call any wifi function.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-11-27:
Does it work if you call AP and then config?

zekageri, 2022-11-28:
Sorry I think this crash is caused by one of my libraries. I have tried to use a fork of the Async tcp lib which does not have locking problems but this resulted in that crash. If I replace that fork with an earlier version, it is working fine.

zekageri, 2022-11-28:
But I will check if it works if I call the config after because I would really need that lib.

zekageri, 2023-01-15:
Turns out it was a library problem.

closed 2023-01-15 by zekageri