espressif/arduino-esp32 #7531: Not able to compile the Ble uart code

rjrajbir, 2022-11-29:
### Board

Esp32 Dev Module

### Device Description

Plain Module onboard

### Hardware Configuration

No

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40 MHZ

### PSRAM enabled

no

### Upload speed

921600

### Description

Showing wrong function

### Sketch

```cpp
/*
    Video: https://www.youtube.com/watch?v=oCMOYS71NIU
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleNotify.cpp
    Ported to Arduino ESP32 by Evandro Copercini

   Create a BLE server that, once we receive a connection, will send periodic notifications.
   The service advertises itself as: 6E400001-B5A3-F393-E0A9-E50E24DCCA9E
   Has a characteristic of: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E - used for receiving data with "WRITE" 
   Has a characteristic of: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E - used to send data with  "NOTIFY"

   The design of creating the BLE server is:
   1. Create a BLE Server
   2. Create a BLE Service
   3. Create a BLE Characteristic on the Service
   4. Create a BLE Descriptor on the characteristic
   5. Start the service.
   6. Start advertising.

   In this example rxValue is the data received (only accessible inside that function).
   And txValue is the data to be sent, in this example just a byte incremented every second. 
*/
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLEServer *pServer = NULL;
[... 168 lines cut ...]
BLE_uart:91:31: error: 'PROPERTY_WRITE' is not a member of 'BLECharacteristic'
            BLECharacteristic::PROPERTY_WRITE
                               ^~~~~~~~~~~~~~
BLE_uart:94:22: error: 'class BLECharacteristic' has no member named 'setCallbacks'
   pRxCharacteristic->setCallbacks(new MyCallbacks());
                      ^~~~~~~~~~~~
BLE_uart:97:13: error: 'class BLEService' has no member named 'start'
   pService->start();
             ^~~~~
C:\Users\Embedded PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\BLE\examples\BLE_uart\BLE_uart.ino: In function 'void loop()':
BLE_uart:108:28: error: 'class BLECharacteristic' has no member named 'notify'
         pTxCharacteristic->notify();
                            ^~~~~~
Multiple libraries were found for "BLEDevice.h"
 Used: C:\Users\Embedded PC\Documents\Arduino\libraries\ArduinoBLE
 Not used: C:\Users\Embedded PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\BLE
Using library ArduinoBLE at version 1.3.2 in folder: C:\Users\Embedded PC\Documents\Arduino\libraries\ArduinoBLE 
Using library BLE at version 2.0.0 in folder: C:\Users\Embedded PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\BLE 
exit status 1
'class BLECharacteristic' has no member named 'getValue'; did you mean 'setValue'?
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

rjrajbir, 2022-11-29:
I have downloaded another ArduinoBLE lib that is present in the Documents>libraries folder, that was causing issue, by deleting that lib it is working now

closed 2022-11-29 by rjrajbir

Sh1-Zu3, 2024-11-30:
nice
https://helloraspberrypi.blogspot.com/2021/07/install-bluepy-on-raspberry-pi-for.html

sivar2311, 2024-11-30:
@Sh1-Zu3 How is that related to this repository respective to ESP32-Arduino?