espressif/arduino-esp32 #7532: BLE unstable/unusable on ESP32 revision 3 board

Staacks, 2022-11-29:
### Board

ESP-32 Dev Kit C V4 (with revision 3 ESP32)

### Device Description

ESP-32 Dev Kit C V4 recently bought from az-delivery with a revision 3 chip, tested on about 50 of these boards.
Same code was tested on an older dev board by az-delivery (https://www.az-delivery.de/products/esp32-developmentboard) as well as the older Dev Kit C board without any issue. The obvious difference of the newer board is that it has an ESP32-D0WD-V3 (revision 3), while the older ones all have an ESP32-D0WDQ6 (revision 1).

### Hardware Configuration

The boards were tested bare - just a USB connection to the PC.

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Linux Mint

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Uploaded the BLE_server.ino example to a new ESP-32 Dev Kit C V4 with ESP32-D0WD-V3 and tryed to connect with nRF Connect after its reset. nRF connect can see the board, but after a connection attempt, it immediately disconnects with the (rather generic) error 133 GATT error. The behavior is slightly erratic as retries sometimes succeed, but failure is the more common outcome.

The same code was uploaded to another ESP-32 Dev Kit C (identical board layout as far as I can tell) with an ESP32-D0WDQ6 (revision 1) and the BLE connection works like a charm. Same goes for an even older dev board by az-delivery (hardware details section).

This behavior was observed with about 50 of these boards during a workshop in which we wanted to teach the use of our Arduino library (phyphoxBLE) to teachers, so while not all occurrences have been investigated, we can certainly say that it is not only affecting the one board I am testing more thoroughly now after the workshop failed spectacularly.

On the phone side I have observed this behavior on a Pixel 3 (Android 12), Pixel 6 (Android 13) and Samsung Galaxy S10 5G (Android 12). On iPhones (iPhone 8 and iPhone 14 Pro) the success rate is higher, but disconnects are common and the initial connect in nRF connect can easily take 10 seconds while the older boards connect immediately (I suspect that iOS does some fallback reconnect work in the background or is just more patient).

### Sketch

```cpp
//This is the exact code from the BLE_server.ino example (except for this comment)

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("Long name works now");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic->setValue("Hello World says Neil");
  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}
```

### Debug Message

```plain
[  1343][V][FreeRTOS.cpp:143] give(): Semaphore giving: name: StartEvt (0x3ffdbe30), owner: start
[  1352][V][BLECharacteristic.cpp:205] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_START_EVT
[  1352][V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: StartEvt (0x3ffdbe30), owner: <N/A>
[  1362][V][BLECharacteristic.cpp:465] handleGATTServerEvent(): << handleGATTServerEvent
[  1371][V][BLEService.cpp:159] start(): << start()
[  1379][V][BLEServer.cpp:280] handleGATTServerEvent(): << handleGATTServerEvent
[  1384][I][BLEDevice.cpp:577] getAdvertising(): create advertising
[  1397][D][BLEDevice.cpp:579] getAdvertising(): get advertising
[  1402][V][BLEDevice.cpp:584] startAdvertising(): >> startAdvertising
[  1409][D][BLEDevice.cpp:579] getAdvertising(): get advertising
[  1414][V][BLEAdvertising.cpp:187] start(): >> start: customAdvData: 0, customScanResponseData: 0
[  1423][D][BLEAdvertising.cpp:199] start(): - advertising service: 4fafc201-1fb5-459e-8fcc-c5c9c331914b
[  1432][V][BLEAdvertising.cpp:252] start(): << start
[  1437][V][BLEDevice.cpp:586] startAdvertising(): << startAdvertising
[  1436][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
Characteristic defined! Now you can read it in your phone!
[  1453][V][BLEUtils.cpp:1056] dumpGapEvent(): [status: 0]
[  1463][D][BLEDevice.cpp:579] getAdvertising(): get advertising
[  1469][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 0]
[  1476][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT
[  1486][V][BLEUtils.cpp:1206] dumpGapEvent(): [status: 0]
[  1491][D][BLEDevice.cpp:579] getAdvertising(): get advertising
[  1497][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 1]
[  1504][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
[  1513][V][BLEUtils.cpp:1074] dumpGapEvent(): [status: 0]
[  1518][D][BLEDevice.cpp:579] getAdvertising(): get advertising
[  1524][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 6]

(Note that this is all dumped before the connection attempt and there is no additional output when nRF Connect tries to connect to the device but fails.)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Staacks, 2023-01-19:
Update:
I have bought several ESP32-DevKitC-32E and ESP32-Wroom-32E-N4 and none of them exhibit the issue. Therefore I would say that this issue can be closed here and I have to discuss this with AZ-Delivery.

closed 2023-01-19 by Staacks

ollivierv, 2023-03-05:
@Staacks I have **EXACTLY** the same problem as you. I have TWO "ESP-32 Dev Kit C V4 (with revision 3 ESP32)" boards from AZ-Delivery. One with _ESP32-D0WDQ6_ and the other with a _ESP32-D0WDQ5_, both use ESP32-D0WD-V3. 
I lost days to figure what happened here, the ESP32-D0WDQ6 work perfectly in BLE (Arduino example), the other, the ESP32-D0WDQ5, appear and disappear while PC is scanning BLE, and connection never succeed !
To be **SURE**, I built and used raw example from the Espressif SDK 5.0.1 (ble/gatt_server), and SAME behaviour ! 

I guess there is a **VERY** problematic PCB/Antenna adaptation with  _ESP32-D0WDQ5_ in this design !