espressif/arduino-esp32 #7535: HardwareSerial pin assignation on ESP32-S3

beckmx, 2022-11-29:
### Board

ESP32-S3-DevKitC-1 v1.1

### Device Description

DevkitC nothing connected, plain USB connection to computer

### Hardware Configuration

Nothing else

### Version

latest master (checkout manually)

### IDE Name

Platformio

### Operating System

Macos 12.5.1

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

When configuring the serial ports using the original definitions from the file HardwareSerial https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp looks like the definition for UART1 and UART2 pins for ESP32-S3 do not correspond with those in the original documentation where the GPIO pins are different https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html.

### Sketch

```cpp
Serial.println("original uart0");
//setup all the ports with initial configuration
Serial1.begin(115200);
Serial2.begin(115200);
```

### Debug Message

```plain
HardwareSerial.cpp

#elif CONFIG_IDF_TARGET_ESP32S3
#define RX1 15

#elif CONFIG_IDF_TARGET_ESP32S3
#define TX1 16
```

### Other Steps to Reproduce

I checked the documentation and looks lke it doesnt match with the pins originally defined

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-11-30:
This is a generic Pin out profile.
It may change for each different board.
ESP32 family allows the application to use any pin for UART 0, 1, 2.
Some boards will define their own RX1, TX1, RX2, TX2 etc...
Example: https://github.com/espressif/arduino-esp32/blob/master/variants/adafruit_qtpy_esp32s3_nopsram/pins_arduino.h#L25-L28

If necessary, just use `Serial1.begin(115200, SERIAL_8N1, RxPin, TxPin);` for instance.

beckmx, 2022-11-30:
@SuGlider thanx for the clarification, it is in fact kinda confusing since the devkit docs indicates like the functionality is in there and I just went to the datasheet and yup it says it can switch to any gpio

closed 2022-11-30 by beckmx