espressif/arduino-esp32 #7536: how to burn in "phy_multiple_init_data.bin" into ESP32 S3

savagerex, 2022-11-30:
### Board

ESP32 S3

### Device Description

ESP32-S3-DevKitC-1

### Hardware Configuration

NONE

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

how to burn in "phy_multiple_init_data.bin" into ESP32 S3

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println("Hello");
  delay(2000);
}
```

### Debug Message

```plain
NONE
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-11-30:
Use esp-idf.  This is well outside the scope of Arduino.

me-no-dev, 2022-11-30:
Hi @savagerex . First you need to upload your sketch with custom partitions.csv in the sketch folder that defines where phy.bin will be located, then you have to manually upload it to that location. You can use the esptool that comes with Arduino (enabling flashing debug in the IDE preferences would show you the command that was used to upload the firmware. you can copy it and add phy.bin to it, then reupload)

VojtechBartoska, 2023-02-23:
Closing es Expired due to no answer. If needed, you can reopen it.

closed 2023-02-23 by VojtechBartoska