espressif/arduino-esp32 #7540: WiFi promiscuous timestamp differs by 140+ms

PavelDrabek, 2022-11-30:
### Board

ESP32 Dev Module

### Device Description

Plain module ESP32-D0WDQ6 (revision 1)

### Hardware Configuration

nothing is attached

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 11 Pro

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Timestamp stored in wifi_pkt_rx_ctrl_t (from wifi_promiscuous_pkt_t) received in wifi promiscuous rx callback function differs from micros() called immediately in that function by more than 140,000 microseconds but code is running smoothly. I would expect it differs by 10 microseconds at most. 

Also micros() returns unsigned long and timestamp in wifi_pkt_rx_ctrl_t is unsigned int which might cause some issues when running for a long time. 

### Sketch

```cpp
#include <WiFi.h>
#include <esp_wifi.h>

void promiscuous_rx_cb(void *buf, wifi_promiscuous_pkt_type_t type) {
  unsigned long t = micros();

  const wifi_promiscuous_pkt_t *ppkt = (wifi_promiscuous_pkt_t *)buf;
  unsigned int timestamp = ppkt->rx_ctrl.timestamp;

  String type_str;
  switch(type){
    case WIFI_PKT_MGMT: type_str = "WIFI_PKT_MGMT"; break;
    case WIFI_PKT_CTRL: type_str = "WIFI_PKT_CTRL"; break;
    case WIFI_PKT_DATA: type_str = "WIFI_PKT_DATA"; break;
    case WIFI_PKT_MISC: type_str = "WIFI_PKT_MISC"; break;
  }

  Serial.printf("rx_cb is called: %s\t rt:%u cbt: %lu, tdiff: %lu\n", type_str, timestamp, t, (t - timestamp));
}

void setup() {
  Serial.begin(115200);
  
  WiFi.mode(WIFI_STA);

  esp_wifi_set_promiscuous(true);
  esp_wifi_set_promiscuous_rx_cb(&promiscuous_rx_cb);
}
 
void loop() {

}
```

### Debug Message

```plain
22:30:59.842 -> rx_cb is called: WIFI_PKT_MGMT	 rt:117792901 cbt: 117938159, tdiff: 145258
22:30:59.874 -> rx_cb is called: WIFI_PKT_MGMT	 rt:117792901 cbt: 117963029, tdiff: 170128
22:30:59.874 -> rx_cb is called: WIFI_PKT_MGMT	 rt:117792901 cbt: 117970676, tdiff: 177775
22:30:59.907 -> rx_cb is called: WIFI_PKT_MGMT	 rt:117834673 cbt: 117980429, tdiff: 145756
22:30:59.907 -> rx_cb is called: WIFI_PKT_MGMT	 rt:117848847 cbt: 117990243, tdiff: 141396
22:30:59.938 -> rx_cb is called: WIFI_PKT_MGMT	 rt:117872995 cbt: 118010494, tdiff: 137499
22:30:59.938 -> rx_cb is called: WIFI_PKT_MGMT	 rt:117892352 cbt: 118035762, tdiff: 143410
22:30:59.970 -> rx_cb is called: WIFI_PKT_MGMT	 rt:117892352 cbt: 118040876, tdiff: 148524
22:30:59.970 -> rx_cb is called: WIFI_PKT_MGMT	 rt:117892352 cbt: 118065452, tdiff: 173100
22:31:00.002 -> rx_cb is called: WIFI_PKT_MGMT	 rt:117892352 cbt: 118082899, tdiff: 190547
22:31:00.002 -> rx_cb is called: WIFI_PKT_MGMT	 rt:117892352 cbt: 118092617, tdiff: 200265
22:31:00.045 -> rx_cb is called: WIFI_PKT_MGMT	 rt:117957417 cbt: 118094616, tdiff: 137199
22:31:00.045 -> rx_cb is called: WIFI_PKT_MGMT	 rt:117975395 cbt: 118112903, tdiff: 137508
22:31:00.077 -> rx_cb is called: WIFI_PKT_MGMT	 rt:117998796 cbt: 118142955, tdiff: 144159
22:31:00.077 -> rx_cb is called: WIFI_PKT_MGMT	 rt:118024925 cbt: 118167859, tdiff: 142934
22:31:00.118 -> rx_cb is called: WIFI_PKT_DATA	 rt:118024925 cbt: 118169032, tdiff: 144107
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-11-30:
This is an ESP-IDF function.  You will have better luck asking at https://github.com/espressif/esp-idf/issues

mrengineer7777, 2022-12-01:
esp_wifi_types.h:
`unsigned timestamp:32;        /**< timestamp. The local time when this packet is received. It is precise only if modem sleep or light sleep is not enabled. unit: microsecond */`

uint32_t is too small to store microseconds or milliseconds.  Could it be UTC seconds?

I believe the correct format specifier for long unsigned is "%llu", but I don't think Arduino core supports that.  I haven't been able to get it to work.

closed 2025-01-16 by Parsaabasi