espressif/arduino-esp32 #7542: return from server.streamFile(datafile, contentType) (size_t) defined as unsigned int, too small

stephenginhampshire, 2022-11-30:
### Board

ESP32-WROOM

### Device Description

ESP32-WROOM

### Hardware Configuration

constexpr int WipeSD_switch_pin = 32;
constexpr int Reset_switch_pin = 33;
constexpr int Start_switch_pin = 27;
constexpr int Running_led_pin = 26;
constexpr int Boot_switch_pin = 0;
constexpr int SD_Active_led_pin = 4;
constexpr int RS485_Enable_pin = 22;
constexpr int RS485_TX_pin = 17;
constexpr int RS485_RX_pin = 16;
constexpr int SS_pin = 5;
constexpr int SCK_pin = 18;
constexpr int MISO_pin = 19;
constexpr int MOSI_pin = 23;
constexpr int ONBOARDLED = 2;

### Version

v2.0.5

### IDE Name

Visual Studio with VMicro

### Operating System

Windows10

### Flash frequency

Crystal is 40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

using the Webserver library the server.streamFile(datafile, contentType); subroutine returns an unsigned int indicating the number of bytes written The maximum value of an unsigned int is 65535  (16 bits), this is too small, could it be changed to an unsigned long?

In library software the return from this subroutine is used to compare to  the size of a (SD) datafile as in:
" if (server.streamFile(datafile, contentType) != datafile.size()) "
-t which is defined as an unsigned int, could this be changed to unsigned long?
both returns return size

### Sketch

```cpp
server.sendHeader("Content-Disposition", "attachment; filename=" + filename);  
if (server.streamFile(datafile, contentType) != datafile.size()) {
   console.print(millis(), DEC);
   console.println("\t\tSent less data than expected!");
}
```

### Debug Message

```plain
I get the error message when the file size exceeds the maximumm size of an unsigned int
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-12-01:
int is 32 bit on esp32.

stephenginhampshire, 2022-12-01:
Thank you lbernstone!
In which case I will post another associated issue.

mrengineer7777, 2023-05-21:
Can this issue be closed?