espressif/arduino-esp32 #7543: ESP32-S3 SPI with mutiple devices missing data

dsilletti, 2022-12-01:
### Board

ESP32-S3-WROOM-1 (all flavours)

### Device Description

2 different SPI devices on the same SPI default bus using different CS pins) or on different SPI custom assigned pins.

### Hardware Configuration

//MOSI: 11 //MISO: 13 //SCK: 12 //CS: 10

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

macOS 12.6

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

using 2 SPI slave devices, reading and writing registers from the 1st device and then writing to the 2nd one (TFT display) in a loop (non in the same time), The SPI devices are always missing some data (usually the 1st transmit). All the following data transfers looks fine. The only workaround possible is to duplicate the 1st SPI operation in the code.

The same code is working perfectly fine on the ESP32-WROOM and ESP32-WROVER and on the ESP32-S3 with a single SPI slave device.

On the ESP32S3 with our without PSRAM (tested on the N8R8, N8, N8R2) if there are 2 SPI devices connected the issue is always there. I tried also adding pull-up resistors with no luck.

### Sketch

```cpp
N/A
```

### Debug Message

```plain
No errors in the logs
```

### Other Steps to Reproduce

The same code is working perfectly fine on the ESP32-WROOM and ESP32-WROVER and on the ESP32-S3 with a single SPI slave device.

On the ESP32S3 with our without PSRAM (tested on the N8R8, N8, N8R2) if there are 2 SPI devices connected the issue is always there. I tried also adding pull-up resistors with no luck.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-12-01:
could you please give minimal example code?

dsilletti, 2022-12-08:
Hi @me-no-dev , I finally fixed the issue, basically I had to rewrite some legacy libs to always create a new SPIClass for the additional bus.
Some legacy libs still do SPI.begin() and while this works on the ESP32, because it probably initialise both the pre-defined VSPI and HSPI, on the ESP32-S3 it initialise just the single SPI defined (FSPI). So rewrite your libs to define new SPIClasses it's the only way to go if you need more than 1 SPI bus on the S3.
Thanks for the support, I will close this issue.

closed 2022-12-08 by dsilletti

me-no-dev, 2022-12-08:
@dsilletti I plan to improve this situation soon :)

dsilletti, 2022-12-09:
Thanks, there is not much documentation around for the ESP32S3, for example that the usable SPI ports are:

FSPI port (SPI2):
//MOSI: GPIO 11 //MISO: GPIO 13 //SCK: GPIO 12 //CS: GPIO 10 (or custom via SPIClass)

HSPI port (SPI3):
GPIO custom pins to assign via SPIClass

MattPChoy, 2023-02-06:
@dsilletti Hi, I'm trying to get an SD card working on the Adafruit Feather ESP32-S3 chip (the one with 2MB of PSRAM which uses the SPI interface as I understand).

Could you provide more detailed code on how to use SPIClass to define custom GPIO Pins? Thanks.

mrnsms, 2023-03-17:
@MattPChoy did you figure out how to do it? I'm also with that problem

tranvnhan, 2023-04-27:
> Thanks, there is not much documentation around for the ESP32S3, for example that the usable SPI ports are:
> 
> FSPI port (SPI2): //MOSI: GPIO 11 //MISO: GPIO 13 //SCK: GPIO 12 //CS: GPIO 10 (or custom via SPIClass)
> 
> HSPI port (SPI3): GPIO custom pins to assign via SPIClass
> 
> so most of the libraries are not able to use the HSPI out of the box (for the FSPI I see you are doing the mapping in the Arduino core SPI lib but probably is not well documented that the name is FSPI and not VSPI)

Could you please provide a minimal example? I did the same with my ESP32-S3 and it still didn't work. I am using SPI to interface with an SD card module. Thanks.