espressif/arduino-esp32 #7545: RMaker Agent connection to Rmaker Cloud

atanudasgupta, 2022-12-02:
### Related area

How to check if RMaker agent is conneced to the cloud?

### Hardware specification

esp32

### Is your feature request related to a problem?

I need to know how to check if RMaker agent or the device is connected to Rainmaker cloud. Something the service provider changes the IP addresses in middle of the night and the device gets disconnected, I end up restarting the WIFI router in the morning again.

How can I check if RMaker Agent is still connected to the cloud and then do a RMakerWiFiReset() or how can allocate a static IP to the device?

### Describe the solution you'd like

if such an API is not there, please suggest alternative

### Describe alternatives you've considered

just checking WIFI.status() does not work, since stale IP may be hanging there.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2022-12-02:
Hello @atanudasgupta, thanks for your contribution. @sanketwadekar Can you please help with triage? Thank you.

mrengineer7777, 2022-12-02:
If this is a Rainmaker question try https://github.com/espressif/esp-rainmaker/issues

atanudasgupta, 2022-12-02:
Hi David,
I am using Arduino with rainmaker .
Regards,Atanu

atanudasgupta, 2022-12-03:
sure I have opend another issue in Rainmaker if that helps