espressif/arduino-esp32 #7547: Arduino LCD display project

JennyGilbert123, 2022-12-04:
### Related area

Error Debugging

### Hardware specification

ESP-32

### Is your feature request related to a problem?

I am working on this basic project where I am using ESP-32 and the LCD model is 1602. There is nothing wrong with the connections, but whenever I 
[image]
[image]
[image]
try to upload the code on Arduino, the following errors appear. Kindly help me how to rectify this problem. 

### Describe the solution you'd like

I would appreciate if someone is willing to help me overcome this problem.

### Describe alternatives you've considered

I have removed the wires and rearranged the circuit multiple times but it doesn't work. 

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2022-12-04:
Please expand that box with the error, and copy/paste the whole error into here as text.

JennyGilbert123, 2022-12-04:
#include <LiquidCrystal_I2C.h>
#include <Wire.h>
#define SDA 13 
#define SCL 14
LiquidCrystal_I2C lcd(0x27,16,2);
void setup() {
  // put your setup code here, to run once:
Wire.begin(SDA, SCL);
lcd.init(); 
lcd.backlight();
lcd.setCursor(0,0);
lcd.print("I love my husband! ");
}

void loop() {
  // put your main code here, to run repeatedly:
lcd.setCursor(0,1);
lcd.print("Counter:"); 
lcd.print(millis() / 1000);
delay(1000);

}

lbernstone, 2022-12-04:
Please read what I posted.  Expand the section with the error, and post the error.  Please note, this forum is for issues with the code here.  It is not a tutorial site, and we do not provide assistance with 3rd party libraries (ie, the Library Manager) or the Arduino IDE.

SuGlider, 2022-12-10:
@JennyGilbert123 - It seems that the `LiquidCrystal_I2c.h` file is not part of the installed Library.
This is the necessary Library: https://www.arduinolibraries.info/libraries/liquid-crystal-i2-c

[image]

JennyGilbert123, 2022-12-13:
Thankyou for providing the link to the library. I have installed it and the program is working now :)