espressif/arduino-esp32 #7548: ESP32, RF24

GOLLAVILLISANJAYKUMAR, 2022-12-05:
### Related area

esp32   RF24

### Hardware specification

ESP32 WROOM DEV KIT

### Is your feature request related to a problem?

c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:15:6: error: prototype for 'void RF24::csn(bool)' does not match any in class 'RF24'
 void RF24::csn(bool mode)
      ^
In file included from c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:11:0:
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.h:78:8: error: candidate is: void RF24::csn(int)
   void csn(int mode);
        ^
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:103:6: error: prototype for 'void RF24::ce(bool)' does not match any in class 'RF24'
 void RF24::ce(bool level)
      ^
In file included from c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:11:0:
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.h:86:8: error: candidate is: void RF24::ce(int)
   void ce(int level);
        ^
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:113:36: error: no 'void RF24::beginTransaction()' member function declared in class 'RF24'
 inline void RF24::beginTransaction()
                                    ^
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:131:34: error: no 'void RF24::endTransaction()' member function declared in class 'RF24'
 inline void RF24::endTransaction()
                                  ^
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:145:6: error: prototype for 'void RF24::read_register(uint8_t, uint8_t*, uint8_t)' does not match any in class 'RF24'
 void RF24::read_register(uint8_t reg, uint8_t* buf, uint8_t len)
      ^
In file included from c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:11:0:
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.h:104:11: error: candidates are: uint8_t RF24::read_register(uint8_t)
   uint8_t read_register(uint8_t reg);
           ^
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.h:96:11: error:                 uint8_t RF24::read_register(uint8_t, uint8_t*, uint8_t)
   uint8_t read_register(uint8_t reg, uint8_t* buf, uint8_t len);
           ^
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp: In member function 'uint8_t RF24::read_register(uint8_t)':
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:220:22: error: 'beginTransaction' was not declared in this scope
     beginTransaction();
                      ^
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:222:5: error: 'status' was not declared in this scope
     status = _spi->transfer(R_REGISTER | reg);
     ^
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:222:14: error: '_spi' was not declared in this scope
     status = _spi->transfer(R_REGISTER | reg);
              ^
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:230:20: error: 'endTransaction' was not declared in this scope
     endTransaction();
                    ^
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp: At global scope:
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:238:6: error: prototype for 'void RF24::write_register(uint8_t, const uint8_t*, uint8_t)' does not match any in class 'RF24'
 void RF24::write_register(uint8_t reg, const uint8_t* buf, uint8_t len)
      ^
In file included from c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:11:0:
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.h:123:11: error: candidates are: uint8_t RF24::write_register(uint8_t, uint8_t)
   uint8_t write_register(uint8_t reg, uint8_t value);
           ^
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.h:114:11: error:                 uint8_t RF24::write_register(uint8_t, const uint8_t*, uint8_t)
   uint8_t write_register(uint8_t reg, const uint8_t* buf, uint8_t len);
           ^
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:281:6: error: prototype for 'void RF24::write_register(uint8_t, uint8_t, bool)' does not match any in class 'RF24'
 void RF24::write_register(uint8_t reg, uint8_t value, bool is_cmd_only)
      ^
In file included from c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:11:0:
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.h:123:11: error: candidates are: uint8_t RF24::write_register(uint8_t, uint8_t)
   uint8_t write_register(uint8_t reg, uint8_t value);
[... 377 lines cut ...]
                             ^
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:2018:41: error: no 'void RF24::toggleAllPipes(bool)' member function declared in class 'RF24'
 void RF24::toggleAllPipes(bool isEnabled)
                                         ^
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:2025:57: error: no 'uint8_t RF24::_data_rate_reg_value(rf24_datarate_e)' member function declared in class 'RF24'
 uint8_t RF24::_data_rate_reg_value(rf24_datarate_e speed)
                                                         ^
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:2058:64: error: no 'uint8_t RF24::_pa_level_reg_value(uint8_t, bool)' member function declared in class 'RF24'
 uint8_t RF24::_pa_level_reg_value(uint8_t level, bool lnaEnable)
                                                                ^
c:\Users\user\Documents\Arduino\libraries\RF24-master\RF24.cpp:2068:77: error: no 'void RF24::setRadiation(uint8_t, rf24_datarate_e, bool)' member function declared in class 'RF24'
 void RF24::setRadiation(uint8_t level, rf24_datarate_e speed, bool lnaEnable)
                                                                             ^

exit status 1

Compilation error: exit status 1

### Describe the solution you'd like

I don't know the reason why I encountered the problem

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

GOLLAVILLISANJAYKUMAR, 2022-12-05:
[RF24.zip](https://github.com/espressif/arduino-esp32/files/10150167/RF24.zip)

VojtechBartoska, 2022-12-05:
Hello @GOLLAVILLISANJAYKUMAR, what's the purpose of this "feature request". I see some library in attached zip but the report explain nothing.

Do you face some issue or is the some kind of feature request?

Please update the description and provide more and clearer info.

Thank you!

Jason2866, 2022-12-05:
This is NOT a forum. This github is ONLY for issues in Arduino esp32 code.
Your problem is a 3rd party software, not the place to address issues with here.

closed 2023-02-23 by VojtechBartoska