espressif/arduino-esp32 #7549: Unable to flash ESP32

primeBiceps, 2022-12-05:
### Board

ESP-WROOM-32

### Device Description

[image]

### Hardware Configuration

GPIO 32-35 are conencted to a decoder and GPIO 22, 23 are inputs. Other pins are floating while programming.

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

When i try uploading the sketch to the custom board with ESP32 I get the following error.

[image]

I tried it BOOT pin pulled to GND, pulled to GND until the uploading reaches connnecting... status, holding the CPU in reset as I read in some forums but none of the options helped.

### Sketch

```cpp
int a1 = 6;
int b1 = 9;
int c1 = 8;
int d1 = 7;

void setup() {
  pinMode(a1, OUTPUT);
  pinMode(b1, OUTPUT);
  pinMode(c1, OUTPUT);
  pinMode(d1, OUTPUT);

}

void loop() {
  digitalWrite(a1, LOW);
  digitalWrite(b1, LOW);
  digitalWrite(c1, LOW);
  digitalWrite(d1, LOW);

  delay(500);

  digitalWrite(a1, HIGH);
  digitalWrite(b1, LOW);
  digitalWrite(c1, HIGH);
  digitalWrite(d1, LOW);
  
  delay(500);

}
```

### Debug Message

```plain
![image](https://user-images.githubusercontent.com/33923770/205628895-f32f597a-4793-4f11-815b-cb294bb861d5.png)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-12-05:
Your module has probably no flash chip. You have to connect a flash chip externally.

lbernstone, 2022-12-05:
That sketch is certainly not going to work, since it is using pins assigned to the SPI flash.  However, that wouldn't keep it from flashing.  You probably have something pulling gpio12 up during boot, so it isn't supplying power properly to the flash.

primeBiceps, 2022-12-07:
> Your module has probably no flash chip. You have to connect a flash chip externally.

I am using an external FTDI module FT232RL.

primeBiceps, 2022-12-07:
> That sketch is certainly not going to work, since it is using pins assigned to the SPI flash. However, that wouldn't keep it from flashing. You probably have something pulling gpio12 up during boot, so it isn't supplying power properly to the flash.

Actually most of those connections are off board, so while flashing the pins are floating. The only on-board connected chips are on pins IO32...IO35 and IO22 and IO23. So pins assigned to the SPI flash and gpio12 don't have anything connected and are floating. I measured voltage on gpio12 and its around 1.3V.

primeBiceps, 2022-12-07:
> The error you see is typical when no flash chip is found or flash chip is defect. I do not talk about the USB serial module used to "talk" to the eap32 via serial.

Ok, I understand what you are saying. The supplyer specs state that the module should have 4MB flash chip.

[image]

Do you know a way to test if the flash chip is defect? Is there any forum about using and setting the external flash?

Thanks for the help.

Jason2866, 2022-12-07:
Just try a different module. If this one works, you know the first one is defect.

VojtechBartoska, 2023-02-16:
Hello, closing this issues as expired due to no answer. If needed, you can reopen this.

closed 2023-02-16 by VojtechBartoska